moveit/moveit2_tutorials #587: moveit_cpp_tutorial-4] [INFO] [1675937187.657785499] [moveit_ros.current_state_monitor]: Didn't received robot state (joint angles) with recent timestamp within 1.000000 seconds. [moveit_cpp_tutorial-4] Check clock synchronization if your are running ROS across multiple machines! [moveit_cpp_tutorial-4] [ERROR] [1675937187.657812738] [move_group_interface]: Failed to fetch current robot state

akhilkp777, 2023-02-09:
I tried to add constraint motion in ros2 moveit cpp code, but  got the above error. some github solutions are suggested to remap the joint_state topic. But ut doesn't work for me. Anyone know how to solve this issue

qboticslabs, 2023-03-03:
@akhilkp777 If you install moveit from apt, you can try the following fix

Execute following command
```
sudo gedit /opt/ros/humble/lib/python3.10/site-packages/moveit_configs_utils/launches.py
```
Add  "use_sim_time": True, on line after
```
"monitor_dynamics": False,
```

So final code will be

```
    move_group_configuration = {
        "publish_robot_description_semantic": True,
        "allow_trajectory_execution": LaunchConfiguration("allow_trajectory_execution"),
        # Note: Wrapping the following values is necessary so that the parameter value can be the empty string
        "capabilities": ParameterValue(
            LaunchConfiguration("capabilities"), value_type=str
        ),
        "disable_capabilities": ParameterValue(
            LaunchConfiguration("disable_capabilities"), value_type=str
        ),
        # Publish the planning scene of the physical robot so that rviz plugin can know actual robot
        "publish_planning_scene": should_publish,
        "publish_geometry_updates": should_publish,
        "publish_state_updates": should_publish,
        "publish_transforms_updates": should_publish,
        "monitor_dynamics": False,
        "use_sim_time": True,        
    }

```
Save it and try execute the code

[Reference](https://github.com/ros-planning/moveit2/issues/1811): 

If you setup moveit from source code, i think this fix is already done, if not you can add this into same file.

yeyanlei, 2023-04-06:
@qboticslabs @akhilkp777  GitHub question, translated into English: @qboticslabs @akhilkp777 I have encountered the same issue, and I have set "use_sim_time": True in both the /opt/ros/humble/lib/python3.10/site-packages/moveit_configs_utils/launches.py file and the moveit.launch.py file. In Rviz, MoveIt can plan and execute, but when using a C++ program, it reports the following error. Could you please provide assistance? Thank you.

ros2 launch moveit2_tutorials move_group_interface_tutorial.launch.py 
[INFO] [launch]: All log files can be found below /home/robot/.ros/log/2023-04-06-15-38-16-583006-c-80920
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [move_group_interface_tutorial-1]: process started with pid [80921]
[move_group_interface_tutorial-1] [INFO] [1680766696.896307773] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.0278734 seconds
[move_group_interface_tutorial-1] [INFO] [1680766696.896364812] [moveit_robot_model.robot_model]: Loading robot model 'panda'...
[move_group_interface_tutorial-1] [INFO] [1680766696.930198795] [move_group_interface]: Ready to take commands for planning group panda_arm.
[move_group_interface_tutorial-1] [INFO] [1680766696.942789451] [moveit_ros.current_state_monitor]: Listening to joint states on topic 'joint_states'
[move_group_interface_tutorial-1] [WARN] [1680766696.944043430] [moveit_ros.current_state_monitor]: Unable to update multi-DOF joint 'virtual_joint':Failure to lookup transform between 'world'and 'panda_link0' with TF exception: "panda_link0" passed to lookupTransform argument source_frame does not exist. 
[move_group_interface_tutorial-1] [INFO] [1680766697.942866961] [moveit_ros.current_state_monitor]: Didn't received robot state (joint angles) with recent timestamp within 1.000000 seconds.
[move_group_interface_tutorial-1] Check clock synchronization if your are running ROS across multiple machines!
[move_group_interface_tutorial-1] [ERROR] [1680766697.942905819] [move_group_interface]: Failed to fetch current robot state
[move_group_interface_tutorial-1] Stack trace (most recent call last):
[move_group_interface_tutorial-1] #4    Object "", at 0xffffffffffffffff, in 
[move_group_interface_tutorial-1] #3    Object "/home/robot/ws_moveit2/install/moveit2_tutorials/lib/moveit2_tutorials/move_group_interface_tutorial", at 0x56052e7debe4, in _start
[move_group_interface_tutorial-1] #2    Source "../csu/libc-start.c", line 392, in __libc_start_main_impl [0x7faeff1c1e3f]
[move_group_interface_tutorial-1] #1    Source "../sysdeps/nptl/libc_start_call_main.h", line 58, in __libc_start_call_main [0x7faeff1c1d8f]
[move_group_interface_tutorial-1] #0    Object "/home/robot/ws_moveit2/install/moveit2_tutorials/lib/moveit2_tutorials/move_group_interface_tutorial", at 0x56052e7dbb67, in main
[move_group_interface_tutorial-1] Segmentation fault (Address not mapped to object [(nil)])
[ERROR] [move_group_interface_tutorial-1]: process has died [pid 80921, exit code -11, cmd '/home/robot/ws_moveit2/install/moveit2_tutorials/lib/moveit2_tutorials/move_group_interface_tutorial --ros-args -r __node:=move_group_interface_tutorial --params-file /tmp/launch_params_gklulxa2 --params-file /tmp/launch_params_nqt__koh --params-file /tmp/launch_params_hr4b0gq_'].

yeyanlei, 2023-04-06:
const moveit::core::JointModelGroup* joint_model_group =
      move_group.getCurrentState()->getJointModelGroup(PLANNING_GROUP);

log: 

[move_group_interface_tutorial-1] [INFO] [1680775155.255027553] [moveit_ros.current_state_monitor]: Didn't receive robot state (joint angles) with recent timestamp within 1.000000 seconds. Requested time 1680775154.254896, but latest received state has time 454.390000.

yeyanlei, 2023-04-07:
some additional information added as follows:

/gazebo:
  enable_performance_metrics
  publish_rate
  qos_overrides./clock.subscription.depth
  qos_overrides./clock.subscription.durability
  qos_overrides./clock.subscription.history
  qos_overrides./clock.subscription.reliability
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  use_sim_time

ros2 param get /gazebo use_sim_time
Boolean value is: True

yeyanlei, 2023-04-07:
I have already resolved this problem, thank you everyone. by adding {"use_sim_time": True}

from launch import LaunchDescription
from launch_ros.actions import Node
from moveit_configs_utils import MoveItConfigsBuilder

def generate_launch_description():
    moveit_config = MoveItConfigsBuilder("moveit_resources_panda").to_moveit_configs()

    # MoveGroupInterface demo executable
    move_group_demo = Node(
        name="move_group_interface_tutorial",
        package="moveit2_tutorials",
        executable="move_group_interface_tutorial",
        output="screen",
        parameters=[
            moveit_config.robot_description,
            moveit_config.robot_description_semantic,
            moveit_config.robot_description_kinematics,
            {"use_sim_time": True},
        ],
    )

    return LaunchDescription([move_group_demo])

henningkayser, 2023-05-17:
@akhilkp777 does `use_sim_time` also fix this issue for you?

okvik, 2023-07-19:
I've encountered the same issue without simulation in Gazebo involved. Setting `use_sim_time:=True` for all the nodes fixes the issue.

closed 2023-12-01 by tylerjw

devika-dudo, 2025-04-16:
Hi, I am also encountering the same issue, but setting use_sim_time to true doesn't resolve my issue  - please send help