moveit/moveit2_tutorials #641: Update MoveIt Setup Assistant documentation for ROS Humble

Robotawi, 2023-03-18:
### Description

The current documentation for MoveIt Setup Assistant still refers to Noetic related instructions. I want to update the documentation to include instructions specifically for ROS Humble, making it easier for users to set up MoveIt 2 robot_config packages.

I am willing to work on this issue and update the documentation as needed. I plan to do the following:

- Update the documentation to include instructions specific to ROS Humble.
- Separate the explanation of the MSA usage from Moveit-Gazebo integration.
- Test the updated documentation and submit a pull request.

I will use the **panda.urdf** model from the **moveit_resources_panda_description** package instead of the official package from Franka Emika to avoid problems due to updating the official package. Maybe later, we can update again once things are more stable. 

Any feedback or suggestions to make a better documentation are welcome. Thank you! 

### Your environment
* ROS Distro: Humble
* OS Version: e.g. Ubuntu 22.04
* Source build
* git commit: 

### Expected behaviour
An up-to-date documentation that provides users with the guidance they need to successfully set up MoveIt 2 robot_config packages on ROS Humble.

airbutt, 2023-03-24:
That is an excellent Idea, In fact it took me a bit just to do the Panda set up from Scratch with humble due to some of the Issues with the UTF-8  and understanding of the controller managers. I managed even to generate my own Robot( 5-Axis) URDF from Autodesk 360 fusion and put it after some adaptions to the MSA. I am just stucked how to easy get use of Gazebo. I was able to write a complex own launchfile with all the controller manager to start and so forth but may be there is an easier way to reuse the MSA created launchfiles ?!  somewhere the sync ( use_sim_time   value=true ) needs to be added as well I bet. So  the I am curious to see how you are doing it. I documented for an assignment all the steps through the MSA (with humble) but not for the panda, but I did it in parallel and it worked well for me.

Robotawi, 2023-03-30:
Thanks for your comment @airbutt! 

Yes, the auto-generated config and launch files make it easier. Did you see the ros2_control_demos [examples](https://github.com/ros-controls/ros2_control_demos) and their [documentation](https://control.ros.org/master/doc/ros2_control_demos/doc/index.html)?
 
I updated the MSA tutorial, and what you mentioned is relevant but it is better to consider in another tutorial that refers to the MSA tutorial. I will make it and update you.