moveit/moveit2_tutorials #664: Fix workaround for multiversion overwriting API html

henningkayser, 2023-04-15:
We are currently deploying the [API docs](https://moveit.picknik.ai/main/doc/api/api.html) by first [building the pages in separate build jobs](https://github.com/ros-planning/moveit2_tutorials/blob/main/.github/workflows/deploy.yml#L69) (including Doxygen and Python autodocs API) and then [copying the build artifacts into the multiversion build directory](https://github.com/ros-planning/moveit2_tutorials/blob/main/.github/workflows/deploy.yml#L120). The [multiversion build](https://github.com/ros-planning/moveit2_tutorials/blob/main/.github/workflows/deploy.yml#L139) then rebuilds the pages, successfully picking up the doxygen tags, but obviously failing to recreate the autodoc references. This is not really a problem since they are provided with the copied build artifacts, but the [Python API html](https://github.com/ros-planning/moveit2_tutorials/blob/main/doc/api/python_api/api.rst) remain emtpy and overwrites the build artifacts version.

The current workaround: [overwriting the api.html with the artifacts version again](https://github.com/ros-planning/moveit2_tutorials/blob/main/.github/workflows/deploy.yml#L143)
Problems:
* The Python API pages won't have the version switcher included by multiversion
* We can't use the autodoc roles for referencing the API from the tutorial without using the same workaround, multiversion will fail for all autodoc rules without notice even though they would function well on local builds

These are not critical or urgent, but somewhat annoying so we should probably investigate alternatives to multiversion that make it easier to support our current workflow.
For now, we will have to "hardcode" references to the Python API docs, and we should add adequate instructions for that.

PeterMitrano, 2023-04-15:
@rhaschke I am going to look into whether we can use this same approach to deploy python docs for MoveIt 1.

PeterMitrano, 2023-04-15:
Another low-priority problem I think I've found:

On [this page](https://moveit.picknik.ai/main/doc/api/python_api/_autosummary/moveit.core.robot_model.html#moveit.core.robot_model.RobotModel), the link to "View on github" gives a 404, trying to link to [this page](https://github.com/ros-planning/moveit2_tutorials/blob/main/doc/api/python_api/_autosummary/moveit.core.robot_model.rst).