moveit/moveit2_tutorials #684: Update RealTime Servo tutorial

abake48, 2023-05-22:
### Description

With the recent changes to MoveIt_Servo(primarily [this PR](https://github.com/ros-planning/moveit2/commit/dc7643502adbeaa41dfd370605460cfd45f26b7f)), the RealTime Servo tutorial no longer builds successfully because `moveit_servo/servo_parameters.h` no longer exists and probably among other things. Will need to update `realtime_servo/servo_cpp_interface_demo.cpp`

### Your environment
* ROS Distro: [Rolling]
* OS Version: Ubuntu 22.04
* Source build
* Commit Hash dc76435 and beyond.

### Steps to reproduce
Build MoveIt 2 Tutorials and MoveIt 2 from source

### Expected behaviour
Everything should build properly

### Backtrace or Console output

```
/home/parallels/moveit2_ws/src/moveit2_tutorials/doc/examples/realtime_servo/src/servo_cpp_interface_demo.cpp:45:10: fatal error: moveit_servo/servo_parameters.h: No such file or directory
   45 | #include <moveit_servo/servo_parameters.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

ibrahiminfinite, 2023-05-24:
Hey @abake48 
I am currently on rolling  (main on moveit2 and moveit2_tutorials) and everything builds without error for me.

sea-bass, 2023-05-24:
@abake48 I just tried this as well and it worked for me as well. I don't even see the include statement in your error message in my `servo_cpp_interface_demo.cpp`, as this was fixed by @AndyZe in https://github.com/ros-planning/moveit2_tutorials/pull/672.

Could it be that you have an older version of the `main` branch, or another branch, checked out?

closed 2023-05-24 by abake48