moveit/moveit2_tutorials #706: Crash when launching isaac_moveit.py

HedgeHao, 2023-06-17:
Following the [tutorial](https://moveit.picknik.ai/main/doc/how_to_guides/isaac_panda/isaac_panda_tutorial.html). When launching Isaac Sim by `./python.sh isaac_moveit.py` error occurs.

```
Traceback (most recent call last):
  File "/home/hedgehao/Documents/ros/moveit2_tutorials/doc/how_to_guides/isaac_panda/launch/isaac_moveit.py", line 260, in <module>
    rs_viewport.dock_in(viewport, omni.ui.DockPosition.RIGHT)
AttributeError: 'NoneType' object has no attribute 'dock_in'
Exception ignored in: <function _make_registry.<locals>._Registry.__del__ at 0x7fc60f10bcb0>
Traceback (most recent call last):
  File "/home/hedgehao/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/extscore/omni.kit.viewport.registry/omni/kit/viewport/registry/registry.py", line 103, in __del__
  File "/home/hedgehao/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/extscore/omni.kit.viewport.registry/omni/kit/viewport/registry/registry.py", line 98, in destroy
TypeError: 'NoneType' object is not callable
Exception ignored in: <function _make_registry.<locals>._Registry.__del__ at 0x7fc60f10bcb0>
Traceback (most recent call last):
  File "/home/hedgehao/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/extscore/omni.kit.viewport.registry/omni/kit/viewport/registry/registry.py", line 103, in __del__
  File "/home/hedgehao/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/extscore/omni.kit.viewport.registry/omni/kit/viewport/registry/registry.py", line 98, in destroy
TypeError: 'NoneType' object is not callable
Exception ignored in: <function SettingChangeSubscription.__del__ at 0x7fc6d7fa0c20>
Traceback (most recent call last):
  File "/home/hedgehao/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/kernel/py/omni/kit/app/_impl/__init__.py", line 114, in __del__
AttributeError: 'NoneType' object has no attribute 'get_settings'
Exception ignored in: <function RegisteredActions.__del__ at 0x7fc5242cfe60>
Traceback (most recent call last):
  File "/home/hedgehao/.local/share/ov/pkg/isaac_sim-2022.2.1/extscache/omni.kit.viewport.menubar.lighting-104.0.9/omni/kit/viewport/menubar/lighting/actions.py", line 347, in __del__
  File "/home/hedgehao/.local/share/ov/pkg/isaac_sim-2022.2.1/extscache/omni.kit.viewport.menubar.lighting-104.0.9/omni/kit/viewport/menubar/lighting/actions.py", line 352, in destroy
TypeError: 'NoneType' object is not callable
2023-06-17 01:55:45 [22,169ms] [Warning] [carb.audio.context] 1 contexts were leaked
./python.sh: line 41: 449961 Segmentation fault      (core dumped) $python_exe "$@" $args
There was an error running python
```

Neo-manchester, 2023-06-25:
Yeah...I got the same problem like that : (

eladpar, 2023-07-11:
got the same

MarqRazz, 2023-07-14:
Really sorry for the slow response here! If you can test out #716 that would be great.

eladpar, 2023-07-16:
> Really sorry for the slow response here! If you can test out #716 that would be great.

Hey now i get this errors:
[image]
`2023-07-16 07:34:15 [12,648ms] [Error] [carb.graphics-vulkan.plugin] VkResult: ERROR_OUT_OF_DEVICE_MEMORY
2023-07-16 07:34:15 [12,648ms] [Error] [carb.graphics-vulkan.plugin] vkAllocateMemory failed for flags: 0.
2023-07-16 07:34:15 [12,648ms] [Error] [gpu.foundation.plugin] Unable to allocate buffer
2023-07-16 07:34:15 [12,648ms] [Error] [gpu.foundation.plugin] Buffer creation failed for the device: 0.
2023-07-16 07:34:15 [12,648ms] [Error] [rtx.raytracing.plugin] Failed to create buffers for picking results!
2023-07-16 07:34:15 [12,651ms] [Error] [carb.graphics-vulkan.plugin] VkResult: ERROR_OUT_OF_DEVICE_MEMORY
2023-07-16 07:34:15 [12,651ms] [Error] [carb.graphics-vulkan.plugin] vkAllocateMemory failed for flags: 0.
2023-07-16 07:34:15 [12,651ms] [Error] [gpu.foundation.plugin] Unable to allocate buffer
2023-07-16 07:34:15 [12,651ms] [Error] [gpu.foundation.plugin] Buffer creation failed for the device: 0.
2023-07-16 07:34:15 [12,651ms] [Error] [rtx.raytracing.plugin] Failed to create buffers for picking results!
2023-07-16 07:34:15 [12,653ms] [Error] [carb.graphics-vulkan.plugin] VkResult: ERROR_OUT_OF_DEVICE_MEMORY
2023-07-16 07:34:15 [12,653ms] [Error] [carb.graphics-vulkan.plugin] vkAllocateMemory failed for flags: 0.
2023-07-16 07:34:15 [12,653ms] [Error] [gpu.foundation.plugin] Texture creation failed for the device: 0.
2023-07-16 07:34:15 [12,653ms] [Error] [gpu.foundation.plugin] Failed to update params for RenderOp 92
2023-07-16 07:34:15 [12,653ms] [Error] [gpu.foundation.plugin] Failed to update params for RenderOp GBuffer RT. Will not execute this or subsequent RenderGraph operations. Aborting RenderGraph execution
2023-07-16 07:34:15 [12,653ms] [Error] [carb.scenerenderer-rtx.plugin] Failed to execute RenderGraph on device 0. Error Code: 7
2023-07-16 07:34:15 [12,721ms] [Error] [carb.graphics-vulkan.plugin] VkResult: ERROR_OUT_OF_DEVICE_MEMORY
2023-07-16 07:34:15 [12,721ms] [Error] [carb.graphics-vulkan.plugin] vkAllocateMemory failed for flags: 0.
2023-07-16 07:34:15 [12,721ms] [Error] [gpu.foundation.plugin] Unable to allocate buffer
2023-07-16 07:34:15 [12,721ms] [Error] [gpu.foundation.plugin] Buffer creation failed for the device: 0.
2023-07-16 07:34:15 [12,721ms] [Error] [rtx.raytracing.plugin] Failed to create buffers for picking results!
2023-07-16 07:34:15 [12,722ms] [Error] [carb.graphics-vulkan.plugin] VkResult: ERROR_OUT_OF_DEVICE_MEMORY
2023-07-16 07:34:15 [12,722ms] [Error] [carb.graphics-vulkan.plugin] vkAllocateMemory failed for flags: 0.
2023-07-16 07:34:15 [12,722ms] [Error] [gpu.foundation.plugin] Unable to allocate buffer
2023-07-16 07:34:15 [12,722ms] [Error] [gpu.foundation.plugin] Buffer creation failed for the device: 0.
2023-07-16 07:34:15 [12,722ms] [Error] [rtx.raytracing.plugin] Failed to create buffers for picking results!
2023-07-16 07:34:15 [12,722ms] [Error] [carb.graphics-vulkan.plugin] VkResult: ERROR_OUT_OF_DEVICE_MEMORY
2023-07-16 07:34:15 [12,722ms] [Error] [carb.graphics-vulkan.plugin] vkAllocateMemory failed.
2023-07-16 07:34:15 [12,722ms] [Error] [gpu.foundation.plugin] Texture creation failed for the device: 0.
2023-07-16 07:34:15 [12,761ms] [Error] [carb.graphics-vulkan.plugin] VkResult: ERROR_OUT_OF_DEVICE_MEMORY
2023-07-16 07:34:15 [12,761ms] [Error] [carb.graphics-vulkan.plugin] vkAllocateMemory failed for flags: 2.
2023-07-16 07:34:15 [12,761ms] [Error] [gpu.foundation.plugin] Texture creation failed for the device: 0.
2023-07-16 07:34:15 [12,761ms] [Error] [carb.graphics-vulkan.plugin] VkResult: ERROR_OUT_OF_DEVICE_MEMORY
2023-07-16 07:34:15 [12,761ms] [Error] [carb.graphics-vulkan.plugin] vkAllocateMemory failed for flags: 2.
2023-07-16 07:34:15 [12,761ms] [Error] [gpu.foundation.plugin] Texture creation failed for the device: 0.`

PR #716 (2023-07-18): Default ROS_DOMAIN_ID if not set for Isaac tutorial
### Description

Fixes #706. 

This PR adds a check to the Isaac python launch script to make sure a valid `ROS_DOMAIN_ID` is set and if not defaults to a value of 0.

### Checklist
- [ ] **Required by CI**: Code is auto formatted using [clang-format](http://moveit.ros.org/documentation/contributing/code)
- [ ] While waiting for someone to review your request, please consider reviewing [another open pull request](https://github.com/ros-planning/moveit2/pulls) to support the maintainers

[//]: # "You can expect a response from a maintainer wit [... cut]
closed 2023-07-18 by MarqRazz