moveit/moveit2_tutorials #714: build failed. "moveit_core::moveit_butterworth_parameters" but the target was not found.

1172534699, 2023-07-02:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: [Foxy|Galactic|Humble|Rolling]
* OS Version: e.g. Ubuntu 22.04
* Source or Binary build?
* If binary, which release version?
* If source, which git commit or tag?

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.

### Expected behaviour
Tell us what should happen

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

tengfoonglam, 2023-07-18:
Hello, I am facing the same problem. But I managed to build by running `colcon build --mixin release` in the `~/ws_moveit/src` directory instead of `~/ws_moveit/` . Not sure why this is the case atm

JafarAbdi, 2023-07-23:
@1172534699 Have you tried updating your system packages? Or a doing clean build?

closed 2023-12-01 by tylerjw