moveit/moveit2_tutorials #715: Moveit Python available for Ros2 Humble?

Icon45, 2023-07-08:
### Description

Hi, I wondered if I can use the python api for moveit2, which is described in Picknik Robotics Documentation, in ros2 humble. I have tried ist but can’t get it to work. Now I’m starting to think that it might not be implemented.

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* binary build
* If binary, which release version?: the one described here: https://moveit.ros.org/install-moveit2/binary/ 

### Steps to reproduce
Write a python node and include the lines ‚from moveit.planning import MoveItPy‘ and 
‚from moveit.core.robot_state import RobotState‘

build the python row two package that includes the node.
Run the node 

### Backtrace or Console output

you‘ll get: no module named ‚moveit‘ found

henningkayser, 2023-08-01:
The humble branch does not feature the python bindings. However, you can source install MoveIt 2 `main` with Humble without any issues.

henningkayser, 2023-08-01:
I'm closing this issue, we'll continue in #729 since other things are also affected.

henningkayser, 2023-08-01:
sorry, closed wrong issue

uriel-luviano, 2023-11-29:
Hi! Is there any way to incorporate the python bindings if I already installed the humble branch? (I need it because the package https://github.com/kroshu/kuka_drivers is not compatible with Iron).
Thanks in advance!

closed 2023-12-01 by tylerjw

uriel-luviano, 2023-12-11:
Hi! I tried building the main branch from a Humble installation but I get this error:
```
struct plan_execution::ExecutableTrajectory’ has no member named ‘description’; did you mean ‘description_
```
Am I doing something wrong?