moveit/moveit2_tutorials #719: Unable to build "moveit_task_constructor_capabilities" package.

chwu-rwth, 2023-07-24:
### Description

Unable to build the package: moveit_task_constructor_capabilities

### Your environment
* ROS Distro: [Humble]
* OS Version: Ubuntu 22.04
* Source: git clone https://github.com/ros-planning/moveit2_tutorials -b main --depth 1

### Steps to reproduce
Was following the instructions from the documentation page: https://moveit.picknik.ai/main/doc/tutorials/getting_started/getting_started.html
Up until : colcon build --mixin release

### Expected behaviour
Summary : X packages finished

### Console output

/home/student/ws_moveit/src/moveit_task_constructor/capabilities/src/execute_task_solution_capability.cpp: In member function ‘bool move_group::ExecuteTaskSolutionCapability::constructMotionPlan(const Solution&, plan_execution::ExecutableMotionPlan&)’:
/home/student/ws_moveit/src/moveit_task_constructor/capabilities/src/execute_task_solution_capability.cpp:145:14: error: ‘struct plan_execution::ExecutableMotionPlan’ has no member named ‘plan_components’; did you mean ‘plan_components_’?
  145 |         plan.plan_components.reserve(solution.sub_trajectory.size());
      |              
      |              plan_components_
/home/student/ws_moveit/src/moveit_task_constructor/capabilities/src/execute_task_solution_capability.cpp:149:22: error: ‘struct plan_execution::ExecutableMotionPlan’ has no member named ‘plan_components’; did you mean ‘plan_components_’?
  149 |                 plan.plan_components.emplace_back();
      |                      
      |                      plan_components_
/home/student/ws_moveit/src/moveit_task_constructor/capabilities/src/execute_task_solution_capability.cpp:150:72: error: ‘struct plan_execution::ExecutableMotionPlan’ has no member named ‘plan_components’; did you mean ‘plan_components_’?
  150 |      plan_execution::ExecutableTrajectory& exec_traj = plan.plan_components.back();
      |                                                             
      |                                                             plan_components_
/home/student/ws_moveit/src/moveit_task_constructor/capabilities/src/execute_task_solution_capability.cpp:154:27: error: ‘struct plan_execution::ExecutableTrajectory’ has no member named ‘description’; did you mean ‘description_’?
  154 |                 exec_traj.description = description;
      |                           
      |                           description_
/home/student/ws_moveit/src/moveit_task_constructor/capabilities/src/execute_task_solution_capability.cpp:171:27: error: ‘struct plan_execution::ExecutableTrajectory’ has no member named ‘trajectory’; did you mean ‘trajectory_’?
  171 |                 exec_traj.trajectory = std::make_shared<robot_trajectory::RobotTrajectory>(model, group);
      |                           
      |                           trajectory_
/home/student/ws_moveit/src/moveit_task_constructor/capabilities/src/execute_task_solution_capability.cpp:172:27: error: ‘struct plan_execution::ExecutableTrajectory’ has no member named ‘trajectory’; did you mean ‘trajectory_’?
  172 |                 exec_traj.trajectory->setRobotTrajectoryMsg(state, sub_traj.trajectory);
      |                           
      |                           trajectory_
/home/student/ws_moveit/src/moveit_task_constructor/capabilities/src/execute_task_solution_capability.cpp:175:27: error: ‘struct plan_execution::ExecutableTrajectory’ has no member named ‘effect_on_success’; did you mean ‘effect_on_success_’?
  175 |                 exec_traj.effect_on_success = [this, sub_traj,
      |                           
      |                           effect_on_success_
gmake[2]: *** [CMakeFiles/moveit_task_constructor_capabilities.dir/build.make:76: CMakeFiles/moveit_task_constructor_capabilities.dir/src/execute_task_solution_capability.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/moveit_task_constructor_capabilities.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
Failed   <<< moveit_task_constructor_capabilities [13.2s, exited with code 2]

FlorianHou, 2023-07-27:
Same issue. But there is no such problem under the ROS Iron version. I also feel very confused.

alphalambdaepsilonxi, 2023-07-29:
I fixed the issue in Humble doing exacetly what the compiler suggested to do. You just have to add _ at exactly the places where the compiler asks you to do that. You just need to modify one file the execute_task_solution_capabilities.cpp
Good luck!

FlorianHou, 2023-07-29:
> I fixed the issue in Humble doing exacetly what the compiler suggested to do. You just have to add _ at exactly the places where the compiler asks you to do that. You just need to modify one file the execute_task_solution_capabilities.cpp Good luck!

Thx~😊

sea-bass, 2023-07-30:
Should be fixed on the `humble` branch now thanks to https://github.com/ros-planning/moveit2_tutorials/pull/722

closed 2023-07-30 by sea-bass

anikk94, 2024-04-04:
When building with moveit2 tutorials, I got the same build errors, made the corrections @FlorianHou mentioned and moved on. I want to push these changes to the ros2 branch which is what is used in the tutorials (by vcs import). Are there any specific rules to making the pull request?
#719