moveit/moveit2_tutorials #723: Incorrect Place Poses Generated in MTC Tutorial

sea-bass, 2023-07-30:
### Description

I found an issue on Humble in the [MTC tutorial](https://moveit.picknik.ai/main/doc/tutorials/pick_and_place_with_moveit_task_constructor/pick_and_place_with_moveit_task_constructor.html), where the place pose generated is rotated and translated incorrectly even though we're just asking to translate in the Y direction by -0.5 meters.

### Your environment
* ROS Distro: [Humble]
* OS Version: Ubuntu 22.04
* Source build

### Steps to reproduce
If you look at the marker at the pick pose:
[image]

Then the place pose RViz marker is correct, but the target object pose and plan is incorrect.
[image]

Is this a bug with `GeneratePlacePose` stage that was introduced at some point?

sea-bass, 2023-07-30:
I can fix the issue and restore it to what the video looks like by changing [this line](https://github.com/ros-planning/moveit2_tutorials/blob/humble/doc/tutorials/pick_and_place_with_moveit_task_constructor/src/mtc_node.cpp#L300)

from

```
      wrapper->setIKFrame(hand_frame);
```

to

```
      wrapper->setIKFrame(grasp_frame_transform, hand_frame);
```

to match with the IK frame set in the grasp generation stage.

This meant elevating the scope of the `grasp_frame_transform` to higher up in the demo... is this the right fix? Would appreciate input from someone more MTC savvy.

rhaschke, 2023-07-31:
Actually, it should work to simply use the object frame:
`wrapper->setIKFrame("object")`

sea-bass, 2023-07-31:
Sure enough, that does work -- thank you! PR incoming.

closed 2023-08-08 by tylerjw