moveit/moveit2_tutorials #728: Python API documentation missing?

pinakjani, 2023-07-31:
### Description

The Moveit2 Python API documentation page is missing.

### Your environment
* ROS Distro: [Humble]
* OS Version: Ubuntu 22.04

### Steps to reproduce
I am working on the Moveit 2 Python API. The documentation page that I was referring until last week went missing today. Is there any update coming in with respect to the documentation page and if so is there another way to access the documentation in the meantime rather than referring the source code directly. 

Following is the link to the page that I was referring to: https://moveit.picknik.ai/main/doc/api/python_api/api.html

henningkayser, 2023-08-01:
Thanks for the heads-up, you are right it's missing. Apparently, the latest Humble deploy build has caused this.

henningkayser, 2023-08-01:
I'm closing this issue, we'll continue in https://github.com/ros-planning/moveit2_tutorials/issues/729 since other things are also affected.

closed 2023-08-01 by henningkayser