moveit/moveit2_tutorials #731: MoveIt2 and Omniverse Isaac Sim connection is not working (demo_isaac)

develiberta, 2023-08-04:
Hi. I want to link MoveIt2 and Isaac Sim in ROS2. I'm following the description on the site (https://docs.omniverse.nvidia.com/isaacsim/latest/tutorial_ros2_moveit.html) ). I am writing an inquiry because the connection is not working well.

**1. Environment**

Ubuntu 22.04, ROS2 Humble
Omniverse Isaac Sim 2022.2.1

**2. context**

As explained in the Omniverse guide, I installed ROS2 Humble and created a fastds.xml file to capture environmental variables. And I proceeded as explained in the MoveIt guide linked to the link. Omniverse runs normally, but when I run Rviz, it fails even if I move the robot, and the following log appears on the console.

**3. Log (Rviz)**

```
nvidia@ubuntu22:~/Omniverse/Moveit/moveit2_tutorials/doc/how_to_guides/isaac_panda$ docker compose up demo_isaac
[+] Running 1/0
 ✔ Container isaac_panda-demo_isaac-1  Created                                                                                           0.1s 
Attaching to isaac_panda-demo_isaac-1
isaac_panda-demo_isaac-1  | Sourced ROS 2 Humble
isaac_panda-demo_isaac-1  | Sourced isaac_moveit_tutorial workspace
isaac_panda-demo_isaac-1  | [INFO] [launch]: All log files can be found below /root/.ros/log/2023-08-04-01-40-55-929367-ubuntu22-1
isaac_panda-demo_isaac-1  | [INFO] [launch]: Default logging verbosity is set to INFO
isaac_panda-demo_isaac-1  | [INFO] [rviz2-1]: process started with pid [52]
isaac_panda-demo_isaac-1  | 
isaac_panda-demo_isaac-1  | [INFO] [static_transform_publisher-2]: process started with pid [54]
isaac_panda-demo_isaac-1  | [INFO] [static_transform_publisher-3]: process started with pid [56]
isaac_panda-demo_isaac-1  | [INFO] [robot_state_publisher-4]: process started with pid [58]
isaac_panda-demo_isaac-1  | [INFO] [move_group-5]: process started with pid [60]
isaac_panda-demo_isaac-1  | [INFO] [ros2_control_node-6]: process started with pid [62]
isaac_panda-demo_isaac-1  | [INFO] [spawner-7]: process started with pid [64]
isaac_panda-demo_isaac-1  | [INFO] [spawner-8]: process started with pid [66]
isaac_panda-demo_isaac-1  | [INFO] [spawner-9]: process started with pid [74]
isaac_panda-demo_isaac-1  | [static_transform_publisher-2] [WARN] [1691113256.465668657] []: Old-style arguments are deprecated; see --help for new-style arguments
isaac_panda-demo_isaac-1  | [static_transform_publisher-3] [WARN] [1691113256.472380100] []: Old-style arguments are deprecated; see --help for new-style arguments
isaac_panda-demo_isaac-1  | [robot_state_publisher-4] [WARN] [1691113256.497841557] [kdl_parser]: The root link panda_link0 has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
isaac_panda-demo_isaac-1  | 
isaac_panda-demo_isaac-1  | [robot_state_publisher-4] [INFO] [1691113256.498552971] [robot_state_publisher]: got segment panda_hand
isaac_panda-demo_isaac-1  | [robot_state_publisher-4] [INFO] [1691113256.498616269] [robot_state_publisher]: got segment panda_leftfinger
isaac_panda-demo_isaac-1  | [robot_state_publisher-4] [INFO] [1691113256.498626819] [robot_state_publisher]: got segment panda_link0
isaac_panda-demo_isaac-1  | [robot_state_publisher-4] [INFO] [1691113256.498631829] [robot_state_publisher]: got segment panda_link1
isaac_panda-demo_isaac-1  | [robot_state_publisher-4] [INFO] [1691113256.498636869] [robot_state_publisher]: got segment panda_link2
isaac_panda-demo_isaac-1  | [robot_state_publisher-4] [INFO] [1691113256.498641499] [robot_state_publisher]: got segment panda_link3
isaac_panda-demo_isaac-1  | [robot_state_publisher-4] [INFO] [1691113256.498646029] [robot_state_publisher]: got segment panda_link4
isaac_panda-demo_isaac-1  | [robot_state_publisher-4] [INFO] [1691113256.498650768] [robot_state_publisher]: got segment panda_link5
isaac_panda-demo_isaac-1  | [robot_state_publisher-4] [INFO] [1691113256.498655308] [robot_state_publisher]: got segment panda_link6
isaac_panda-demo_isaac-1  | [robot_state_publisher-4] [INFO] [1691113256.498659808] [robot_state_publisher]: got segment panda_link7
isaac_panda-demo_isaac-1  | [robot_state_publisher-4] [INFO] [1691113256.498664348] [robot_state_publisher]: got segment panda_link8
isaac_panda-demo_isaac-1  | [robot_state_publisher-4] [INFO] [1691113256.498668938] [robot_state_publisher]: got segment panda_rightfinger
isaac_panda-demo_isaac-1  | [static_transform_publisher-2] [INFO] [1691113256.502854511] [static_transform_publisher]: Spinning until stopped - publishing transform
isaac_panda-demo_isaac-1  | [static_transform_publisher-2] translation: ('0.000000', '0.000000', '0.000000')
isaac_panda-demo_isaac-1  | [static_transform_publisher-2] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
isaac_panda-demo_isaac-1  | [static_transform_publisher-2] from 'world' to 'panda_link0'
isaac_panda-demo_isaac-1  | [static_transform_publisher-3] [INFO] [1691113256.503739830] [static_transform_publisher]: Spinning until stopped - publishing transform
isaac_panda-demo_isaac-1  | [static_transform_publisher-3] translation: ('0.040000', '0.000000', '0.040000')
isaac_panda-demo_isaac-1  | [static_transform_publisher-3] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
isaac_panda-demo_isaac-1  | [static_transform_publisher-3] from 'panda_hand' to 'sim_camera'
isaac_panda-demo_isaac-1  | [ros2_control_node-6] [INFO] [1691113256.505609297] [resource_manager]: Loading hardware 'PandaFakeSystem' 
isaac_panda-demo_isaac-1  | [move_group-5] [INFO] [1691113256.525525713] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00299423 seconds
isaac_panda-demo_isaac-1  | [move_group-5] [INFO] [1691113256.525639420] [moveit_robot_model.robot_model]: Loading robot model 'panda'...
isaac_panda-demo_isaac-1  | [ros2_control_node-6] [INFO] [1691113256.539457578] [resource_manager]: Initialize hardware 'PandaFakeSystem' 
isaac_panda-demo_isaac-1  | [move_group-5] [WARN] [1691113256.540651141] [kdl_parser]: The root link panda_link0 has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
isaac_panda-demo_isaac-1  | [move_group-5] [INFO] [1691113256.540804967] [moveit_kinematics_base.kinematics_base]: Joint weights for group 'panda_arm': 1 1 1 1 1 1 1
isaac_panda-demo_isaac-1  | [rviz2-1] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
isaac_panda-demo_isaac-1  | [ros2_control_node-6] [INFO] [1691113256.564638372] [resource_manager]: Successful initialization of hardware 'PandaFakeSystem'
isaac_panda-demo_isaac-1  | [ros2_control_node-6] [INFO] [1691113256.564843427] [resource_manager]: Loading hardware 'PandaHandFakeSystem' 
isaac_panda-demo_isaac-1  | [ros2_control_node-6] [INFO] [1691113256.564871367] [resource_manager]: Initialize hardware 'PandaHandFakeSystem' 
isaac_panda-demo_isaac-1  | [move_group-5] [INFO] [1691113256.567011877] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
isaac_panda-demo_isaac-1  | [move_group-5] [INFO] [1691113256.567263241] [moveit.ros_planning_interface.moveit_cpp]: Listening to 'joint_states' for joint states
isaac_panda-demo_isaac-1  | [move_group-5] [INFO] [1691113256.568940852] [moveit_ros.current_state_monitor]: Listening to joint states on topic 'joint_states'
isaac_panda-demo_isaac-1  | [move_group-5] [INFO] [1691113256.569382552] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
[... 190 lines cut ...]
isaac_panda-demo_isaac-1  | [spawner-7] [ERROR] [1691113266.821315791] [spawner_joint_state_broadcaster]: Controller manager not available
isaac_panda-demo_isaac-1  | [spawner-9] [ERROR] [1691113266.868391874] [spawner_panda_hand_controller]: Controller manager not available
isaac_panda-demo_isaac-1  | [spawner-8] [ERROR] [1691113266.879465556] [spawner_panda_arm_controller]: Controller manager not available
isaac_panda-demo_isaac-1  | [ERROR] [spawner-7]: process has died [pid 64, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner joint_state_broadcaster --controller-manager /controller_manager --ros-args'].
isaac_panda-demo_isaac-1  | [ERROR] [spawner-9]: process has died [pid 74, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner panda_hand_controller -c /controller_manager --ros-args'].
[... cut ...] plugin does warn you if it does not use the requested planner.
isaac_panda-demo_isaac-1  | [move_group-5] [INFO] [1691113276.841061297] [moveit_move_group_default_capabilities.move_action_capability]: START_STATE_IN_COLLISION
isaac_panda-demo_isaac-1  | [rviz2-1] [INFO] [1691113276.841681052] [move_group_interface]: Plan and Execute request aborted
isaac_panda-demo_isaac-1  | [rviz2-1] [ERROR] [1691113276.842549752] [move_group_interface]: MoveGroupInterface::move() failed or timeout reached
```

Zhaogaog, 2023-08-07:
Maybe i have met the similar problem with u.Have you fixed this problem or is there any progress?

develiberta, 2023-08-21:
@Zhaogaog 
This issue has be corrected and verified on the main brach.
Can you please test again?

FYI when you rebuild you might need to specify no cache so it gets the latest changes.
docker compose build base --no-cache

closed 2023-08-21 by develiberta