moveit/moveit2_tutorials #732: Error while launching demo tutorial

Shobuj-Paul, 2023-08-05:
### Description
While launching the `demo.launch.py` node, I get an error saying that the package `kinova_gen3_7dof_robotiq_2f_85_moveit_config` is not found but trying to install it using `sudo apt install ros-rolling-kinova_gen3_7dof_robotiq_2f_85_moveit_config` gives the following error:
```bash
E: Unable to locate package ros-rolling-moveit-kinova_gen3_7dof_robotiq_2f_85_moveit_config
```

### Your environment
* ROS Distro: Rolling
* OS Version: Ubuntu 22.04
* Source Build
* Commit (f226955d549b4f5e650d17b96ba1174302d7adcd)

### Steps to reproduce
Follow the instructions on [this](https://moveit.picknik.ai/main/doc/tutorials/quickstart_in_rviz/quickstart_in_rviz_tutorial.html) page for getting started with Moveit.

### Expected behaviour
Should launch the demo as shown on the tutorials page.

### Backtrace or Console output

```bash
[INFO] [launch]: All log files can be found below /home/shobuj/.ros/log/2023-08-05-15-56-39-060518-home-station-30181
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): "package 'kinova_gen3_7dof_robotiq_2f_85_moveit_config' not found, searching: ['/home/shobuj/test_ws/install/moveit_tests', '/home/shobuj/moveit_ws/install/moveit2_tutorials', '/home/shobuj/moveit_ws/install/moveit_visual_tools', '/home/shobuj/moveit_ws/install/rviz_visual_tools', '/home/shobuj/moveit_ws/install/moveit_task_constructor_visualization', '/home/shobuj/moveit_ws/install/moveit_task_constructor_demo', '/home/shobuj/moveit_ws/install/moveit_task_constructor_capabilities', '/home/shobuj/moveit_ws/install/moveit_task_constructor_core', '/home/shobuj/moveit_ws/install/rviz_marker_tools', '/home/shobuj/moveit_ws/install/rosparam_shortcuts', '/home/shobuj/moveit_ws/install/pilz_industrial_motion_planner_testutils', '/home/shobuj/moveit_ws/install/moveit_runtime', '/home/shobuj/moveit_ws/install/moveit', '/home/shobuj/moveit_ws/install/moveit_planners', '/home/shobuj/moveit_ws/install/pilz_industrial_motion_planner', '/home/shobuj/moveit_ws/install/moveit_task_constructor_msgs', '/home/shobuj/moveit_ws/install/moveit_ros_control_interface', '/home/shobuj/moveit_ws/install/moveit_plugins', '/home/shobuj/moveit_ws/install/moveit_simple_controller_manager', '/home/shobuj/moveit_ws/install/moveit_setup_assistant', '/home/shobuj/moveit_ws/install/moveit_setup_srdf_plugins', '/home/shobuj/moveit_ws/install/moveit_setup_core_plugins', '/home/shobuj/moveit_ws/install/moveit_setup_controllers', '/home/shobuj/moveit_ws/install/moveit_setup_app_plugins', '/home/shobuj/moveit_ws/install/moveit_setup_framework', '/home/shobuj/moveit_ws/install/moveit_servo', '/home/shobuj/moveit_ws/install/moveit_ros', '/home/shobuj/moveit_ws/install/moveit_ros_visualization', '/home/shobuj/moveit_ws/install/moveit_py', '/home/shobuj/moveit_ws/install/moveit_hybrid_planning', '/home/shobuj/moveit_ws/install/moveit_ros_planning_interface', '/home/shobuj/moveit_ws/install/moveit_ros_benchmarks', '/home/shobuj/moveit_ws/install/moveit_ros_warehouse', '/home/shobuj/moveit_ws/install/moveit_ros_robot_interaction', '/home/shobuj/moveit_ws/install/moveit_ros_perception', '/home/shobuj/moveit_ws/install/moveit_resources_prbt_pg70_support', '/home/shobuj/moveit_ws/install/moveit_resources_prbt_moveit_config', '/home/shobuj/moveit_ws/install/moveit_ros_move_group', '/home/shobuj/moveit_ws/install/moveit_planners_ompl', '/home/shobuj/moveit_ws/install/moveit_kinematics', '/home/shobuj/moveit_ws/install/moveit_ros_planning', '/home/shobuj/moveit_ws/install/moveit_ros_occupancy_map_monitor', '/home/shobuj/moveit_ws/install/moveit_resources_prbt_support', '/home/shobuj/moveit_ws/install/moveit_resources_prbt_ikfast_manipulator_plugin', '/home/shobuj/moveit_ws/install/moveit_resources', '/home/shobuj/moveit_ws/install/moveit_resources_pr2_description', '/home/shobuj/moveit_ws/install/moveit_resources_panda_moveit_config', '/home/shobuj/moveit_ws/install/dual_arm_panda_moveit_config', '/home/shobuj/moveit_ws/install/moveit_resources_panda_description', '/home/shobuj/moveit_ws/install/moveit_resources_fanuc_moveit_config', '/home/shobuj/moveit_ws/install/moveit_resources_fanuc_description', '/home/shobuj/moveit_ws/install/moveit_planners_stomp', '/home/shobuj/moveit_ws/install/moveit_planners_chomp', '/home/shobuj/moveit_ws/install/moveit_chomp_optimizer_adapter', '/home/shobuj/moveit_ws/install/chomp_motion_planner', '/home/shobuj/moveit_ws/install/moveit_core', '/home/shobuj/moveit_ws/install/moveit_msgs', '/home/shobuj/moveit_ws/install/moveit_configs_utils', '/home/shobuj/moveit_ws/install/moveit_common', '/opt/ros/rolling']"
```

sea-bass, 2023-08-05:
You should replace all the underscores with dashes, so

```
sudo apt install ros-rolling-kinova_gen3-7dof-robotiq-2f-85-moveit-config
```

Does this work?

... but also, you can always use `rosdep install` on your workspace to have the binaries be automatically installed!

Shobuj-Paul, 2023-08-05:
Hi,
My bad, I had originally put dashes, I think I put underscores as a copying mistake, I will update the issue accordingly.
I used the following rosdep command to update my packages from my `~/moveit_ws/src` directory.
```bash
sudo apt update && rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y
```
This was the output:
```bash
executing command [sudo -H apt-get install -y ros-rolling-kinova-gen3-7dof-robotiq-2f-85-moveit-config]
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package ros-rolling-kinova-gen3-7dof-robotiq-2f-85-moveit-config
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install -y ros-rolling-kinova-gen3-7dof-robotiq-2f-85-moveit-config] failed
  apt: Failed to detect successful installation of [ros-rolling-kinova-gen3-7dof-robotiq-2f-85-moveit-config]
```

sea-bass, 2023-08-05:
Hmm, then likely the Kinova configs are still not available from binaries.

Can you clone the moveit_resources and build it from source, as shown here?

https://github.com/ros-planning/moveit2_tutorials/blob/main/moveit2_tutorials.repos#L22

Shobuj-Paul, 2023-08-05:
This is the stderr output from building the packages.
```bash
--- stderr: kortex_api                                                                                                                    
CMake Warning (dev) at /home/shobuj/.local/lib/python3.10/site-packages/cmake/data/share/cmake-3.26/Modules/FetchContent.cmake:1282 (message):
  The DOWNLOAD_EXTRACT_TIMESTAMP option was not given and policy CMP0135 is
  not set.  The policy's OLD behavior will be used.  When using a URL
  download, the timestamps of extracted files should preferably be that of
  the time of extraction, otherwise code that depends on the extracted
  contents might not be rebuilt if the URL changes.  The OLD behavior
  preserves the timestamps from the archive instead, but this is usually not
  what you want.  Update your project to the NEW behavior or specify the
  DOWNLOAD_EXTRACT_TIMESTAMP option with a value of true to avoid this
  robustness issue.
Call Stack (most recent call first):
  CMakeLists.txt:4 (FetchContent_Declare)
This warning is for project developers.  Use -Wno-dev to suppress it.

--- stderr: kortex_driver                                                                                        
CMake Warning (dev) at /home/shobuj/.local/lib/python3.10/site-packages/cmake/data/share/cmake-3.26/Modules/FetchContent.cmake:1282 (message):
  The DOWNLOAD_EXTRACT_TIMESTAMP option was not given and policy CMP0135 is
  not set.  The policy's OLD behavior will be used.  When using a URL
  download, the timestamps of extracted files should preferably be that of
  the time of extraction, otherwise code that depends on the extracted
  contents might not be rebuilt if the URL changes.  The OLD behavior
  preserves the timestamps from the archive instead, but this is usually not
  what you want.  Update your project to the NEW behavior or specify the
  DOWNLOAD_EXTRACT_TIMESTAMP option with a value of true to avoid this
  robustness issue.
Call Stack (most recent call first):
  CMakeLists.txt:3 (FetchContent_Declare)
This warning is for project developers.  Use -Wno-dev to suppress it.
```

This is the following output on trying to run the same launch command.

```bash
[INFO] [launch]: All log files can be found below /home/shobuj/.ros/log/2023-08-05-18-50-07-203075-home-station-55917
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): package not found: "package 'robotiq_description' not found, searching: ['/home/shobuj/moveit_ws/install/kinova_gen3_7dof_robotiq_2f_85_moveit_config', '/home/shobuj/moveit_ws/install/kinova_gen3_6dof_robotiq_2f_85_moveit_config', '/home/shobuj/moveit_ws/install/kortex_bringup', '/home/shobuj/moveit_ws/install/kortex_driver', '/home/shobuj/moveit_ws/install/kortex_description', '/home/shobuj/moveit_ws/install/kortex_api', '/home/shobuj/test_ws/install/moveit_tests', '/home/shobuj/moveit_ws/install/moveit2_tutorials', '/home/shobuj/moveit_ws/install/moveit_visual_tools', '/home/shobuj/moveit_ws/install/rviz_visual_tools', '/home/shobuj/moveit_ws/install/moveit_task_constructor_visualization', '/home/shobuj/moveit_ws/install/moveit_task_constructor_demo', '/home/shobuj/moveit_ws/install/moveit_task_constructor_capabilities', '/home/shobuj/moveit_ws/install/moveit_task_constructor_core', '/home/shobuj/moveit_ws/install/rviz_marker_tools', '/home/shobuj/moveit_ws/install/rosparam_shortcuts', '/home/shobuj/moveit_ws/install/pilz_industrial_motion_planner_testutils', '/home/shobuj/moveit_ws/install/moveit_runtime', '/home/shobuj/moveit_ws/install/moveit', '/home/shobuj/moveit_ws/install/moveit_planners', '/home/shobuj/moveit_ws/install/pilz_industrial_motion_planner', '/home/shobuj/moveit_ws/install/moveit_task_constructor_msgs', '/home/shobuj/moveit_ws/install/moveit_ros_control_interface', '/home/shobuj/moveit_ws/install/moveit_plugins', '/home/shobuj/moveit_ws/install/moveit_simple_controller_manager', '/home/shobuj/moveit_ws/install/moveit_setup_assistant', '/home/shobuj/moveit_ws/install/moveit_setup_srdf_plugins', '/home/shobuj/moveit_ws/install/moveit_setup_core_plugins', '/home/shobuj/moveit_ws/install/moveit_setup_controllers', '/home/shobuj/moveit_ws/install/moveit_setup_app_plugins', '/home/shobuj/moveit_ws/install/moveit_setup_framework', '/home/shobuj/moveit_ws/install/moveit_servo', '/home/shobuj/moveit_ws/install/moveit_ros', '/home/shobuj/moveit_ws/install/moveit_ros_visualization', '/home/shobuj/moveit_ws/install/moveit_py', '/home/shobuj/moveit_ws/install/moveit_hybrid_planning', '/home/shobuj/moveit_ws/install/moveit_ros_planning_interface', '/home/shobuj/moveit_ws/install/moveit_ros_benchmarks', '/home/shobuj/moveit_ws/install/moveit_ros_warehouse', '/home/shobuj/moveit_ws/install/moveit_ros_robot_interaction', '/home/shobuj/moveit_ws/install/moveit_ros_perception', '/home/shobuj/moveit_ws/install/moveit_resources_prbt_pg70_support', '/home/shobuj/moveit_ws/install/moveit_resources_prbt_moveit_config', '/home/shobuj/moveit_ws/install/moveit_ros_move_group', '/home/shobuj/moveit_ws/install/moveit_planners_ompl', '/home/shobuj/moveit_ws/install/moveit_kinematics', '/home/shobuj/moveit_ws/install/moveit_ros_planning', '/home/shobuj/moveit_ws/install/moveit_ros_occupancy_map_monitor', '/home/shobuj/moveit_ws/install/moveit_resources_prbt_support', '/home/shobuj/moveit_ws/install/moveit_resources_prbt_ikfast_manipulator_plugin', '/home/shobuj/moveit_ws/install/moveit_resources', '/home/shobuj/move [... cut ...] ws/install/moveit_resources_fanuc_description', '/home/shobuj/moveit_ws/install/moveit_planners_stomp', '/home/shobuj/moveit_ws/install/moveit_planners_chomp', '/home/shobuj/moveit_ws/install/moveit_chomp_optimizer_adapter', '/home/shobuj/moveit_ws/install/chomp_motion_planner', '/home/shobuj/moveit_ws/install/moveit_core', '/home/shobuj/moveit_ws/install/moveit_msgs', '/home/shobuj/moveit_ws/install/moveit_configs_utils', '/home/shobuj/moveit_ws/install/moveit_common', '/opt/ros/rolling']"
````

sea-bass, 2023-08-05:
Yes, that package has its own dependencies such as the robotiq package in that error message. Make sure those are installed as well.

Shobuj-Paul, 2023-08-05:
> Yes, that package has its own dependencies such as the robotiq package in that error message. Make sure those are installed as well.

Rest of the other dependencies are installed, could you point out the robotiq-description package? I could only find [this](https://github.com/ros-industrial/robotiq) which is only for old ROS 1 versions.

sea-bass, 2023-08-05:
Yep, this one!
https://github.com/Kinovarobotics/ros2_kortex/blob/main/ros2_kortex.repos#L20

Shobuj-Paul, 2023-08-06:
I keep running into this one error while building. `ros-rolling-serial-driver` is installed so I am not sure which _serial_ the error is referring to.

```bash
--- stderr: robotiq_driver                                                                    
CMake Error at CMakeLists.txt:15 (find_package):
  By not providing "Findserial.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "serial", but
  CMake did not find one.

  Could not find a package configuration file provided by "serial" with any
  of the following names:

    serialConfig.cmake
    serial-config.cmake

  Add the installation prefix of "serial" to CMAKE_PREFIX_PATH or set
  "serial_DIR" to a directory containing one of the above files.  If "serial"
  provides a separate development package or SDK, be sure it has been
  installed.
```

sea-bass, 2023-08-06:
That requires this repo: https://github.com/PickNikRobotics/ros2_robotiq_gripper/blob/main/ros2_robotiq_gripper.rolling.repos#L4

Shobuj-Paul, 2023-08-06:
> That requires this repo: https://github.com/PickNikRobotics/ros2_robotiq_gripper/blob/main/ros2_robotiq_gripper.rolling.repos#L4
> 
> Sorry, I know chasing down vcstool dependencies is not the most fun thing.

Thanks for helping me, the tutorial finally ran in RViz. Should I create a PR for updating moveit2_tutorials.repos with all the required repositories?

sea-bass, 2023-08-06:
@moriarty @abake48 Do these source repos need to be updated, or is there a short term to have binaries released for rolling?

abake48, 2023-08-06:
It should be `sudo apt install ros-rolling-kinova-gen3-7dof-robotiq-2f-85-moveit-config`

moriarty, 2023-08-07:
The binaries are on ros-testing, not ros main yet, they'll hit main whenever the next sync happens: https://discourse.ros.org/c/release/16

https://docs.ros.org/en/rolling/Installation/Testing.html

moriarty, 2023-08-07:
re: the missing serial dependency, you only need the robotiq_description package package not the robotiq_driver package.

The kinova arm communicates directly with the robotiq_gripper which is why you don't need the robotiq_driver.
The robotiq_description is still needed.

If you build from source you can add a COLCON_IGNORE in robotiq_driver to skip the robotiq_driver.
The binary of the robotiq_driver is skipped for now I need to make time to release the an updated serial package and then release the driver.
https://github.com/ros2-gbp/ros2_robotiq_gripper-release/blob/master/rolling.ignored

Shobuj-Paul, 2023-09-05:
Since this has been resolved and ros2_kortex has been added to .repos, I am closing this issue. Thanks for helping out!

closed 2023-09-05 by Shobuj-Paul