moveit/moveit2_tutorials #733: can not make the panda move when i click 'plan & execute'

Zhaogaog, 2023-08-05:
I’m using ubuntu22.04,ros2humble, and when I follow this tutorial([9. MoveIt 2 — isaacsim latest documentation](https://docs.omniverse.nvidia.com/isaacsim/latest/tutorial_ros2_moveit.html))(https://moveit.picknik.ai/main/doc/how_to_guides/isaac_panda/isaac_panda_tutorial.html), I have a couple problems:
First of all: “controller manager not available”,All I can ensure is that I have the controller manager installed. I can make sure that I have installed the controller manager. And the ‘process has died’ appears.
Secondly: in the rviz2 interface, there is a warning like 'requesting initial scene failed '.
Lastly, when I move the trackball and give control commands, the arm does not move when I click ‘plan & execute’ and I get this error in the terminal 'Movegroup action client/server not ready '.
Help ！！！！！！
[image]
[image]
[image]

bmaxdk, 2023-08-12:
Could you check if the planning scene topic is correctly configured in Rviz2? Make sure your robot_description parameter is loaded and available.

Zhaogaog, 2023-08-14:
Is there anything wrong with the 'panda_moveit_config.rviz' or the  'isaac_demo.launch.py'.Actually,i am new and i do not know how to check  if the planning scene topic is correctly configured in Rviz2?Have you run the demo successfully? Can anybody provide me with the correct 'panda_moveit_config.rviz' and the  'isaac_demo.launch.py'? Thank you!

tylerjw, 2023-12-01:
Closing this for now, please re-open if you are still experiencing this problem.