moveit/moveit2_tutorials #736: CI is failing with segfault on teardown of test

tylerjw, 2023-08-08:
### Description

```
process has died [pid 5764, exit code -6, cmd '/opt/ros/rolling/lib/rviz2/rviz2 -d /home/runner/work/moveit2_tutorials/moveit2_tutorials/.work/target_ws/install/moveit2_tutorials/share/moveit2_tutorials/launch/kinova_moveit_config_demo.rviz --ros-args -r __node:=rviz2 --params-file
```

Example Failure: https://github.com/ros-planning/moveit2_tutorials/actions/runs/5715858695/job/15486491621?pr=727

sea-bass, 2023-08-08:
Looking at the failure, its that there is a `/move_action` action server that is not available:
https://github.com/ros-planning/moveit2_tutorials/actions/runs/5715858695/job/15486491621?pr=727#step:6:844

tylerjw, 2023-08-08:
It seems wrong that it is segfaulting even if the action server doesn't exist. We should fix it so it prints an error and exits with an error code (maybe by throwing an exception) instead of segfaulting. We should also figure out why it is looking for an action server that isn't there.

sea-bass, 2023-08-08:
Yeah, I think I found why in the PR -- turns out we were launching the executable generated from `bringup_test.cpp` twice... lol

As for the segfault with the control node, adding a delay after receiving a response from the action server (and not tearing down immediately) got rid of it for me, at least locally.

tylerjw, 2023-08-08:
Can you submit a PR to add that delay?

sea-bass, 2023-08-08:
https://github.com/ros-planning/moveit2_tutorials/pull/740

closed 2023-08-08 by tylerjw