moveit/moveit2_tutorials #744: colcon build --mixin release Failure with moveit 2

adityasomic, 2023-08-10:
Hello everyone , 

I know this issue has been discussed many times on the forum but after following every issue and its solution on the github I am not able to build the packages. Also Instructions given on the official site of moveit [https://moveit.picknik.ai/main/doc/tutorials/getting_started/getting_started.html](url) is not working properly as after every attempt it showing me the same result. Can anyone please suggest me a perfect steps for installing and building the packages of the moveit2.  

Note I also followed this [https://moveit.ros.org/install-moveit2/source/](url)

System: Ubuntu 22.04
Ros2 : Humble

[image]

adityasomic, 2023-08-10:
Thank you for your reply I followed it and it gave me a message 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package ros-humble-object-reocgnition-msgs

Then again I erased the repository and tried using this [https://moveit.ros.org/install-moveit2/source/#foxy-galactic-humble---stable](url)
So I received this error 
[image]

sea-bass, 2023-08-10:
Have you run `sudo rosdep init; rosdep update; sudo apt-get update` before trying this?

adityasomic, 2023-08-14:
Hi @sea-bass , Sorry for the late reply , yes I tried these steps earlier and it did not work , instead of that I installed the neotic with move it one which is working good.

sea-bass, 2023-08-14:
Great, glad you were able to get this resolved!

closed 2023-08-14 by sea-bass