moveit/moveit2_tutorials #761: Failed to call service get_planning_scene

shaimiz007, 2023-09-03:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Humble
* OS Version: e.g. Ubuntu 22.04
* Source or Binary build?

1. running the launch file [mtc_demo.launch.py](https://github.com/ros-planning/moveit2_tutorials/blob/humble/doc/tutorials/pick_and_place_with_moveit_task_constructor/launch/mtc_demo.launch.py) on ros2-humbe produce the warning
2. [rviz2-1] [INFO] [1693754311.894494125] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Failed to call service get_planning_scene, have you launched move_group or called psm.providePlanningSceneService()?
3 It sticking the flow of the pick and place tutorial
4. A work around can be launching ros2 launch panda_moveit_config demo.launch.py instead 
[image]
T

Norm1lu, 2023-11-22:
i met the same question

lazuforever, 2024-07-18:
do you have an answer?