moveit/moveit2_tutorials #767: Found empty JointState message

ravijo, 2023-09-20:
### Description

I am following the [MoveIt tutorials](https://moveit.picknik.ai/main/doc/examples/motion_planning_pipeline/motion_planning_pipeline_tutorial.html) and have stumbled upon a problem. The command `ros2 launch moveit2_tutorials motion_planning_pipeline_tutorial.launch.py` reports `Found empty JointState message`

### Your environment
* ROS Distro: Iron
* OS Version: Ubuntu 22.04.3 LTS
* ROS Source or Binary build: Binary
* Moveit Source or Binary build: Source

### Steps to reproduce
```
$ ros2 launch moveit2_tutorials move_group.launch.py
$ ros2 launch moveit2_tutorials motion_planning_pipeline_tutorial.launch.py
```

### Console output
* <details>
  <summary>move_group.launch.py</summary>

  ```
  $ ros2 launch moveit2_tutorials move_group.launch.py
  [INFO] [launch]: All log files can be found below /home/user/.ros/log/2023-09-20-16-59-01-264998-dell-86229
  [INFO] [launch]: Default logging verbosity is set to INFO
  [INFO] [rviz2-1]: process started with pid [86232]
  [INFO] [static_transform_publisher-2]: process started with pid [86234]
  [INFO] [robot_state_publisher-3]: process started with pid [86236]
  [INFO] [move_group-4]: process started with pid [86238]
  [INFO] [ros2_control_node-5]: process started with pid [86246]
  [INFO] [ros2 run controller_manager spawner panda_arm_controller-6]: process started with pid [86268]
  [INFO] [ros2 run controller_manager spawner panda_hand_controller-7]: process started with pid [86271]
  [INFO] [ros2 run controller_manager spawner joint_state_broadcaster-8]: process started with pid [86274]
  [rviz2-1] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
  [static_transform_publisher-2] [INFO] [1695196742.346904891] [static_transform_publisher]: Spinning until stopped - publishing transform
  [static_transform_publisher-2] translation: ('0.000000', '0.000000', '0.000000')
  [static_transform_publisher-2] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
  [static_transform_publisher-2] from 'world' to 'panda_link0'
  [robot_state_publisher-3] [WARN] [1695196742.348367274] [kdl_parser]: The root link panda_link0 has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
  [robot_state_publisher-3] [INFO] [1695196742.348480487] [robot_state_publisher]: got segment panda_hand
  [robot_state_publisher-3] [INFO] [1695196742.348525490] [robot_state_publisher]: got segment panda_leftfinger
  [robot_state_publisher-3] [INFO] [1695196742.348544040] [robot_state_publisher]: got segment panda_link0
  [robot_state_publisher-3] [INFO] [1695196742.348561232] [robot_state_publisher]: got segment panda_link1
  [robot_state_publisher-3] [INFO] [1695196742.348577366] [robot_state_publisher]: got segment panda_link2
  [robot_state_publisher-3] [INFO] [1695196742.348593199] [robot_state_publisher]: got segment panda_link3
  [robot_state_publisher-3] [INFO] [1695196742.348609461] [robot_state_publisher]: got segment panda_link4
  [robot_state_publisher-3] [INFO] [1695196742.348625792] [robot_state_publisher]: got segment panda_link5
  [robot_state_publisher-3] [INFO] [1695196742.348641367] [robot_state_publisher]: got segment panda_link6
  [robot_state_publisher-3] [INFO] [1695196742.348657261] [robot_state_publisher]: got segment panda_link7
  [robot_state_publisher-3] [INFO] [1695196742.348673093] [robot_state_publisher]: got segment panda_link8
  [robot_state_publisher-3] [INFO] [1695196742.348689131] [robot_state_publisher]: got segment panda_rightfinger
  [move_group-4] [INFO] [1695196742.477195930] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.0264601 seconds
  [move_group-4] [INFO] [1695196742.477255395] [moveit_robot_model.robot_model]: Loading robot model 'panda'...
  [ros2_control_node-5] [WARN] [1695196742.481171246] [controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
  [ros2_control_node-5] [INFO] [1695196742.481637517] [resource_manager]: Loading hardware 'PandaFakeSystem' 
  [ros2_control_node-5] [INFO] [1695196742.482587596] [resource_manager]: Initialize hardware 'PandaFakeSystem' 
  [ros2_control_node-5] [WARN] [1695196742.482763580] [mock_generic_system]: Parsing of optional initial interface values failed or uses a deprecated format. Add initial values for every state interface in the ros2_control.xacro. For example: 
  [ros2_control_node-5] <state_interface name="velocity"> 
  [ros2_control_node-5]   <param name="initial_value">0.0</param> 
  [ros2_control_node-5] </state_interface>
  [ros2_control_node-5] [INFO] [1695196742.482833200] [resource_manager]: Successful initialization of hardware 'PandaFakeSystem'
  [ros2_control_node-5] [INFO] [1695196742.482965750] [resource_manager]: Loading hardware 'PandaHandFakeSystem' 
  [ros2_control_node-5] [INFO] [1695196742.482983184] [resource_manager]: Initialize hardware 'PandaHandFakeSystem' 
  [ros2_control_node-5] [INFO] [1695196742.483025632] [resource_manager]: Successful initialization of hardware 'PandaHandFakeSystem'
  [ros2_control_node-5] [INFO] [1695196742.483244107] [resource_manager]: 'configure' hardware 'PandaHandFakeSystem' 
  [ros2_control_node-5] [INFO] [1695196742.483278891] [resource_manager]: Successful 'configure' of hardware 'PandaHandFakeSystem'
  [ros2_control_node-5] [INFO] [1695196742.483315547] [resource_manager]: 'activate' hardware 'PandaHandFakeSystem' 
  [ros2_control_node-5] [INFO] [1695196742.483348149] [resource_manager]: Successful 'activate' of hardware 'PandaHandFakeSystem'
  [ros2_control_node-5] [INFO] [1695196742.483385011] [resource_manager]: 'configure' hardware 'PandaFakeSystem' 
  [ros2_control_node-5] [INFO] [1695196742.483417771] [resource_manager]: Successful 'configure' of hardware 'PandaFakeSystem'
  [ros2_control_node-5] [INFO] [1695196742.483456766] [resource_manager]: 'activate' hardware 'PandaFakeSystem' 
[... 249 lines cut ...]
  - 1.571
  - 0.785
  velocity:
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  effort:
  - .nan
  - .nan
  - .nan
  - .nan
  - .nan
  - .nan
  - .nan
  - .nan
  - .nan
  ---
  ```
  </details>

ravijo, 2023-10-03:
Update

It works on ROS Humble upto some extent. Below are the logs:

```
$ ros2 launch moveit2_tutorials motion_planning_api_tutorial.launch.py
[INFO] [launch]: All log files can be found below /home/user/.ros/log/2023-10-03-06-10-04-802015-dell-2523
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [motion_planning_api_tutorial-1]: process started with pid [2530]
[motion_planning_api_tutorial-1] [INFO] [1696313405.093249169] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00431912 seconds
[motion_planning_api_tutorial-1] [INFO] [1696313405.093315649] [moveit_robot_model.robot_model]: Loading robot model 'panda'...
[motion_planning_api_tutorial-1] [INFO] [1696313405.143881242] [motion_planning_api_tutorial]: Using planning interface 'OMPL'
[motion_planning_api_tutorial-1] [INFO] [1696313405.146246596] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00213542 seconds
[motion_planning_api_tutorial-1] [INFO] [1696313405.146264625] [moveit_robot_model.robot_model]: Loading robot model 'panda'...
[motion_planning_api_tutorial-1] [INFO] [1696313405.174273099] [move_group_interface]: Ready to take commands for planning group panda_arm.
[motion_planning_api_tutorial-1] [INFO] [1696313405.175977917] [motion_planning_api_tutorial.remote_control]: RemoteControl Ready.
[motion_planning_api_tutorial-1] [INFO] [1696313405.176075675] [motion_planning_api_tutorial.remote_control]: Waiting to continue: Press 'next' in the RvizVisualToolsGui window to start the demo
[motion_planning_api_tutorial-1] [INFO] [1696313408.399002217] [motion_planning_api_tutorial.remote_control]: ... continuing
[motion_planning_api_tutorial-1] [ERROR] [1696313408.399314911] [moveit_robot_state.conversions]: Found empty JointState message
[motion_planning_api_tutorial-1] [INFO] [1696313408.399331515] [moveit_planning_interface.planning_interface]: The timeout for planning must be positive (0.000000 specified). Assuming one second instead.
[motion_planning_api_tutorial-1] [WARN] [1696313408.399346802] [moveit.ompl_planning.model_based_planning_context]: It looks like the planning volume was not specified.
[motion_planning_api_tutorial-1] [INFO] [1696313408.399620870] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'panda_arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[motion_planning_api_tutorial-1] [INFO] [1696313408.462488180] [motion_planning_api_tutorial.remote_control]: Waiting to continue: Press 'next' in the RvizVisualToolsGui window to continue the demo
[motion_planning_api_tutorial-1] [INFO] [1696313411.842841182] [motion_planning_api_tutorial.remote_control]: ... continuing
[motion_planning_api_tutorial-1] [ERROR] [1696313411.842930671] [moveit_robot_state.conversions]: Found empty JointState message
[motion_planning_api_tutorial-1] [INFO] [1696313411.842944733] [moveit_planning_interface.planning_interface]: The timeout for planning must be positive (0.000000 specified). Assuming one second instead.
[motion_planning_api_tutorial-1] [WARN] [1696313411.842955330] [moveit.ompl_planning.model_based_planning_context]: It looks like the planning volume was not specified.
[motion_planning_api_tutorial-1] [INFO] [1696313411.843088253] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'panda_arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[motion_planning_api_tutorial-1] [INFO] [1696313411.883744284] [motion_planning_api_tutorial.remote_control]: Waiting to continue: Press 'next' in the RvizVisualToolsGui window to continue the demo
[motion_planning_api_tutorial-1] [INFO] [1696313415.204878067] [motion_planning_api_tutorial.remote_control]: ... continuing
[motion_planning_api_tutorial-1] [ERROR] [1696313415.205031333] [moveit_robot_state.conversions]: Found empty JointState message
[motion_planning_api_tutorial-1] [INFO] [1696313415.205064965] [moveit_planning_interface.planning_interface]: The timeout for planning must be positive (0.000000 specified). Assuming one second instead.
[motion_planning_api_tutorial-1] [WARN] [1696313415.205077733] [moveit.ompl_planning.model_based_planning_context]: It looks like the planning volume was not specified.
[motion_planning_api_tutorial-1] [INFO] [1696313415.205233270] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'panda_arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[motion_planning_api_tutorial-1] [INFO] [1696313415.269014417] [motion_planning_api_tutorial.remote_control]: Waiting to continue: Press 'next' in the RvizVisualToolsGui window to continue the demo
[motion_planning_api_tutorial-1] [INFO] [1696313420.501440511] [motion_planning_api_tutorial.remote_control]: ... continuing
[motion_planning_api_tutorial-1] [ERROR] [1696313420.501666876] [moveit_robot_state.conversions]: Found empty JointState message
[motion_planning_api_tutorial-1] [INFO] [1696313420.501679984] [moveit_planning_interface.planning_interface]: The timeout for planning must be positive (0.000000 specified). Assuming one second instead.
[motion_planning_api_tutorial-1] [INFO] [1696313420.501856293] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'panda_arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when [... cut ...]
[motion_planning_api_tutorial-1] [INFO] [1696313603.897592122] [moveit.ompl_planning.model_based_planning_context]: panda_arm: Allocating specialized state sampler for state space
[motion_planning_api_tutorial-1] [INFO] [1696313604.020316735] [motion_planning_api_tutorial.remote_control]: Waiting to continue: Press 'next' in the RvizVisualToolsGui window to exit the demo
[motion_planning_api_tutorial-1] [INFO] [1696313719.011950316] [motion_planning_api_tutorial.remote_control]: ... continuing
[INFO] [motion_planning_api_tutorial-1]: process has finished cleanly [pid 2530]
```

tylerjw, 2023-12-01:
We were not able to reproduce.

closed 2023-12-01 by tylerjw