moveit/moveit2_tutorials #770: Package 'panda_moveit_config' not found in STOMP Motion Planner

ravijo, 2023-09-25:
### Description

I am getting started with the STOMP motion planner by following [the documentation](https://moveit.picknik.ai/main/doc/how_to_guides/stomp_planner/stomp_planner.html). But the command `ros2 launch panda_moveit_config demo.launch.py` mentioned in [the documentation](https://moveit.picknik.ai/main/doc/how_to_guides/stomp_planner/stomp_planner.html#running-the-demo) is showing the following error.
```
Package 'panda_moveit_config' not found
```

### Your environment
* ROS Distro: Humble and Iron (found in both versions)
* OS Version: Ubuntu 22.04.3 LTS
* ROS Source or Binary build: Binary
* MoveIt Source or Binary build: Source (`humble` branch for Humble, `main` branch for Iron)

### Steps to reproduce
* ROS Humble
  * Executing `demo.launch.py` as shown below:
    ```
    user@dell:~/moveit_ws_humble$ ros2 launch panda_moveit_config demo.launch.py
    Package 'panda_moveit_config' not found: "package 'panda_moveit_config' not found, searching: ['/home/user/moveit_ws_humble/install/pilz_industrial_motion_planner_testutils', '/home/user/moveit_ws_humble/install/moveit_runtime', '/home/user/moveit_ws_humble/install/moveit2_tutorials', '/home/user/moveit_ws_humble/install/moveit', '/home/user/moveit_ws_humble/install/moveit_planners', '/home/user/moveit_ws_humble/install/pilz_industrial_motion_planner', '/home/user/moveit_ws_humble/install/moveit_visual_tools', '/home/user/moveit_ws_humble/install/moveit_task_constructor_visualization', '/home/user/moveit_ws_humble/install/moveit_task_constructor_demo', '/home/user/moveit_ws_humble/install/moveit_task_constructor_capabilities', '/home/user/moveit_ws_humble/install/moveit_task_constructor_core', '/home/user/moveit_ws_humble/install/moveit_ros_control_interface', '/home/user/moveit_ws_humble/install/moveit_plugins', '/home/user/moveit_ws_humble/install/moveit_simple_controller_manager', '/home/user/moveit_ws_humble/install/moveit_setup_assistant', '/home/user/moveit_ws_humble/install/moveit_setup_srdf_plugins', '/home/user/moveit_ws_humble/install/moveit_setup_core_plugins', '/home/user/moveit_ws_humble/install/moveit_setup_controllers', '/home/user/moveit_ws_humble/install/moveit_setup_app_plugins', '/home/user/moveit_ws_humble/install/moveit_setup_framework', '/home/user/moveit_ws_humble/install/moveit_servo', '/home/user/moveit_ws_humble/install/moveit_ros', '/home/user/moveit_ws_humble/install/moveit_ros_visualization', '/home/user/moveit_ws_humble/install/moveit_hybrid_planning', '/home/user/moveit_ws_humble/install/moveit_ros_planning_interface', '/home/user/moveit_ws_humble/install/moveit_ros_benchmarks', '/home/user/moveit_ws_humble/install/moveit_ros_warehouse', '/home/user/moveit_ws_humble/install/moveit_ros_robot_interaction', '/home/user/moveit_ws_humble/install/moveit_ros_perception', '/home/user/moveit_ws_humble/install/moveit_resources_prbt_pg70_support', '/home/user/moveit_ws_humble/install/moveit_resources_prbt_moveit_config', '/home/user/moveit_ws_humble/install/moveit_ros_move_group', '/home/user/moveit_ws_humble/install/moveit_planners_ompl', '/home/user/moveit_ws_humble/install/moveit_ros_planning', '/home/user/moveit_ws_humble/install/moveit_ros_occupancy_map_monitor', '/home/user/moveit_ws_humble/install/moveit_resources_prbt_ikfast_manipulator_plugin', '/home/user/moveit_ws_humble/install/moveit_planners_chomp', '/home/user/moveit_ws_humble/install/moveit_kinematics', '/home/user/moveit_ws_humble/install/moveit_chomp_optimizer_adapter', '/home/user/moveit_ws_humble/install/chomp_motion_planner', '/home/user/moveit_ws_humble/install/moveit_core', '/home/user/moveit_ws_humble/install/moveit_configs_utils', '/home/user/moveit_ws_humble/install/srdfdom', '/home/user/moveit_ws_humble/install/rviz_marker_tools', '/home/user/moveit_ws_humble/install/rosparam_shortcuts', '/home/user/moveit_ws_humble/install/moveit_task_constructor_msgs', '/home/user/moveit_ws_humble/install/moveit_resources_prbt_support', '/home/user/moveit_ws_humble/install/moveit_resources', '/home/user/moveit_ws_humble/install/moveit_resources_pr2_description', '/home/user/moveit_ws_humble/install/moveit_resources_panda_moveit_config', '/home/user/moveit_ws_humble/install/moveit_resources_panda_description', '/home/user/moveit_ws_humble/install/moveit_resources_fanuc_moveit_config', '/home/user/moveit_ws_humble/install/moveit_resources_fanuc_description', '/home/user/moveit_ws_humble/install/moveit_common', '/home/user/moveit_ws_humble/install/launch_param_builder', '/opt/ros/humble']"
    ```
  * Surprisingly the package `panda_moveit_config` is available in the source folder as shown below:
    ```
    user@dell:~/moveit_ws_humble$ ll ./src/moveit_resources/panda_moveit_config
    total 40
    drwxrwxr-x  4 user user 4096  9月 25 11:08 ./
    drwxrwxr-x 10 user user 4096  9月 25 11:08 ../
    -rw-rw-r--  1 user user 4510  9月 25 11:08 CHANGELOG.rst
    -rw-rw-r--  1 user user  324  9月 25 11:08 CMakeLists.txt
    drwxrwxr-x  2 user user 4096  9月 25 11:08 config/
    drwxrwxr-x  2 user user 4096  9月 25 11:08 launch/
    -rw-rw-r--  1 user user 1741  9月 25 11:08 package.xml
    -rw-rw-r--  1 user user  199  9月 25 11:08 README.md
    -rw-rw-r--  1 user user  292  9月 25 11:08 .setup_assistant
    ```
  * The package `panda_moveit_config` is also installed by binary as shown below:
    ```
    user@dell:~/moveit_ws_humble$ sudo apt install ros-humble-moveit-resources-panda-moveit-config 
    [sudo] password for user: 
    Reading package lists... Done
    Building dependency tree... Done
    Reading state information... Done
    ros-humble-moveit-resources-panda-moveit-config is already the newest version (2.0.6-1jammy.20230822.190940).
    0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
    ```
* ROS Iron
  * Executing `demo.launch.py` as shown below:
    ```
    user@dell:~/moveit_ws_iron$ ros2 launch panda_moveit_config demo.launch.py
    Package 'panda_moveit_config' not found: "package 'panda_moveit_config' not found, searching: ['/home/user/moveit_ws_iron/install/moveit2_tutorials', '/home/user/moveit_ws_iron/install/hello_moveit', '/home/user/moveit_ws_iron/install/moveit_visual_tools', '/home/user/moveit_ws_iron/install/rviz_visual_tools', '/home/user/moveit_ws_iron/install/mtc_tutorial', '/home/user/moveit_ws_iron/install/moveit_task_constructor_visualization', '/home/user/moveit_ws_iron/install/moveit_task_constructor_demo', '/home/user/moveit_ws_iron/install/moveit_task_constructor_capabilities', '/home/user/moveit_ws_iron/install/moveit_task_constructor_core', '/home/user/moveit_ws_iron/install/rviz_marker_tools', '/home/user/moveit_ws_iron/install/rosparam_shortcuts', '/home/user/moveit_ws_iron/install/pilz_industrial_motion_planner_testutils', '/home/user/moveit_ws_iron/install/moveit_runtime', '/home/user/moveit_ws_iron/install/moveit', '/home/user/moveit_ws_iron/install/kinova_gen3_7dof_robotiq_2f_85_moveit_config', '/home/user/moveit_ws_iron/install/kinova_gen3_6dof_robotiq_2f_85_moveit_config', '/home/user/moveit_ws_iron/install/moveit_planners', '/home/user/moveit_ws_iron/install/pilz_industrial_motion_planner', '/home/user/moveit_ws_iron/install/pick_ik', '/home/user/moveit_ws_iron/install/moveit_task_constructor_msgs', '/home/user/moveit_ws_iron/install/moveit_ros_control_interface', '/home/user/moveit_ws_iron/install/moveit_plugins', '/home/user/moveit_ws_iron/install/moveit_simple_controller_manager', '/home/user/moveit_ws_iron/install/moveit_setup_assistant', '/home/user/moveit_ws_iron/install/moveit_setup_srdf_plugins', '/home/user/moveit_ws_iron/install/moveit_setup_core_plugins', '/home/user/moveit_ws_iron/install/moveit_setup_controllers', '/home/user/moveit_ws_iron/install/moveit_setup_app_plugins', '/home/user/moveit_ws_iron/install/moveit_setup_framework', '/home/user/moveit_ws_iron/install/moveit_servo', '/home/user/moveit_ws_iron/install/moveit_ros', '/home/user/moveit_ws_iron/install/moveit_ros_visualization', '/home/user/moveit_ws_iron/install/moveit_py', '/home/user/moveit_ws_iron/install/moveit_hybrid_planning', '/home/user/moveit_ws_iron/install/moveit_ros_planning_interface', '/home/user/moveit_ws_iron/install/moveit_ros_benchmarks', '/home/user/moveit_ws_iron/install/moveit_ros_warehouse', '/home/user/moveit_ws_iron/install/moveit_ros_robot_interaction', '/home/user/moveit_ws_iron/install/moveit_ros_perception', '/home/user/moveit_ws_iron/install/moveit_resources_prbt_pg70_support', '/home/user/moveit_ws_iron/install/moveit_resources_prbt_moveit_ [... cut ...]
  * The package `panda_moveit_config` is also installed by binary as shown below:
    ```
    user@dell:~/moveit_ws_iron$ sudo apt install ros-iron-moveit-resources-panda-moveit-config 
    [sudo] password for user: 
    Reading package lists... Done
    Building dependency tree... Done
    Reading state information... Done
    ros-iron-moveit-resources-panda-moveit-config is already the newest version (2.1.1-1jammy.20230908.192824).
    0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
    ```

sjahr, 2023-09-25:
Oh, that seems to be a bug. The package name should be `moveit_resources_panda_moveit_config`. Can you try running

```bash
ros2 launch moveit_resources_panda_moveit_config demo.launch.py
```

ravijo, 2023-09-25:
> Can you try running
> 
> ```shell
> ros2 launch moveit_resources_panda_moveit_config demo.launch.py
> ```

It works in both Humble and Iron. 

Please note that I had to add "Motion Planning" inside RViz by myself. After adding it to the display, I could see STOMP, but only in Iron. In the case of Humble, I could not see STOMP in the dropdown menu. Please see a screenshot of RViz shown in Iron below.

[image]

However, I still don't get the view of having a box in the scene, as shown in [the documentation](https://moveit.picknik.ai/main/doc/how_to_guides/stomp_planner/stomp_planner.html#stomp-motion-planner). I apologize if I am missing something.

closed 2023-09-25 by sjahr

sjahr, 2023-09-25:
Yes, STOMP is not available in humble installed as Debian. You need to build moveit main on humble from source if you want to use STOMP there. We usually only backport bug fixes after a version is released to keep the API stable (see [this blogpost](https://moveit.ros.org/moveit%202/ros/2023/05/31/balancing-stability-and-development.html) for a more detailed explanation). However, since the STOMP ROS 2 port is unlikely to break the API you could consider back porting it yourself and contribute this back to upstream moveit2. We'd appreciate this contribution :smiley:

ravijo, 2023-09-26:
@sjahr 

Thank you so much for sharing a link to the MoveIt blog post. I didn't know such an excellent blog existed! Bookmarking the URL now :)

BTW [the STOMP documentation](https://moveit.picknik.ai/main/doc/how_to_guides/stomp_planner/stomp_planner.html#stomp-motion-planner) shows a collision object (green box) and iterative trajectories for the arm. The command `ros2 launch moveit_resources_panda_moveit_config demo.launch.py` gets me to a different view. May I request you to provide your guidance to achieve it, please?

PR #771 (2023-09-25): Replace panda_moveit_config -> moveit_resources_panda_moveit_config
### Description

Fixes #770 

### Checklist
- [ ] **Required by CI**: Code is auto formatted using [clang-format](http://moveit.ros.org/documentation/contributing/code)
- [ ] While waiting for someone to review your request, please consider reviewing [another open pull request](https://github.com/ros-planning/moveit2/pulls) to support the maintainers

[//]: # "You can expect a response from a maintainer within 7 days. If you haven't heard anything by then, feel free to ping the thread. Thank you!"