moveit/moveit2_tutorials #784: motion_planning_python_api_tutorial cannot be executed correctly.

yulat214, 2023-10-06:
### Description

When executing motion_planning_python_api_tutorial.launch.py from moveit2_tutorial, Rviz shows `NO PLANNING LIBRARY`.
Also, Plan&Execute is not executable.
When executing, I receive the INFO message: `Failed to call service get_planning_scene, have you launched move_group or called psm.providePlanningSceneService()?` and the process stops.

I tried #724 , but it didn't work.

### Your environment
* ROS Distro: [Humble|Rolling]
* OS Version: Ubuntu 22.04
* Source or Binary build?: source
* If source, which git commit or tag?: main

### Steps to reproduce
Following [this tutorial](https://moveit.picknik.ai/main/doc/examples/motion_planning_python_api/motion_planning_python_api_tutorial.html).

### Backtrace or Console output
I am concerned that planning scene monitor and others are stopping in the middle of the process.

<details><summary>full text</summary>
[INFO] [launch]: All log files can be found below /root/.ros/log/2023-10-06-04-40-55-535376-reclot-280
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [motion_planning_python_api_tutorial.py-1]: process started with pid [283]
[INFO] [robot_state_publisher-2]: process started with pid [285]
[INFO] [ros2_control_node-3]: process started with pid [287]
[INFO] [rviz2-4]: process started with pid [289]
[INFO] [static_transform_publisher-5]: process started with pid [291]
[INFO] [ros2 run controller_manager spawner panda_arm_controller-6]: process started with pid [293]
[INFO] [ros2 run controller_manager spawner panda_hand_controller-7]: process started with pid [295]
[INFO] [ros2 run controller_manager spawner joint_state_broadcaster-8]: process started with pid [298]
[static_transform_publisher-5] [INFO] [1696567255.794767540] [static_transform_publisher]: Spinning until stopped - publishing transform
[static_transform_publisher-5] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-5] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-5] from 'world' to 'panda_link0'
[ros2_control_node-3] [WARN] [1696567255.810406263] [controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ros2_control_node-3] [INFO] [1696567255.811235566] [resource_manager]: Loading hardware 'PandaFakeSystem'
[ros2_control_node-3] [INFO] [1696567255.811889504] [resource_manager]: Initialize hardware 'PandaFakeSystem'
[ros2_control_node-3] [WARN] [1696567255.811923472] [mock_generic_system]: Parsing of optional initial interface values failed or uses a deprecated format. Add initial values for every state interface in the ros2_control.xacro. For example:
[ros2_control_node-3] <state_interface name="velocity">
[ros2_control_node-3]   <param name="initial_value">0.0</param>
[ros2_control_node-3] </state_interface>
[ros2_control_node-3] [INFO] [1696567255.811934021] [resource_manager]: Successful initialization of hardware 'PandaFakeSystem'
[ros2_control_node-3] [INFO] [1696567255.811953861] [resource_manager]: Loading hardware 'PandaHandFakeSystem'
[ros2_control_node-3] [INFO] [1696567255.811963012] [resource_manager]: Initialize hardware 'PandaHandFakeSystem'
[ros2_control_node-3] [INFO] [1696567255.811972934] [resource_manager]: Successful initialization of hardware 'PandaHandFakeSystem'
[ros2_control_node-3] [INFO] [1696567255.812008182] [resource_manager]: 'configure' hardware 'PandaHandFakeSystem'
[ros2_control_node-3] [INFO] [1696567255.812013258] [resource_manager]: Successful 'configure' of hardware 'PandaHandFakeSystem'
[ros2_control_node-3] [INFO] [1696567255.812017885] [resource_manager]: 'activate' hardware 'PandaHandFakeSystem'
[ros2_control_node-3] [INFO] [1696567255.812021586] [resource_manager]: Successful 'activate' of hardware 'PandaHandFakeSystem'
[ros2_control_node-3] [INFO] [1696567255.812025066] [resource_manager]: 'configure' hardware 'PandaFakeSystem'
[ros2_control_node-3] [INFO] [1696567255.812027974] [resource_manager]: Successful 'configure' of hardware 'PandaFakeSystem'
[ros2_control_node-3] [INFO] [1696567255.812033238] [resource_manager]: 'activate' hardware 'PandaFakeSystem'
[ros2_control_node-3] [INFO] [1696567255.812036349] [resource_manager]: Successful 'activate' of hardware 'PandaFakeSystem'
[robot_state_publisher-2] [WARN] [1696567255.814573442] [kdl_parser]: The root link panda_link0 has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[robot_state_publisher-2] [INFO] [1696567255.815098229] [robot_state_publisher]: Robot initialized
[ros2_control_node-3] [INFO] [1696567255.815686658] [controller_manager]: update rate is 100 Hz
[ros2_control_node-3] [INFO] [1696567255.815803861] [controller_manager]: RT kernel is recommended for better performance
[rviz2-4] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
[motion_planning_python_api_tutorial.py-1] [INFO] [1696567255.885074041] [moveit_cpp_initializer]: Initialize rclcpp
[motion_planning_python_api_tutorial.py-1] [INFO] [1696567255.885883945] [moveit_cpp_initializer]: Initialize node parameters
[motion_planning_python_api_tutorial.py-1] [INFO] [1696567255.885911109] [moveit_cpp_initializer]: Initialize node and executor
[motion_planning_python_api_tutorial.py-1] [INFO] [1696567255.896240461] [moveit_cpp_initializer]: Spin separate thread
[motion_planning_python_api_tutorial.py-1] [INFO] [1696567255.898770578] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00243386 seconds
[motion_planning_python_api_tutorial.py-1] [INFO] [1696567255.898793938] [moveit_robot_model.robot_model]: Loading robot model 'panda'...
[motion_planning_python_api_tutorial.py-1] [WARN] [1696567255.909931764] [kdl_parser]: The root link panda_link0 has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[motion_planning_python_api_tutorial.py-1] [INFO] [1696567255.910037951] [moveit_kinematics_base.kinematics_base]: Joint weights for group 'panda_arm': 1 1 1 1 1 1 1
[motion_planning_python_api_tutorial.py-1] [INFO] [1696567255.931855557] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[motion_planning_python_api_tutorial.py-1] [INFO] [1696567255.931943673] [moveit.ros_planning_interface.moveit_cpp]: Listening to '/joint_states' for joint states
[motion_planning_python_api_tutorial.py-1] [INFO] [1696567255.932672073] [moveit_ros.current_state_monitor]: Listening to joint states on topic '/joint_states'
[... 48 lines cut ...]
[ros2_control_node-3] [INFO] [1696567256.257455784] [panda_hand_controller]: Action status changes will be monitored at 20Hz.
[ros2 run controller_manager spawner panda_hand_controller-7] [INFO] [1696567256.276775163] [spawner_panda_hand_controller]: Configured and activated panda_hand_controller
[ros2_control_node-3] [INFO] [1696567256.277717416] [controller_manager]: Loading controller 'panda_arm_controller'
[ros2_control_node-3] [WARN] [1696567256.285325571] [panda_arm_controller]: [Deprecated]: "allow_nonzero_velocity_at_trajectory_end" is set to true. The default behavior will change to false.
[ros2_control_node-3] [INFO] [1696567256.286266215] [controller_manager]: Loading controller 'joint_state_broadcaster'
[ros2 run controller_manager spawner panda_arm_controller-6] [INFO] [1696567256.287005537] [spawner_panda_arm_controller]: Loaded panda_arm_controller
[ros2_control_node-3] [INFO] [1696567256.296183667] [controller_manager]: Configuring controller 'panda_arm_controller'
[ros2_control_node-3] [INFO] [1696567256.296307097] [panda_arm_controller]: No specific joint names are used for command interfaces. Using 'joints' parameter.
[ros2_control_node-3] [INFO] [1696567256.296343010] [panda_arm_controller]: Command interfaces are [position] and state interfaces are [position velocity].
[ros2_control_node-3] [INFO] [1696567256.296359398] [panda_arm_controller]: Using 'splines' interpolation method.
[ros2 run controller_manager spawner joint_state_broadcaster-8] [INFO] [1696567256.296775658] [spawner_joint_state_broadcaster]: Loaded joint_state_broadcaster
[ros2_control_node-3] [INFO] [1696567256.297415221] [panda_arm_controller]: Action status changes will be monitored at 20.00 Hz.
[... cut ...]
[INFO] [ros2 run controller_manager spawner panda_arm_controller-6]: process has finished cleanly [pid 293]
[INFO] [ros2 run controller_manager spawner panda_hand_controller-7]: process has finished cleanly [pid 295]
[INFO] [ros2 run controller_manager spawner joint_state_broadcaster-8]: process has finished cleanly [pid 298]
[rviz2-4] [INFO] [1696567261.474883007] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Failed to call service get_planning_scene, have you launched move_group or called psm.providePlanningSceneService()?
</details>

peterdavidfagan, 2023-10-12:
> When executing motion_planning_python_api_tutorial.launch.py from moveit2_tutorial, Rviz shows NO PLANNING LIBRARY.

This is expected, the Python API is supposed to be used directly within your script in order to define the motion plans you want. If you want to interact via rviz it may be worthwhile launching rviz with `move_group`.

> When executing, I receive the INFO message: Failed to call service get_planning_scene, have you launched move_group or called psm.providePlanningSceneService()? and the process stops.

I need to check this issue.

yulat214, 2023-10-31:
Thank you for confirming.
Sorry it took so long to see this.
I have tried many times since then.

I had been getting this error in various environments, but after several builds it suddenly fixed itself.

It now works, but I haven't done anything special to it other than build it.

adrielmicom, 2023-11-01:
Hello, I have the same problem, any updates? 
`[rviz2-4] [INFO] [1698842496.286118580] [rviz2.planning_scene_monitor]: Failed to call service get_planning_scene, have you launched move_group or called psm.providePlanningSceneService()?`  

@yulat214  could you tell me how you built that last environment?

Thank you for your time

yulat214, 2023-11-02:
I forgot something important.
The tutorial assumes several movements, but I have only been able to confirm that moving to a random posture and placing objects works.

In [moveit2 tutorial motion planning python API](https://moveit.picknik.ai/main/doc/examples/motion_planning_python_api/motion_planning_python_api_tutorial.html), if you comment out the code shown in `Single Pipeline Planning - Pose Goal` and `Single Pipeline Planning - Custom Constraints`, does it work?
It seems that if there is a flaw somewhere in the code, everything will not work.

I have confirmed that it works on Ubuntu 22.04, ROS Rolling.
Best of luck!

yulat214, 2023-11-02:
By the way, I just tried `ros2 launch moveit2_tutorials motion_planning_python_api_tutorial.launch.py` about 10 times for checking, and 8 of them did not work.

`ros2 launch moveit2_tutorials motion_planning_python_api_tutorial.launch.py example_file:=motion_planning_python_api_planning_scene.py` and a picture of a successful execution is also shown.
[image]

adrielmicom, 2023-11-03:
If  i comment `single Pipeline Planning - Pose Goal and Single Pipeline Planning - Custom `Constraints,` it doesn’t work.
I get this when I run it (https://github.com/adrielmicom/moveit2_tutorials/blob/adrielmicom-patch-1/issues/784_1)

On the other hand, without commenting anything and following the tutorial, I get (https://github.com/adrielmicom/moveit2_tutorials/blob/adrielmicom-patch-1/issues/784_2).
With the same error whether using Planning Scene or not.  `motion_planning_python_api_planning_scene.py-1] [ERROR] [1698999635.573165687] [moveit_3093295755.occupancy_map_monitor]: No 3D sensor plugin(s) defined for octomap updates
[ERROR] [motion_planning_python_api_planning_scene.py-1]: process has died [pid 59413, exit code -11, cmd '/home/adri/ws_moveit/install/moveit2_tutorials/lib/moveit2_tutorials/motion_planning_python_api_planning_scene.py --ros-args -r __node:=moveit_py --params-file /tmp/launch_params_l8tvxx2c'].` 

I think this provides more information by being able to see the entire output.

yulat214, 2023-11-09:
I have seen the same errors many times. Sometimes the process does not die even with that error. The cause is unknown.

Sometimes I can do it and sometimes I can't. I am not sure how to run them definitely.

I will try some more, Please let me know if you find out anything.