moveit/moveit2_tutorials #787: how to use moveit with gazebo

wxydejoy, 2023-10-10:
### Description

how to use moveit with gazebo

### Your environment
* ROS Distro: humble
* OS Version:  Ubuntu 22.04
*  Binary build
### Please give me some help

I only found https://github.com/lbr-stack/lbr_fri_ros2_stack

Robotawi, 2023-10-10:
I was thinking of explaining MoveIt 2 & Gazebo (Fortress) integration as a sub-task of porting the [Multiple Robot Arms Tutorial](https://ros-planning.github.io/moveit_tutorials/doc/multiple_robot_arms/multiple_robot_arms_tutorial.html) to ROS 2. [This issue](https://github.com/ros-planning/moveit2_tutorials/issues/556) is related. The integration can be separated from the multiple arms tutorial to make it easier to understand. I will be able to dedicate time to this very soon.

wxydejoy, 2023-10-11:
thanks， Wish you all the best.

closed 2023-10-11 by wxydejoy