moveit/moveit2_tutorials #795: file 'pilz_mtc.launch.py' was not found

ravijo, 2023-10-23:
### Description
I am trying to setup [Pilz Industrial Motion Planner](https://moveit.picknik.ai/main/doc/how_to_guides/pilz_industrial_motion_planner/pilz_industrial_motion_planner.html#pilz-industrial-motion-planner) by following the documentation in ROS 2 Humble. Unfortunately, the command `ros2 launch moveit2_tutorials pilz_mtc.launch.py` is not working.

```
$ ros2 launch moveit2_tutorials pilz_mtc.launch.py
file 'pilz_mtc.launch.py' was not found in the share directory of package 'moveit2_tutorials' which is at '/home/user/moveit2_ws/install/moveit2_tutorials/share/moveit2_tutorials'
```

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04.3 LTS
* ROS Source or Binary build: Binary
* Moveit Source or Binary build: Source (humble branch)

### Steps to reproduce
```
$ ros2 launch moveit2_tutorials demo.launch.py rviz_config:=panda_hello_moveit.rviz
$ ros2 launch moveit2_tutorials mtc_demo.launch.py
$ ros2 launch moveit2_tutorials pilz_mtc.launch.py
file 'pilz_mtc.launch.py' was not found in the share directory of package 'moveit2_tutorials' which is at '/home/user/moveit2_ws/install/moveit2_tutorials/share/moveit2_tutorials'
```

### Objective
I just need a working demo of Pilz Industrial Motion Planner. It could be in ROS 1 or ROS 2.  I could not find any. Unfortunately, the ROS 1 demo is referring to a [read-online repository](https://docs.ros.org/en/melodic/api/moveit_tutorials/html/doc/pilz_industrial_motion_planner/pilz_industrial_motion_planner.html#action-interface) and it is also [showing errors reported here](https://github.com/PilzDE/pilz_tutorials/issues/37). 

Is there a working demo of Pilz Industrial Motion Planner available online?

sea-bass, 2023-10-23:
I think this tutorial broke in the changing of default robots for the Kinova Gen3.

Just fixed it all up with a combination of 2 PRs, if you'd like to test now:
* https://github.com/ros-planning/moveit2_tutorials/pull/796
* https://github.com/ros-planning/moveit_resources/pull/190

Otherwise, we'll get this merged soon!

sea-bass, 2023-10-23:
Oh by the way, this tutorial is only on the `main` branch of the tutorials, and not on `humble` -- so that may be why the file itself is not found.

We'll either need to backport that tutorial to Humble, or you can try using the `main` branch of the tutorials -- noting that if you do this, you'll need to be on ROS 2 Rolling.

ravijo, 2023-10-24:
@sea-bass 

Thank you very much for the quick response. It works like a charm!

Adding commands for reference only:

```
$ source /opt/ros/rolling/setup.bash
$ source ~/ws_moveit2_ros2_rolling/install/setup.bash
$ ros2 launch moveit_resources_panda_moveit_config demo.launch.py
$ ros2 run moveit2_tutorials pilz_move_group
```

Thank you!

closed 2023-10-24 by sjahr

ravijo, 2023-10-24:
@sea-bass @sjahr 

Sorry for commenting on a closed issue. Just wanted to know how to port this to the UR robot. The [panda_moveit_config](https://github.com/ros-planning/moveit_resources/tree/3d6a737f95fb7e5f3cfdc306d12b0bb6cbdadfb2/panda_moveit_config/config) contains [pilz_industrial_motion_planner_planning.yaml](https://github.com/ros-planning/moveit_resources/blob/3d6a737f95fb7e5f3cfdc306d12b0bb6cbdadfb2/panda_moveit_config/config/pilz_industrial_motion_planner_planning.yaml), but [ur_moveit_config](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/tree/5b538e2403a3d31526c4a07c2bd484a5265a2713/ur_moveit_config/config) does not have it.

Thanks!

ravijo, 2023-10-26:
Adding this comment for future readers. A demonstration of Pilz Industrial Motion Planner in MoveIt with Universal Robots is now available here: https://github.com/ravijo/ur_pilz_demo

Thanks

sjahr, 2023-10-26:
Sorry, I just saw your comment. Glad you solved it! Just open a new issue next time to increase visibility