moveit/moveit2_tutorials #801: Sequence Tutorial for Pilz Industrial Motion Planner

ravijo, 2023-10-27:
### Description
I am looking for Sequence Tutorial for Pilz Industrial Motion Planner in MoveIt 2.

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Source or Binary build: Source

### Additional Info
Below is the code snippet showing sequence tutorial  in Python for ROS Melodic:
```
    # Repeat the previous steps with a sequence command
    sequence = Sequence()
    sequence.append(Lin(goal=Pose(position=Point(0.2, 0, 0.8)), vel_scale=0.1, acc_scale=0.1))
    sequence.append(Circ(goal=Pose(position=Point(0.2, -0.2, 0.8)), center=Point(0.1, -0.1, 0.8), acc_scale=0.1))
    sequence.append(Ptp(goal=pose_after_relative, vel_scale=0.2))

    r.move(sequence)

    # Move to start goal for sequence demonstration
    r.move(Ptp(goal=start_joint_values))

    # Blend sequence
    blend_sequence = Sequence()
    blend_sequence.append(Lin(goal=Pose(position=Point(0.2, 0, 0.7)), acc_scale=0.05), blend_radius=0.01)
    blend_sequence.append(Lin(goal=Pose(position=Point(0.2, 0.1, 0.7)), acc_scale=0.05))

    r.move(blend_sequence)
```
Source:  https://docs.ros.org/en/melodic/api/pilz_robot_programming/html/

sea-bass, 2023-10-27:
I don't think this part of the tutorial has been ported over, so you're welcome to give it a shot! I'd suggest looking at the integration test in the `moveit2` repo that does a sequence with Pilz:

https://github.com/ros-planning/moveit2/blob/main/moveit_planners/pilz_industrial_motion_planner/test/integration_tests/src/integrationtest_sequence_action.cpp

thanh7040, 2024-01-19:
@ravijo Did you able to implement it?. Can you give me a hint?

sea-bass, 2024-08-01:
We had someone recently contribute this tutorial, and the PR is a few small comments away from being merged!
https://github.com/moveit/moveit2_tutorials/pull/917

ravijo, 2024-08-01:
Thanks for the update @sea-bass

closed 2024-08-01 by ravijo