moveit/moveit2_tutorials #804: MoveItPy tutorial seems not working

ravijo, 2023-10-29:
### Description

I am trying to get started with Motion Planning Python API by following [the official tutorial](https://moveit.picknik.ai/main/doc/examples/motion_planning_python_api/motion_planning_python_api_tutorial.html). RViz window pops up, but the Panda arm is not moving. RViz shows a warning saying, "Requesting initial scene failed," as shown in the screenshot below:

[image]

### Your environment
* ROS Distro: Tried on Humble and Iron
* OS Version: Ubuntu 22.04.3 LTS
* ROS Source or Binary build: Binary
* Moveit Source or Binary build: main branch

### Steps to reproduce
I am executing the following command mentioned in [the documentation](https://moveit.picknik.ai/main/doc/examples/motion_planning_python_api/motion_planning_python_api_tutorial.html#getting-started).
```console
$ ros2 launch moveit2_tutorials motion_planning_python_api_tutorial.launch.py
```

### Expected behaviour
I expect the Panda arm to move.

### Backtrace or Console output
* ROS Humble (Moveit main Branch)
    ```console
    $ ros2 launch moveit2_tutorials motion_planning_python_api_tutorial.launch.py
    [INFO] [launch]: All log files can be found below /home/user/.ros/log/2023-10-29-12-23-29-760558-dell22-16552
    [INFO] [launch]: Default logging verbosity is set to INFO
    [INFO] [motion_planning_python_api_tutorial.py-1]: process started with pid [16553]
    [INFO] [robot_state_publisher-2]: process started with pid [16555]
    [INFO] [ros2_control_node-3]: process started with pid [16557]
    [INFO] [rviz2-4]: process started with pid [16559]
    [INFO] [static_transform_publisher-5]: process started with pid [16561]
    [INFO] [ros2 run controller_manager spawner panda_arm_controller-6]: process started with pid [16563]
    [INFO] [ros2 run controller_manager spawner panda_hand_controller-7]: process started with pid [16565]
    [INFO] [ros2 run controller_manager spawner joint_state_broadcaster-8]: process started with pid [16567]
    [rviz2-4] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
    [static_transform_publisher-5] [INFO] [1698549810.026716929] [static_transform_publisher]: Spinning until stopped - publishing transform
    [static_transform_publisher-5] translation: ('0.000000', '0.000000', '0.000000')
    [static_transform_publisher-5] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
    [static_transform_publisher-5] from 'world' to 'panda_link0'
    [robot_state_publisher-2] [WARN] [1698549810.034222775] [kdl_parser]: The root link panda_link0 has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
    [ros2_control_node-3] [INFO] [1698549810.034337568] [resource_manager]: Loading hardware 'PandaFakeSystem' 
    [ros2_control_node-3] [INFO] [1698549810.034831281] [resource_manager]: Initialize hardware 'PandaFakeSystem' 
    [ros2_control_node-3] [WARN] [1698549810.034851566] [mock_generic_system]: Parsing of optional initial interface values failed or uses a deprecated format. Add initial values for every state interface in the ros2_control.xacro. For example: 
    [ros2_control_node-3] <state_interface name="velocity"> 
    [ros2_control_node-3]   <param name="initial_value">0.0</param> 
    [ros2_control_node-3] </state_interface>
    [ros2_control_node-3] [INFO] [1698549810.034858474] [resource_manager]: Successful initialization of hardware 'PandaFakeSystem'
    [ros2_control_node-3] [INFO] [1698549810.034874124] [resource_manager]: Loading hardware 'PandaHandFakeSystem' 
    [ros2_control_node-3] [INFO] [1698549810.034881629] [resource_manager]: Initialize hardware 'PandaHandFakeSystem' 
    [ros2_control_node-3] [INFO] [1698549810.034888470] [resource_manager]: Successful initialization of hardware 'PandaHandFakeSystem'
    [ros2_control_node-3] [INFO] [1698549810.034909869] [resource_manager]: 'configure' hardware 'PandaFakeSystem' 
    [ros2_control_node-3] [INFO] [1698549810.034914097] [resource_manager]: Successful 'configure' of hardware 'PandaFakeSystem'
    [ros2_control_node-3] [INFO] [1698549810.034918749] [resource_manager]: 'activate' hardware 'PandaFakeSystem' 
    [ros2_control_node-3] [INFO] [1698549810.034922177] [resource_manager]: Successful 'activate' of hardware 'PandaFakeSystem'
    [ros2_control_node-3] [INFO] [1698549810.034925245] [resource_manager]: 'configure' hardware 'PandaHandFakeSystem' 
    [ros2_control_node-3] [INFO] [1698549810.034928386] [resource_manager]: Successful 'configure' of hardware 'PandaHandFakeSystem'
    [ros2_control_node-3] [INFO] [1698549810.034931591] [resource_manager]: 'activate' hardware 'PandaHandFakeSystem' 
    [ros2_control_node-3] [INFO] [1698549810.034934645] [resource_manager]: Successful 'activate' of hardware 'PandaHandFakeSystem'
    [robot_state_publisher-2] [INFO] [1698549810.036400306] [robot_state_publisher]: got segment panda_hand
    [robot_state_publisher-2] [INFO] [1698549810.036697872] [robot_state_publisher]: got segment panda_leftfinger
    [robot_state_publisher-2] [INFO] [1698549810.036930449] [robot_state_publisher]: got segment panda_link0
    [robot_state_publisher-2] [INFO] [1698549810.037149145] [robot_state_publisher]: got segment panda_link1
    [ros2_control_node-3] [INFO] [1698549810.037176510] [controller_manager]: update rate is 100 Hz
    [ros2_control_node-3] [INFO] [1698549810.037313693] [controller_manager]: RT kernel is recommended for better performance
    [robot_state_publisher-2] [INFO] [1698549810.037690384] [robot_state_publisher]: got segment panda_link2
    [robot_state_publisher-2] [INFO] [1698549810.037911819] [robot_state_publisher]: got segment panda_link3
    [robot_state_publisher-2] [INFO] [1698549810.038125495] [robot_state_publisher]: got segment panda_link4
    [robot_state_publisher-2] [INFO] [1698549810.038336649] [robot_state_publisher]: got segment panda_link5
    [robot_state_publisher-2] [INFO] [1698549810.038545227] [robot_state_publisher]: got segment panda_link6
[... 186 lines cut ...]
    [motion_planning_python_api_tutorial.py-1] [INFO] [1698553200.078382398] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
    [motion_planning_python_api_tutorial.py-1] [INFO] [1698553200.078406980] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
    [motion_planning_python_api_tutorial.py-1] [INFO] [1698553200.078693158] [moveit_1639447295.trajectory_execution_manager]: Trajectory execution is managing controllers
    [motion_planning_python_api_tutorial.py-1] [INFO] [1698553200.082973297] [moveit_py.pose_goal]: MoveItPy instance created
    [motion_planning_python_api_tutorial.py-1] Traceback (most recent call last):
    [motion_planning_python_api_tutorial.py-1]   File "/home/user/ws_moveit2_main_iron/install/moveit2_tutorials/lib/moveit2_tutorials/motion_planning_python_api_tutorial.py", line 176, in <module>
    [motion_planning_python_api_tutorial.py-1]     main()
    [motion_planning_python_api_tutorial.py-1]   File "/home/user/ws_moveit2_main_iron/install/moveit2_tutorials/lib/moveit2_tutorials/motion_planning_python_api_tutorial.py", line 71, in main
    [motion_planning_python_api_tutorial.py-1]     panda_arm.set_start_state(configuration_name="ready")
    [motion_planning_python_api_tutorial.py-1] AttributeError: 'moveit.planning.PlanningComponent' object has no attribute 'set_start_state'. Did you mean: 'get_start_state'?
    [motion_planning_python_api_tutorial.py-1] [INFO] [1698553200.111919548] [moveit_1639447295.moveit_cpp]: Deleting MoveItCpp
    [motion_planning_python_api_tutorial.py-1] [INFO] [1698553200.117838194] [moveit_1639447295.planning_scene_monitor]: Stopped publishing maintained planning scene.
    [motion_planning_python_api_tutorial.py-1] [INFO] [1698553200.118418570] [moveit_1639447295.planning_scene_monitor]: Stopping world geometry monitor
    [motion_planning_python_api_tutorial.py-1] [INFO] [1698553200.118552100] [moveit_1639447295.planning_scene_monitor]: Stopping planning scene monitor
    [INFO] [ros2 run controller_manager spawner panda_arm_controller-6]: process has finished cleanly [pid 80154]
    [INFO] [ros2 run controller_manager spawner joint_state_broadcaster-8]: process has finished cleanly [pid 80158]
[... cut ...]
    [ros2 run controller_manager spawner panda_hand_controller-7] [INFO] [1698553200.819829624] [spawner_panda_hand_controller]: Configured and activated panda_hand_controller
    [INFO] [ros2 run controller_manager spawner panda_hand_controller-7]: process has finished cleanly [pid 80156]
    [rviz2-4] [INFO] [1698553205.111913437] [rviz2.planning_scene_monitor]: GetPlanningScene service call to get_planning_scene timed out. at /home/user/ws_moveit2_main_iron/src/moveit2/moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp:607
    ```

sea-bass, 2023-10-29:
What version is your `moveit_resources` repo? I submitted a fix related to this issue for the Panda configuration ~6 days ago in https://github.com/ros-planning/moveit_resources/pull/190 -- can you ensure you have that, rebuid, and try again?

sea-bass, 2023-10-29:
Actually, never mind, might be a duplicate of https://github.com/ros-planning/moveit2_tutorials/issues/784

ravijo, 2023-10-30:
@sea-bass 

Thanks for quick response. Using https://github.com/ros-planning/moveit_resources/pull/190, the RViz warning is gone. But the robot arm is still not moving. There is an `AttributeError` displayed on the terminal. 

Below are the commands:

```
$ cd moveit_resources
$ git log
commit 3d6a737f95fb7e5f3cfdc306d12b0bb6cbdadfb2 (HEAD -> ros2, origin/ros2, origin/HEAD)
Author: Sebastian Castro <4603398+sea-bass@users.noreply.github.com>
Date:   Tue Oct 24 02:56:55 2023 -0400

    Fix Panda demo launch for MoveIt 2 tutorials (#190)

$ git fetch origin pull/190/head:moveitpy
$ colcon build --mixin release

$ ros2 launch moveit2_tutorials motion_planning_python_api_tutorial.launch.py
[INFO] [launch]: All log files can be found below /home/user/.ros/log/2023-10-30-12-32-56-639336-dell-87156
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [motion_planning_python_api_tutorial.py-1]: process started with pid [87159]
[INFO] [robot_state_publisher-2]: process started with pid [87161]
[INFO] [ros2_control_node-3]: process started with pid [87163]
[INFO] [rviz2-4]: process started with pid [87165]
[INFO] [static_transform_publisher-5]: process started with pid [87167]
[INFO] [ros2 run controller_manager spawner panda_arm_controller-6]: process started with pid [87169]
[INFO] [ros2 run controller_manager spawner panda_hand_controller-7]: process started with pid [87171]
[INFO] [ros2 run controller_manager spawner joint_state_broadcaster-8]: process started with pid [87173]
[static_transform_publisher-5] [INFO] [1698636776.975412383] [static_transform_publisher]: Spinning until stopped - publishing transform
[static_transform_publisher-5] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-5] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-5] from 'world' to 'panda_link0'
[ros2_control_node-3] [WARN] [1698636776.976797857] [controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ros2_control_node-3] [INFO] [1698636776.977012410] [resource_manager]: Loading hardware 'PandaFakeSystem' 
[ros2_control_node-3] [INFO] [1698636776.977513324] [resource_manager]: Initialize hardware 'PandaFakeSystem' 
[ros2_control_node-3] [WARN] [1698636776.977546227] [mock_generic_system]: Parsing of optional initial interface values failed or uses a deprecated format. Add initial values for every state interface in the ros2_control.xacro. For example: 
[ros2_control_node-3] <state_interface name="velocity"> 
[ros2_control_node-3]   <param name="initial_value">0.0</param> 
[ros2_control_node-3] </state_interface>
[ros2_control_node-3] [INFO] [1698636776.977553846] [resource_manager]: Successful initialization of hardware 'PandaFakeSystem'
[ros2_control_node-3] [INFO] [1698636776.977567124] [resource_manager]: Loading hardware 'PandaHandFakeSystem' 
[ros2_control_node-3] [INFO] [1698636776.977573618] [resource_manager]: Initialize hardware 'PandaHandFakeSystem' 
[ros2_control_node-3] [INFO] [1698636776.977580122] [resource_manager]: Successful initialization of hardware 'PandaHandFakeSystem'
[ros2_control_node-3] [INFO] [1698636776.977610189] [resource_manager]: 'configure' hardware 'PandaHandFakeSystem' 
[ros2_control_node-3] [INFO] [1698636776.977613660] [resource_manager]: Successful 'configure' of hardware 'PandaHandFakeSystem'
[ros2_control_node-3] [INFO] [1698636776.977617504] [resource_manager]: 'activate' hardware 'PandaHandFakeSystem' 
[ros2_control_node-3] [INFO] [1698636776.977620448] [resource_manager]: Successful 'activate' of hardware 'PandaHandFakeSystem'
[ros2_control_node-3] [INFO] [1698636776.977623605] [resource_manager]: 'configure' hardware 'PandaFakeSystem' 
[ros2_control_node-3] [INFO] [1698636776.977626420] [resource_manager]: Successful 'configure' of hardware 'PandaFakeSystem'
[ros2_control_node-3] [INFO] [1698636776.977630753] [resource_manager]: 'activate' hardware 'PandaFakeSystem' 
[ros2_control_node-3] [INFO] [1698636776.977633511] [resource_manager]: Successful 'activate' of hardware 'PandaFakeSystem'
[rviz2-4] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[ros2_control_node-3] [INFO] [1698636776.981055760] [controller_manager]: update rate is 100 Hz
[ros2_control_node-3] [INFO] [1698636776.981196726] [controller_manager]: RT kernel is recommended for better performance
[robot_state_publisher-2] [WARN] [1698636776.986207136] [kdl_parser]: The root link panda_link0 has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[robot_state_publisher-2] [INFO] [1698636776.987972301] [robot_state_publisher]: got segment panda_hand
[robot_state_publisher-2] [INFO] [1698636776.988152969] [robot_state_publisher]: got segment panda_leftfinger
[robot_state_publisher-2] [INFO] [1698636776.988171135] [robot_state_publisher]: got segment panda_link0
[robot_state_publisher-2] [INFO] [1698636776.988182048] [robot_state_publisher]: got segment panda_link1
[robot_state_publisher-2] [INFO] [1698636776.988206463] [robot_state_publisher]: got segment panda_link2
[robot_state_publisher-2] [INFO] [1698636776.988216985] [robot_state_publisher]: got segment panda_link3
[... cut ...]
[motion_planning_python_api_tutorial.py-1] [INFO] [1698636778.406755408] [moveit_644873175.planning_scene_monitor]: Stopping planning scene monitor
[INFO] [ros2 run controller_manager spawner joint_state_broadcaster-8]: process has finished cleanly [pid 87173]
[ERROR] [motion_planning_python_api_tutorial.py-1]: process has died [pid 87159, exit code 1, cmd '/home/user/ws_moveit2_main_iron/install/moveit2_tutorials/lib/moveit2_tutorials/motion_planning_python_api_tutorial.py --ros-args -r __node:=moveit_py --params-file /tmp/launch_params_jwfxa8wc'].
```

LeonYo153, 2024-04-27:
Hello, I wonder if it's solved now? I met the same problem

mikeferguson, 2024-12-18:
Iron is now EOL (debs were quite out of date all along too), and the python interface is not available on Humble

closed 2024-12-18 by mikeferguson