moveit/moveit2_tutorials #806: Cannot run "Pick and Place with MoveIt Task Constructor" tutorial with provided docker image

VincentDelort, 2023-10-30:
### Description

Hi,

I cannot run the "Pick and Place with MoveIt Task Constructor" tutorial demo with the provided `moveit2_container` docker image : demo fails at the task execution step (last step) of [`5. Running the demo`](https://moveit.picknik.ai/main/doc/tutorials/pick_and_place_with_moveit_task_constructor/pick_and_place_with_moveit_task_constructor.html#running-the-demo)

### Your environment
* Host OS: Ubuntu 20.04
* Running environment: `rolling-tutorial` docker image provided by the MoveIt2 tutorial
  * ROS Distro: Rolling

Image inspect:
```text
[
    {
        "Id": "sha256:a39ea3a666ff742bbaf5a35dec2eaca35900c48edb939cea3ae43bd1ca7e4bc9",
        "RepoTags": [
            "ghcr.io/ros-planning/moveit2_tutorials:rolling-tutorial"
        ],
        "RepoDigests": [
            "ghcr.io/ros-planning/moveit2_tutorials@sha256:c3baa1a87e49cce8d4fcfa1b39d3fb5bdbb6a019837f9efb6e90e723bd4b8fea"
        ],
        "Parent": "",
        "Comment": "buildkit.dockerfile.v0",
        "Created": "2023-10-03T16:07:45.863603919Z",
        "Container": "",
        "ContainerConfig": {
            "Hostname": "",
            "Domainname": "",
            "User": "",
            "AttachStdin": false,
            "AttachStdout": false,
            "AttachStderr": false,
            "Tty": false,
            "OpenStdin": false,
            "StdinOnce": false,
            "Env": null,
            "Cmd": null,
            "Image": "",
            "Volumes": null,
            "WorkingDir": "",
            "Entrypoint": null,
            "OnBuild": null,
            "Labels": null
        },
        "DockerVersion": "",
        "Author": "",
        "Config": {
            "Hostname": "",
            "Domainname": "",
            "User": "",
            "AttachStdin": false,
            "AttachStdout": false,
            "AttachStderr": false,
            "Tty": false,
            "OpenStdin": false,
            "StdinOnce": false,
            "Env": [
                "PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin",
                "LANG=C.UTF-8",
                "LC_ALL=C.UTF-8",
                "ROS_DISTRO=rolling",
                "TERM=xterm",
                "ROS_UNDERLAY=/root/ws_moveit/install"
            ],
            "Cmd": [
                "bash"
            ],
            "ArgsEscaped": true,
            "Image": "",
[... 438 lines cut ...]
[move_group-4]         963: 
[move_group-4]         964:   /// std::make_unique for arrays of unknown bound
[move_group-4]         965:   template<typename _Tp>
[move_group-4] #7    Object "/opt/ros/rolling/lib/librclcpp.so", at 0x7fc19d562405, in 
[move_group-4] #6    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7fc19d2c64d7, in __cxa_throw
[move_group-4] #5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7fc19d2c6276, in std::terminate()
[move_group-4] #4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7fc19d2c620b, in 
[move_group-4] #3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7fc19d2bab9d, in 
[move_group-4] #2    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7fc19cff87f2, in abort
[move_group-4] #1    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7fc19d012475, in raise
[move_group-4] #0    Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7fc19d0669fc, in pthread_kill
[move_group-4] Aborted (Signal sent by tkill() 281907 0)
[ERROR] [move_group-4]: process has died [pid 281907, exit code -6, cmd '/root/ws_moveit/install/moveit_ros_move_group/lib/moveit_ros_move_group/move_group --ros-args --params-file /tmp/launch_params_6idegugl --params-file /tmp/launch_params_p8lkx9dq'].
```

In RViz, the planned path is displayed (transparent object) but not executed:

[image]

Note: if trying to replace the interpolation planner by the sample or cartesian planner, the demo crashes even sooner (either during planning or during task initialisation).

-----------
I have checked for other Issues, and I do not thing that this is a dupplicate... 

Thank you for any help you can provide!

rhaschke, 2023-10-31:
Looks like the `sampling_frequency` passed to TrajectoryMonitor is zero:

```
[move_group-4]         412:     double sampling_frequency = 0.0;
[move_group-4]         413:     node_->get_parameter_or("plan_execution.record_trajectory_state_frequency", sampling_frequency, 0.0);
[move_group-4]       > 414:     trajectory_monitor_ = std::make_shared<planning_scene_monitor::TrajectoryMonitor>(
[move_group-4]         415:         planning_scene_monitor_->getStateMonitor(), sampling_frequency);
```

sea-bass, 2023-12-16:
I actually resolved this on main in https://github.com/ros-planning/moveit2/pull/2423 -- I guess the Docker image must not yet have this change?