moveit/moveit2_tutorials #810: Motion planning python api tutorial with Universal Robot

Mennosytsma, 2023-11-02:
### Description
For my projects I encountered the issue that a lot of the tutorials presented are not directly suitable to apply for a Universal Robot.
As discussed in other issues like [#2014](https://github.com/ros-planning/moveit2/issues/2014) and [#3](https://github.com/ros-planning/moveit2_tutorials/issues/3) there are more people that would like to use other robots with the `moveit_py`. 
With the examples given in #2014 and [#2483](https://github.com/ros-planning/moveit2/issues/2483) I managed to get the tutorial also working for the Universal robots.  
To get the tutorial functional with the original yaml that is provided for the Panda robot I added more planner configs in the launch file in a way proposed in #808 

I have uploaded the code for this [here](https://github.com/Mennosytsma/moveit2_tutorials/tree/moveitpy_ur/doc/examples/motion_planning_python_api)

### Environment
* ROS Distro: [Rolling]
* OS Version: e.g. Ubuntu 22.04
* Source build [moveit 09019d2](https://github.com/ros-planning/moveit2/commit/09019d277993a4d2b9a2d48ea99110c27b8e70a6)

## Question
I have tested the basic functionality of these launch files and hope that it is useful for other people as well, but is this suitable for a pull request/ is it okay to have a duplicated tutorial?

sjahr, 2023-11-30:
@Mennosytsma Yes! Please create a PR with your proposed tutorial. If necessary we can reduce duplication throughout the review process

tylerjw, 2023-12-01:
Please open a PR, that would be awesome!

ozgur-kurt, 2024-05-30:
@Mennosytsma Could you help that how to integrate into UR ROS2 packages ?

FarStryke21, 2025-03-05:
@Mennosytsma 

Thanks a lot for documenting this. You saved me from a very big headache. 

On a side note, this method of updating the MoveIt config can be extended for the Octomap portion as well, which was demonstrated [here](https://moveit.picknik.ai/main/doc/examples/perception_pipeline/perception_pipeline_tutorial.html). Because of how UR defines their move group launch file, adding the sensors to the MoveIt config in the way defined in the tutorial is not directly possible.

I loaded the parameters defined in a `sensors_3d.yaml` and added it to the planning scene monitor parameters.

```
sensors_3d_yaml = load_yaml("ur_moveit_config", "config/sensors_3d.yaml")
planning_scene_monitor_parameters.update(sensors_3d_yaml)

```
These are later passed into the `move_group_node `in lines 201-221 in your [launch file](https://github.com/Mennosytsma/moveit2_tutorials/blob/moveitpy_ur/doc/examples/motion_planning_python_api/launch/motion_planning_python_api_tutorial_ur.launch.py). 

One thing to note here is when you create the `moveit_py_node` on line 322, passing the same planning scene monitor parameters has a little unexpected behavior. You can get away with popping the sensors key-value pair from the parameters, but then commands from `moveit_py_node `does not respect octomap as a collision object. 

This is something I am looking to solve. Have you already submitted a PR to UR's robot driver repo?