moveit/moveit2_tutorials #824: First demo fails to launch and gives error

Alxkory, 2023-11-30:
### Description

Receive error output while attempting to launch first demo of the tutorial
likely related to [issue #821](https://github.com/ros-planning/moveit2_tutorials/issues/821)

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Source or Binary build? Source
* If source, which git commit or tag? 71ee840

### Steps to reproduce
set up Ubuntu 22.04.3 iso on VMware workstation 16.
let Ubuntu update and restart.
install ROS Humble following : https://docs.ros.org/en/humble/Installation/Ubuntu-Install-Debians.html
install Moveit tutorial following: https://moveit.picknik.ai/main/doc/tutorials/getting_started/getting_started.html
run: ros2 launch moveit2_tutorials demo.launch.py

### Expected behaviour
Tutorial workspace should compile without errors
Demo should launch

### Backtrace or Console output

[console output](https://gist.github.com/Alxkory/00c000ce1794f9f734a3b629b5afa9ac)

sjahr, 2023-11-30:
This is directly related to #821

Abishalini, 2023-12-01:
This should be fixed by the changes in this PR - https://github.com/ros-planning/moveit2_tutorials/pull/826
Run `vcs import src < src/moveit2_tutorials/moveit2_tutorials.repos` and build again.

sjahr, 2023-12-01:
@Alxkory Can you test and review https://github.com/ros-planning/moveit2_tutorials/pull/827

tylerjw, 2023-12-01:
I verified this is working locally. Re-open if you are still experiencing this.