moveit/moveit2_tutorials #828: Motion plan succeeds but execution fails

Abishalini, 2023-12-04:
### Description

Ran the basic demo - `ros2 launch moveit2_tutorials demo.launch.py`.
The motion plan returns a success but cannot execute the motion.

### Your environment
* ROS Distro: Rolling
* OS Version: e.g. Ubuntu 22.04
* Source or Binary build? Source
* If binary, which release version?
* If source, which git commit or tag? latest main of moveit2 tutorials.

### Steps to reproduce
Run  `ros2 launch moveit2_tutorials demo.launch.py`. In the RViz Motion planning widget, set the start and end goal for the arm group and hit plan and execute. 

### Expected behavior
The arm should move if the motion plan succeeds

### Backtrace or Console output

If the start state given to the motion plan is different from the current state of the robot, planning will succeed but when we try to execute, there will be an error saying the start goal is not close to the current state
```
[move_group-4] [ERROR] [1701709565.380752655] [move_group.trajectory_execution_manager]: 
[move_group-4] Invalid Trajectory: start point deviates from current robot state more than 0.01
[move_group-4] joint 'joint_2': expected: 0.26, current: 0
[move_group-4] [INFO] [1701709565.380801452] [move_group]: Execution completed: ABORTED
[rviz2-1] [INFO] [1701709565.380980906] [rviz2.move_group_interface]: Execute request aborted
[rviz2-1] [ERROR] [1701709565.381804294] [rviz2.move_group_interface]: MoveGroupInterface::execute() failed or timeout reached
```
If the start state is the current state, the motion plan will succeed but it still does not execute the plan. There might be some floating point comparison mishap happening in the joint trajectory controller
```
ros2_control_node-5] [INFO] [1701709582.650210573] [joint_trajectory_controller]: Received new action goal
[ros2_control_node-5] [ERROR] [1701709582.650235815] [joint_trajectory_controller]: Velocity of last trajectory point of joint joint_1 is not zero: -0.000000
[move_group-4] [ERROR] [1701709582.650458094] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Goal was rejected by server
[move_group-4] [INFO] [1701709582.650459360] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: joint_trajectory_controller started execution
[move_group-4] [WARN] [1701709582.650488791] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Goal request rejected
[move_group-4] [ERROR] [1701709582.650489316] [move_group.trajectory_execution_manager]: Failed to send trajectory part 1 of 1 to controller joint_trajectory_controller
[move_group-4] [INFO] [1701709582.650503038] [move_group.trajectory_execution_manager]: Completed trajectory execution with status ABORTED ...
[move_group-4] [INFO] [1701709582.650558209] [move_group]: Execution completed: ABORTED
[rviz2-1] [INFO] [1701709582.650782042] [rviz2.move_group_interface]: Execute request aborted
[rviz2-1] [ERROR] [1701709582.651759097] [rviz2.move_group_interface]: MoveGroupInterface::execute() failed or timeout reached

```

sea-bass, 2023-12-04:
I just tested on main/rolling and it worked for me... could you try a `sudo apt-get update` to see if maybe some of the `ros2_control` packages need update?

Abishalini, 2023-12-04:
I did a `sudo apt update` yesterday and I have the 4.1.0 version of `ros2_control` and `ros2_controllers`. Is that what you have too?

Abishalini, 2023-12-04:
To get this working, I cloned `ros2_controllers` and modified how the floating point comparison happens [here](https://github.com/ros-controls/ros2_controllers/blob/23f944f26e7e21764be3acdb6610cd59866c9bfc/joint_trajectory_controller/src/joint_trajectory_controller.cpp#L1372) in the JTC code.

Changed it to 
```
static double EPSILON = 0.0001;
if (fabs(trajectory.points.back().velocities.at(i)) > EPSILON)
{
  ...
}
```

sea-bass, 2024-02-11:
@Abishalini can this issue be closed now? I think you fixed it in the JTC implementation, yeah?

Abishalini, 2024-02-12:
Yep the fix has been merged in `ros2_controllers`

closed 2024-02-12 by Abishalini