moveit/moveit2_tutorials #831: failing to build with moveit_calibration on moveit2

shrutichakraborty, 2023-12-05:
### Description

Hi§ I would like to use moveit hand-eye calibration plugin. I have used the noetic version before but now I am trying to use it on ros2. I ran the following commands in the src folder of my moveit2_ws to install the plugin : 

`git clone https://github.com/ros-planning/moveit_calibration.git`

`rosdep install -y --from-paths . --ignore-src --rosdistro humble`

[image]

And on running colcon build the build fails as well...it seems like it still has dependency on catkin. Any ideas how I can run the plugin on moveit2? 

### Your environment
* ROS Distro: Humble
* OS Version: e.g. Ubuntu 22.04

shrutichakraborty, 2023-12-08:
@Omdiwan , thanks, I know that, that's why I am running `colcon build` but if you look at the screenshot you will see that the build fails as the error seems to the rosdep keys were not resolved for moveit_calibration_gui and plugins as it is for some reason looking for rosdep definition for catkin. To me that seems like the authors have not completely migrated the moveit_calibration stack to ros2 and I would like to know if indeed that is the case of there is something I am missing out on? 

Thanks

Varun0157, 2024-08-08:
Hey @shrutichakraborty, have you found a fix? I have the same issue. 

ROS Distro: Humble
OS: Ubuntu 22.04 LTS

mikeferguson, 2024-12-18:
moveit_calibration is not yet fully ported to ROS2 - please check out https://github.com/moveit/moveit_calibration/pull/118 for more details

closed 2024-12-18 by mikeferguson