moveit/moveit2_tutorials #839: moveit2 assistant——Problems with auto-generated configuration files

hhz0328, 2023-12-14:
### Description

Problems with auto-generated configuration files

### My environment
* ROS Distro: [iron]
* OS Version: e.g. Ubuntu 22.04
* PC: x86

### Problem
When I use the moveit2 assistant and run demo.launch.py after configuring my urdf robot model. there is a problem with the rviz2 interface, not being able to display the robot model, and the interface exiting automatically.

### Terminal Errors
[rviz2-4] Error:   Semantic description is not specified for the same robot as the URDF
[rviz2-4]          at line 664 in ./src/model.cpp
[rviz2-4] Error:   Joint 'slipway' declared as part of group 'easy' is not known to the URDF
[rviz2-4]          at line 168 in ./src/model.cpp
[rviz2-4] Error:   Joint 'joint' declared as part of group 'easy' is not known to the URDF
[rviz2-4]          at line 168 in ./src/model.cpp
[rviz2-4] Warning: Group 'easy' is empty.
[rviz2-4]          at line 245 in ./src/model.cpp
[rviz2-4] Error:   Joint 'joint' declared as part of group state 'start' is not known to the URDF
[rviz2-4]          at line 346 in ./src/model.cpp
[rviz2-4] Error:   Joint 'slipway' declared as part of group state 'start' is not known to the URDF
[rviz2-4]          at line 346 in ./src/model.cpp
[rviz2-4] Warning: Link 'joint_Link' is not known to URDF. Cannot disable/enable collisons.
[rviz2-4]          at line 577 in ./src/model.cpp
[rviz2-4] Warning: Link 'slipway_Link' is not known to URDF. Cannot disable/enable collisons.
[rviz2-4]          at line 577 in ./src/model.cpp
[rviz2-4] Warning: Link 'joint_Link' is not known to URDF. Cannot disable/enable collisons.
[rviz2-4]          at line 572 in ./src/model.cpp
[rviz2-4] [INFO] [1702451602.140792438] [moveit_3133492212.rdf_loader]: Loaded robot model in 0.221612 seconds
[rviz2-4] [INFO] [1702451602.140932148] [moveit_robot_model.robot_model]: Loading robot model 'arm_robot'...
[rviz2-4] [WARN] [1702451602.874949326] [moveit_robot_model.robot_model]: Group 'easy' must have at least one valid joint
[rviz2-4] [WARN] [1702451602.874991766] [moveit_robot_model.robot_model]: Failed to add group 'easy'
[rviz2-4] [ERROR] [1702451602.875036145] [moveit_robot_model.robot_model]: Group state 'start' specified for group 'easy', but that group does not exist
[rviz2-4] [INFO] [1702451605.052677731] [moveit_3133492212.planning_scene_monitor]: Starting planning scene monitor
[rviz2-4] [INFO] [1702451605.055958382] [moveit_3133492212.planning_scene_monitor]: Listening to '/monitored_planning_scene'
[rviz2-4] [INFO] [1702451606.673219372] [interactive_marker_display_93855605001424]: Connected on namespace: /rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic
[rviz2-4] [WARN] [1702451606.748399556] [moveit_planning_scene.planning_scene]: Setting the scene for model 'user_test_model' but model 'arm_robot' is loaded.
[rviz2-4] [ERROR] [1702451606.750036961] [moveit_exceptions.exceptions]: Variable 'slipway' is not known to model 'arm_robot'
[rviz2-4] Exception thrown.
[rviz2-4] [WARN] [1702451606.752286925] [moveit_planning_scene.planning_scene]: Setting the scene for model 'user_test_model' but model 'arm_robot' is loaded.
[rviz2-4] [ERROR] [1702451606.752363879] [moveit_exceptions.exceptions]: Variable 'slipway' is not known to model 'arm_robot'
[rviz2-4] Exception thrown.
[rviz2-4] terminate called after throwing an instance of 'moveit::Exception'
[rviz2-4]   what():  Variable 'slipway' is not known to model 'arm_robot'
[ERROR] [rviz2-4]: process has died [pid 24649, exit code -6, cmd '/opt/ros/iron/lib/rviz2/rviz2 -d /home/ubuntu/ws_moveit/install/arm_robot_moveit_config/share/arm_robot_moveit_config/config/moveit.rviz --ros-args --params-file /tmp/launch_params_ezauxk5i --params-file /tmp/launch_params_9hyqdz8p'].

hhz0328, 2023-12-14:
I tried to modify **joint_limits.yaml** and **moveit_controllers.yaml**.In the **joint_limits.yaml** file, I set the value of all **has_velocity_limits** to false.In the **moveit_controllers.yaml** file,I removed duplicate **action_ns: follow_joint_trajectory** and **default: true**.
By doing the above, I solved the display problem in rviz2, but when I click the **Execute** button, the robot still can't motion plan. Is there any solution?
[image]

mikeferguson, 2024-12-18:
Iron is now EOL

closed 2024-12-18 by mikeferguson