moveit/moveit2_tutorials #862: moveit2__tutorials: Failed to load any planning pipelines.

zhangbo9426426, 2024-01-18:
### Description

Hi, I have installed moveit2 and can start demo normally by `ros2 launch dual_arm_moveit2 demo.launch.py`. But when I continued to run `ros2 run moveit2_tutorials moveit_cpp_tutorial` and tried to use the advanced api moveitcpp, the terminal got an error.The following is all the information output by the terminal:

> [INFO] [1703218871.704853446] [moveit_cpp_tutorial]: Initialize node
> [INFO] [1703218872.714803138] [moveit_cpp_tutorial]: Starting MoveIt Tutorials...
> [INFO] [1703218874.362360608] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 1.64742 seconds
> [INFO] [1703218874.362414467] [moveit_robot_model.robot_model]: Loading robot model 'panda'...
> [WARN] [1703218874.373798051] [moveit_ros.robot_model_loader]: No kinematics plugins defined. Fill and load kinematics.yaml!
> [INFO] [1703218874.388775160] [moveit.ros_planning_interface.moveit_cpp]: Listening to 'joint_states' for joint states
> [INFO] [1703218874.391539423] [moveit_ros.current_state_monitor]: Listening to joint states on topic 'joint_states'
> [INFO] [1703218874.395096161] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
> [INFO] [1703218874.396032978] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
> [INFO] [1703218874.396193747] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
> [INFO] [1703218874.396649030] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/planning_scene'
> [INFO] [1703218874.396749676] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
> [INFO] [1703218874.397136859] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'collision_object'
> [INFO] [1703218874.397506483] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'planning_scene_world' for planning scene world geometry
> [WARN] [1703218874.398091735] [moveit.ros.occupancy_map_monitor.middleware_handle]: Resolution not specified for Octomap. Assuming resolution = 0.1 instead
> [ERROR] [1703218874.398131114] [moveit.ros.occupancy_map_monitor.middleware_handle]: No 3D sensor plugin(s) defined for octomap updates
> [ERROR] [1703218874.399007046] [moveit.ros_planning_interface.moveit_cpp]: Failed to load any planning pipelines.
> [FATAL] [1703218874.399057056] [moveit.ros_planning_interface.moveit_cpp]: Failed to load planning pipelines from parameter server
> terminate called after throwing an instance of 'std::runtime_error'
>   what():  Failed to load planning pipelines from parameter server
> Stack trace (most recent call last):
> ros-planning/moveit2#13   Object "", at 0xffffffffffffffff, in 
> ros-planning/moveit2#12   Object "/home/zhangbo/moveit2_ws/install/moveit2_tutorials/lib/moveit2_tutorials/moveit_cpp_tutorial", at 0x55e98eec15e4, in _start
> ros-planning/moveit2#11   Source "../csu/libc-start.c", line 392, in __libc_start_main_impl [0x7fb04a229e3f]
> ros-planning/moveit2#10   Source "../sysdeps/nptl/libc_start_call_main.h", line 58, in __libc_start_call_main [0x7fb04a229d8f]
> ros-planning/moveit2#9    Object "/home/zhangbo/moveit2_ws/install/moveit2_tutorials/lib/moveit2_tutorials/moveit_cpp_tutorial", at 0x55e98eebfab1, in main
> ros-planning/moveit2#8    Object "/home/zhangbo/moveit2_ws/install/moveit_ros_planning/lib/libmoveit_cpp.so.2.5.5", at 0x7fb04afca572, in moveit_cpp::MoveItCpp::MoveItCpp(std::shared_ptr<rclcpp::Node> const&)
> ros-planning/moveit2#7    Object "/home/zhangbo/moveit2_ws/install/moveit_ros_planning/lib/libmoveit_cpp.so.2.5.5", at 0x7fb04afc31d5, in moveit_cpp::MoveItCpp::MoveItCpp(std::shared_ptr<rclcpp::Node> const&, moveit_cpp::MoveItCpp::Options const&) [clone .cold]
> ros-planning/moveit2#6    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7fb04a6ae4d7, in __cxa_throw
> ros-planning/moveit2#5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7fb04a6ae276, in std::terminate()
> ros-planning/moveit2#4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7fb04a6ae20b, in 
> ros-planning/moveit2#3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7fb04a6a2b9d, in 
> ros-planning/moveit2#2    Source "./stdlib/abort.c", line 79, in abort [0x7fb04a2287f2]
> ros-planning/moveit2#1    Source "../sysdeps/posix/raise.c", line 26, in raise [0x7fb04a242475]
> #0  | Source "./nptl/pthread_kill.c", line 89, in __pthread_kill_internal
>     | Source "./nptl/pthread_kill.c", line 78, in __pthread_kill_implementation
>       Source "./nptl/pthread_kill.c", line 44, in __pthread_kill [0x7fb04a2969fc]
> Aborted (Signal sent by tkill() 3675 1000)
> [ros2run]: Aborted

### Your environment
* ROS Distro: Humble
* OS Version:  Ubuntu 22.04
* Source or Binary build? Source
* If source, which branch? Sorry, I'm not sure. I followed these instructions to clone, I think these instructions can tell which branch I am installed.
`git clone https://github.com/ros-planning/moveit2_tutorials -b humble --depth 1  moveit2_ws/src/moveit2_tutorials`
`vcs import < moveit2_tutorials/moveit2_tutorials.repos`

zhangbo9426426, 2024-02-11:
@sea-bass Thank you for your reply.
I checked to see if there was any other binary version of moveit by do
 `apt list --installed | grep ros-humble-moveit`
 and there was one:
` ros-humble-moveit-msgs/jammy,now 2.1-1jammy.20231117.191327 amd64`
and I uninstall `ros-humble-moveit-msgs `by executing `sudo apt remove ros-humble-moveit*`
then running `ros2 launch dual_arm_moveit2 demo.launch.py` again, which causes RVIZ to fail to start with the following error message:

> PluginlibFactory: The plugin for class 'moveit_rviz_plugin/PlanningScene' failed to load. Error: Failed to load library /home/zhangbo/moveit2_ws/install/moveit_ros_visualization/lib/libmoveit_planning_scene_rviz_plugin.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library dlopen error: libmoveit_msgs__rosidl_typesupport_cpp.so:

After I re-installed the plugin using `sudo apt install ros-humble-moveit-msgs`, I was able to start RVIZ normally, but unfortunately I still came back to the problem I originally mentioned

Looking forward to your response.

sea-bass, 2024-02-11:
I think you may want to uninstall that moveit_msgs package and instead build it from source alongside your source build of moveit2. Could you try that?

zhangbo9426426, 2024-02-11:
> I think you may want to uninstall that moveit_msgs binary and instead build it from source alongside your source build of moveit2. Could you try that?
I have uninstalled binary moveit_msgs and built moveit_msgs from the source code, and then restarted the demo to enter the rviz interface normally. But when I run` ros2 run moveit2_tutorials moveit_cpp_tutorial` the initial problem still arises

sea-bass, 2024-02-11:
I see. Two things, then.

1. I didn't notice this before, but can you instead run `ros2 launch moveit2_tutorials moveit_cpp_tutorial.launch.py`? These errors may actually be that a lot of the launch parameters are missing if you just run the standalone executable with `ros2 run`.
2. If the above doesn't work, could you share your entire console log?

zhangbo9426426, 2024-02-11:
I was able to run `ros2 launch moveit2_tutorials moveit_cpp_tutorial.launch.py` normally, and at the same time hit the “next” button in RvizVisualToolGui in the RVIZ console and the arm would work normally.

In fact, my main purpose is to use the new MovitCPP API in ROS2 when writing programs in c++. But I found that every time the “moveit_cpp_tutorial.cpp“ demo program interrupt occurs at `auto moveit_cpp_ptr = std::make_shared<moveit_cpp::MoveItCpp>(node); `  by outputting information in the terminal.

sea-bass, 2024-02-11:
Yes, because the node needs certain parameters to work correctly, and if you call `ros2 run` on that executable, the node has no overridden parameters.

It shouldn't segfault, and anyone is welcome to fix this with better error handling, but also it's expected that MoveIt nodes need parameters to work correctly.

zhangbo9426426, 2024-02-11:
> Yes, because the node needs certain parameters to work correctly, and if you call `ros2 run` on that executable, the node has no overridden parameters unless you start adding things like `--ros-args -p ...` to the end of that command...
> 
> ... which launch files make a little easier to do.
> 
> It shouldn't segfault, and anyone is welcome to fix this with better error handling, but also it's expected that MoveIt nodes need parameters to work correctly.

I see. Thank you very much for your response.

sea-bass, 2024-02-11:
Of course! I'll go ahead and close this issue, but please let us know as anything else comes up.

closed 2024-02-11 by sea-bass