moveit/moveit2_tutorials #865: MOVEIT MoveIt Quickstart in RViz - Robot trail not showing up?

ACTSdesigner, 2024-01-22:
### Description

MOVEIT MoveIt Quickstart in RViz - Robot trail not showing up?

### Your environment
* ROS Distro: Iron
* OS Version: Ubuntu 22.04.3 LTS
* Binary build
* GNU bash, version 5.1.16(1)-release (x86_64-pc-linux-gnu)

### Steps to reproduce
I'm attempting to run the MOVEIT MoveIt Quickstart in RViz(https://moveit.picknik.ai/main/doc/tutorials/quickstart_in_rviz/quickstart_in_rviz_tutorial.html#moveit-quickstart-in-rviz) and when I get down to the Step 4: Use Motion Planning with the Kinova Gen 3, I don't see the intermediate grey robot trail point. I've done a clean install of Ubuntu, ROS Iron, MOVEIT2 with no change, the robot trail points are not generated, although I can get the robot to move from the start to the target position, so the path planning is taking place and the path execution also works.    

### Expected behaviour
When I get down to the Step 4: Use Motion Planning with the Kinova Gen 3, I don't see the intermediate grey robot trail point as shown in the screenshot. 

[Document8.docx](https://github.com/ros-planning/moveit2_tutorials/files/14013985/Document8.docx)

### Backtrace or Console output

[Document7.docx](https://github.com/ros-planning/moveit2_tutorials/files/14013491/Document7.docx)

See highlighted in \yellow warnings.

rleung31wh, 2024-03-02:
Ran into the exact same issue here (no trail shown when I checked "Show Trail"). Also using Ubuntu 22.04 with ROS Humble. Any solution or workaround?

Sgt-Harry, 2024-04-09:
Same Issue. 
And I think it may be related to some Warning above
like

>  [move_group-4] [INFO] [1712679511.502278833] [move_group.moveit.moveit.planners.ompl.model_based_planning_context]: Planner configuration 'manipulator' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.

Maybe I think

mikeferguson, 2024-12-18:
Iron is now EOL

closed 2024-12-18 by mikeferguson