moveit/moveit2_tutorials #875: Tutorial fails when instructions followed and CycloneDDS used on Iron

dsandber, 2024-02-22:
I followed the instructions for the MoveIt2 source build here:

https://moveit.ros.org/install-moveit2/source/

This involved using Cyclone DDS.  

When I ran the following:

`ros2 launch moveit2_tutorials motion_planning_python_api_tutorial.launch.py`

I got a lot of errors (attached) and the arm didn't move.  When I removed the `export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp` line and used the default middle-ware for Ubuntu Iron binary installation, everything worked fine.  Perhaps this issue comes from mixing the Ubuntu 22.04 Iron ROS2 binary installation with a MoveIt2 source installation?

I'd suggest changing the instructions above to not recommend using the Cyclone DDS on an iron build.  The main ROS2 instructions I believe also don't recommend it for Iron.  This might be the cause of #866 and #804 as well.

[failed.log](https://github.com/ros-planning/moveit2_tutorials/files/14372239/failed.log)

mikeferguson, 2024-12-18:
Iron is now EOL

closed 2024-12-18 by mikeferguson

AniketCOAST, 2025-04-09:
@dsandber If you have enabled ufw (firewall), then either disable it or add rules for enabling multicasting. That should solve your issue:
https://github.com/moveit/moveit2/issues/3418#issuecomment-2787587857