moveit/moveit2_tutorials #876: MotionPlanning plan but do not execute

fjanovitz, 2024-02-29:
### Description

I'm using Moveit in Rviz to move a UR5 robot both in a URSim and a real robot, in both i can do the planning, it shows the trajectory but when i try to execute it always give me a "Failed" message. I don't know if it is a simple thing or not, but i didn't found anything similar to this problem in ROS Forums, Stack Exchange, etc.

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Source or Binary build? Source
* If binary, which release version?
* If source, which git commit or tag? The lastest

### Steps to reproduce
In the first terminal:
`docker run --rm -it -p 5900:5900 -p 6080:6080 -v ${HOME}/.ursim/urcaps:/urcaps -v ${HOME}/.ursim/programs:/ursim/programs --name ursim_cb3 universalrobots/ursim_cb3
`
In the second terminal:
`ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5 robot_ip:=172.17.0.2 launch_rviz:=false initial_joint_controller:=joint_trajectory_controller
`
In the third terminal:
`ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur5 launch_rviz:=true
`
After that, just go to Rviz and try to use the MotionPlanning.

### Expected behaviour
Robots should move according to the planned trajectory

### Backtrace or Console output
This appeared on the third terminal

[image]

If it's a simple problem, I'm sorry for opening an Issue to solve it, I recently started using ROS/Moveit and I'm having some difficulties.

rhaschke, 2024-02-29:
The approach to use scaled_joint_trajectory_controller is correct for UR robots. But I don't understand why the controller rejects the trajectory goal. Are there any error messages from the ROS controller manager running the controller? Probably this was started in yet another terminal.

fjanovitz, 2024-03-04:
In the second terminal, i had the following error:
[image]

These two lines appeared always when i tried to execute, i had found similar error messages but when i tried to do the respective solutions (when they had a solution), it did not solve the problem.