moveit/moveit2_tutorials #880: i cannot successfully run "ros2 launch moveit2_tutorials motion_planning_pipeline_tutorial.launch.py",The terminal shows file was not found in the share

yangming517, 2024-03-11:
### Your environment
* ROS Distro: [Humble]
* OS Version: Ubuntu 22.04
* Source or Binary build?
* If binary, which release version?
* If source, which git commit or tag?

### Description
I followed the offcial tutorial:getting started [https://moveit.picknik.ai/main/doc/tutorials/tutorials.html](url)
However, the following operations are different from the official ones
git clone https://github.com/ros-planning/moveit2_tutorials **-b humble** 
colcon build **--cmake-args -DCMAKE_BUILD_TYPE=Release --parallel-workers 1**

Obviously,command with "-b humble" will result in missing some  packages,such as 
"motion_planning_pipeline、parallel planning...."。I want to know if humble doesn't support these packages or these functions?

Because if I execute the  command "git clone https://github.com/ros-planning/moveit2_tutorials" without "-b humble"
The process that colcon build will fail.

The terminal shows that:
ym@ym-PC:~/ws_moveit2test2$ colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release --parallel-workers 1
Starting >>> moveit_common
Finished <<< moveit_common [0.20s]                   
Starting >>> moveit_msgs
Finished <<< moveit_msgs [2.21s]                      
Starting >>> moveit_configs_utils
--- stderr: moveit_configs_utils                    
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< moveit_configs_utils [0.78s]
Starting >>> moveit_resources_panda_description
Finished <<< moveit_resources_panda_description [0.19s]                   
Starting >>> moveit_resources_panda_moveit_config
Finished <<< moveit_resources_panda_moveit_config [0.20s]                   
Starting >>> moveit_resources_fanuc_description
Finished <<< moveit_resources_fanuc_description [0.20s]                   
Starting >>> moveit_resources_fanuc_moveit_config
Finished <<< moveit_resources_fanuc_moveit_config [0.20s]                   
Starting >>> robotiq_description
Finished <<< robotiq_description [0.19s]                   
Starting >>> moveit_task_constructor_msgs
Finished <<< moveit_task_constructor_msgs [0.73s]                      
Starting >>> rviz_marker_tools
Finished <<< rviz_marker_tools [0.24s]                   
Starting >>> kortex_description
Finished <<< kortex_description [0.19s]                    
Starting >>> moveit_resources_pr2_description
Finished <<< moveit_resources_pr2_description [0.21s]                    
Starting >>> moveit_core
Finished <<< moveit_core [0.60s]                       
Starting >>> moveit_ros_occupancy_map_monitor
Finished <<< moveit_ros_occupancy_map_monitor [0.28s]                    
Starting >>> moveit_ros_planning
Finished <<< moveit_ros_planning [0.41s]                      
Starting >>> moveit_kinematics
Finished <<< moveit_kinematics [0.31s]                    
Starting >>> moveit_ros_move_group
Finished <<< moveit_ros_move_group [0.35s]                       
Starting >>> moveit_ros_warehouse
Finished <<< moveit_ros_warehouse [0.31s]                      
Starting >>> moveit_ros_robot_interaction
Finished <<< moveit_ros_robot_interaction [0.29s]                      
Starting >>> moveit_simple_controller_manager
Finished <<< moveit_simple_controller_manager [0.25s]                        
Starting >>> moveit_ros_planning_interface
Finished <<< moveit_ros_planning_interface [0.30s]                     
Starting >>> moveit_ros_visualization
Finished <<< moveit_ros_visualization [0.56s]                       
Starting >>> moveit_planners_ompl
Finished <<< moveit_planners_ompl [0.32s]                       
Starting >>> moveit_setup_framework
Finished <<< moveit_setup_framework [0.31s]                     
Starting >>> moveit_planners_stomp
Finished <<< moveit_planners_stomp [0.27s]                     
Starting >>> moveit_setup_app_plugins
Finished <<< moveit_setup_app_plugins [0.38s]                        
Starting >>> moveit_setup_controllers
Finished <<< moveit_setup_controllers [0.42s]                        
Starting >>> moveit_setup_core_plugins
Finished <<< moveit_setup_core_plugins [0.34s]                      
Starting >>> moveit_setup_srdf_plugins
Finished <<< moveit_setup_srdf_plugins [0.43s]                       
Starting >>> moveit_setup_assistant
Finished <<< moveit_setup_assistant [0.31s]                       
Starting >>> moveit_resources_prbt_support
Finished <<< moveit_resources_prbt_support [0.20s]                     
Starting >>> moveit_resources_prbt_ikfast_manipulator_plugin
Finished <<< moveit_resources_prbt_ikfast_manipulator_plugin [0.27s]
Starting >>> moveit_plugins
Finished <<< moveit_plugins [0.20s]                     
Starting >>> moveit_ros_benchmarks
Finished <<< moveit_ros_benchmarks [0.30s]                     
Starting >>> kortex_api
Finished <<< kortex_api [0.20s]                     
Starting >>> rviz_visual_tools
Finished <<< rviz_visual_tools [0.44s]                       
Starting >>> serial
Finished <<< serial [0.22s]                     
Starting >>> chomp_motion_planner
Finished <<< chomp_motion_planner [0.28s]                        
Starting >>> moveit_ros_perception
Finished <<< moveit_ros_perception [0.31s]                       
Starting >>> moveit_resources_prbt_moveit_config
Finished <<< moveit_resources_prbt_moveit_config [0.21s]                     
Starting >>> moveit_ros
Finished <<< moveit_ros [0.21s]                     
Starting >>> kortex_driver
Finished <<< kortex_driver [0.25s]                     
Starting >>> robotiq_driver
Finished <<< robotiq_driver [0.31s]                     
Starting >>> pick_ik
--- stderr: pick_ik                                
/home/ym/ws_moveit2test2/build/pick_ik/tests/test-pick_ik: symbol lookup error: /opt/ros/humble/lib/libpick_ik_plugin.so: undefined symbol: _ZN10kinematics14KinematicsBase6LOGGERE
CMake Error at /home/ym/ws_moveit2test2/build/pick_ik/_deps/catch2-src/extras/CatchAddTests.cmake:60 (message):
  Error running test executable
  '/home/ym/ws_moveit2test2/build/pick_ik/tests/test-pick_ik':

    Result: 127
    Output: 

gmake[2]: *** [tests/CMakeFiles/test-pick_ik.dir/build.make:340：tests/test-pick_ik] 错误 1
gmake[2]: *** 正在删除文件“tests/test-pick_ik”
gmake[1]: *** [CMakeFiles/Makefile2:244：tests/CMakeFiles/test-pick_ik.dir/all] 错误 2
gmake: *** [Makefile:146：all] 错误 2
---
Failed   <<< pick_ik [0.37s, exited with code 2]

Summary: 43 packages finished [17.8s]
  1 package failed: pick_ik
  2 packages had stderr output: moveit_configs_utils pick_ik
  25 packages not processed

who can tell me How to experience all the features of moveit2_tutotials,thanks in advance.

sea-bass, 2024-03-11:
As discussed in https://github.com/PickNikRobotics/pick_ik/issues/65 and https://github.com/PickNikRobotics/pick_ik/issues/64, could you try removing your MoveIt binary installs that may be lingering? Thanks!

sea-bass, 2024-03-11:
There are no binaries for MoveIt Task Constructor. But it seems like you may have mismatched versions of both that repo and `moveit_visual_tools`. Can you make sure you've got the `ros2` branches of both of those, respectively?

yangming517, 2024-03-12:
@sea-bass Hi,I have big problem.Though I successfully build,I find that I almost meet the same problem " planning scene requesting initial scene failed"every time when i launch the "demo.launch.py""mtc_demo.launch.py"and so forth.
when i launch demo.launch.py,the terminal shows above
`ym@ym-PC:~/ws_moveit2test2$ ros2 launch moveit2_tutorials demo.launch.py 
[INFO] [launch]: All log files can be found below /home/ym/.ros/log/2024-03-12-18-44-23-219971-ym-PC-54476
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [54477]
[INFO] [static_transform_publisher-2]: process started with pid [54479]
[INFO] [robot_state_publisher-3]: process started with pid [54481]
[INFO] [move_group-4]: process started with pid [54483]
[INFO] [ros2_control_node-5]: process started with pid [54485]
[INFO] [spawner-6]: process started with pid [54487]
[INFO] [spawner-7]: process started with pid [54489]
[INFO] [spawner-8]: process started with pid [54491]
[static_transform_publisher-2] [INFO] [1710240263.707630945] [static_transform_publisher]: Spinning until stopped - publishing transform
[static_transform_publisher-2] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-2] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-2] from 'world' to 'base_link'
[robot_state_publisher-3] [INFO] [1710240263.714744396] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1710240263.714831188] [robot_state_publisher]: got segment bracelet_link
[robot_state_publisher-3] [INFO] [1710240263.714844536] [robot_state_publisher]: got segment end_effector_link
[robot_state_publisher-3] [INFO] [1710240263.714853070] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-3] [INFO] [1710240263.714860779] [robot_state_publisher]: got segment half_arm_1_link
[robot_state_publisher-3] [INFO] [1710240263.714868104] [robot_state_publisher]: got segment half_arm_2_link
[robot_state_publisher-3] [INFO] [1710240263.714875550] [robot_state_publisher]: got segment robotiq_85_base_link
[robot_state_publisher-3] [INFO] [1710240263.714883530] [robot_state_publisher]: got segment robotiq_85_left_finger_link
[robot_state_publisher-3] [INFO] [1710240263.714891318] [robot_state_publisher]: got segment robotiq_85_left_finger_tip_link
[robot_state_publisher-3] [INFO] [1710240263.714899085] [robot_state_publisher]: got segment robotiq_85_left_inner_knuckle_link
[robot_state_publisher-3] [INFO] [1710240263.714906223] [robot_state_publisher]: got segment robotiq_85_left_knuckle_link
[robot_state_publisher-3] [INFO] [1710240263.714913488] [robot_state_publisher]: got segment robotiq_85_right_finger_link
[robot_state_publisher-3] [INFO] [1710240263.714921052] [robot_state_publisher]: got segment robotiq_85_right_finger_tip_link
[robot_state_publisher-3] [INFO] [1710240263.714928568] [robot_state_publisher]: got segment robotiq_85_right_inner_knuckle_link
[robot_state_publisher-3] [INFO] [1710240263.714935950] [robot_state_publisher]: got segment robotiq_85_right_knuckle_link
[robot_state_publisher-3] [INFO] [1710240263.714943704] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-3] [INFO] [1710240263.714950948] [robot_state_publisher]: got segment spherical_wrist_1_link
[robot_state_publisher-3] [INFO] [1710240263.714958836] [robot_state_publisher]: got segment spherical_wrist_2_link
[robot_state_publisher-3] [INFO] [1710240263.714966297] [robot_state_publisher]: got segment world
[ros2_control_node-5] [INFO] [1710240263.726391030] [controller_manager]: Subscribing to '~/robot_description' topic for robot description file.
[ros2_control_node-5] [INFO] [1710240263.728860363] [controller_manager]: update rate is 1000 Hz
[ros2_control_node-5] [INFO] [1710240263.728981494] [controller_manager]: RT kernel is recommended for better performance
[ros2_control_node-5] [INFO] [1710240263.729455679] [controller_manager]: Received robot description file.
[ros2_control_node-5] [INFO] [1710240263.729743812] [resource_manager]: Loading hardware 'KortexMultiInterfaceHardware' 
[ros2_control_node-5] [INFO] [1710240263.730274349] [resource_manager]: Initialize hardware 'KortexMultiInterfaceHardware' 
[ros2_control_node-5] [WARN] [1710240263.730290356] [mock_generic_system]: Parameter 'fake_sensor_commands' has been deprecated from usage. Use'mock_sensor_commands' instead.
[ros2_control_node-5] [WARN] [1710240263.730302076] [mock_generic_system]: Parsing of optional initial interface values failed or uses a deprecated format. Add initial values for every state interface in the ros2_control.xacro. For example: 
[ros2_control_node-5] <state_interface name="velocity"> 
[ros2_control_node-5]   <param name="initial_value">0.0</param> 
[ros2_control_node-5] </state_interface>
[ros2_control_node-5] [INFO] [1710240263.730309566] [resource_manager]: Successful initialization of hardware 'KortexMultiInterfaceHardware'
[ros2_control_node-5] [INFO] [1710240263.730324238] [resource_manager]: Loading hardware 'RobotiqGripperHardwareInterface' 
[ros2_control_node-5] [INFO] [1710240263.730331867] [resource_manager]: Initialize hardware 'RobotiqGripperHardwareInterface' 
[ros2_control_node-5] [WARN] [1710240263.730337605] [mock_generic_system]: Parameter 'fake_sensor_commands' has been deprecated from usage. Use'mock_sensor_commands' instead.
[... cut ...]
[rviz2-1] [INFO] [1710240272.901257916] [rviz2.planning_scene_monitor]: Failed to call service get_planning_scene, have you launched move_group or called psm.providePlanningSceneService()?
[rviz2-1] [INFO] [1710240272.901300927] [rviz2.motion_planning_frame]: group gripper
[rviz2-1] [INFO] [1710240272.901305756] [rviz2.motion_planning_frame]: Constructing new MoveGroup connection for group 'gripper' in namespace ''
and the rviz interface show that 
[image]

`does it related to  the moveit2 installation method of binary.i install moveit2 by binary .

closed 2024-03-25 by yangming517

Mactarvish, 2024-07-25:
Same error (havn't try https://github.com/moveit/moveit2_tutorials/issues/719 mentioned above yet) but I checked the source code and there're exactly `trajectory` `description` intead of `trajectory_` `descrition_` , why compiler found `xx_` not `xx` ? Cannot understand