moveit/moveit2_tutorials #881: MTC Task Execution Failure - Moveit2 Pick n Place MTC demo

trentatsa, 2024-03-11:
### Description
When running the moveit2_tutorials pick and place demo, the plan succeeds however the execution fails. The reason behind this is detection of collision 'object' and 'panda_leftfinger' duing the " approach object" trajectory execution. The reason behind this is that after the gripper opens in  the demo, its seems like the trajectory execution does not use the latest planning scene i.e. it uses the planning scene when the gripper is still closed on startup. If I set the initail postion of the gripper to be open the demo pick and place runs successfully i.e. during the "approach object" and "retreat after place" the panda_leftfinger and panda_rightfinger are in the open position as seen by the trajectory execution. Im suspicious that it may be something to do with the ExecuteTaskSolutionCapability of move_group.

### Your environment
* ROS Distro: Iron
* OS Version: Ubuntu 22.04

### Steps to reproduce
add allow_nonzero_velocity_at_trajectory_end: true to the ros_controllers.yaml of moveit_resources_panda_moveit_config so that it as below. This is done otherwise you get the error: 
```
panda_arm_controller:
  ros__parameters:
    allow_nonzero_velocity_at_trajectory_end: true
    command_interfaces:
      - position
    state_interfaces:
      - position
      - velocity
    joints:
      - panda_joint1
      - panda_joint2
      - panda_joint3
      - panda_joint4
      - panda_joint5
      - panda_joint6
      - panda_joint7

panda_hand_controller:
  ros__parameters:
    joint: panda_finger_joint1

```
This is done otherwise you get the following error in control:
```
[ros2_control_node-5] [ERROR] [1710009229.370527846] [panda_arm_controller]: Velocity of last trajectory point of joint panda_joint1 is not zero: 0.002099733316440
[move_group-4] [INFO] [1710009229.370758219] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: panda_arm_controller started execution
[move_group-4] [WARN] [1710009229.370770581] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Goal request rejected
[move_group-4] [ERROR] [1710009229.370775217] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Goal was rejected by server
[move_group-4] [ERROR] [1710009229.370808155] [move_group.moveit.trajectory_execution_manager]: Failed to send trajectory part 1 of 1 to controller panda_arm_controller
[move_group-4] [INFO] [1710009229.370833203] [move_group.moveit.trajectory_execution_manager]: Completed trajectory execution with status ABORTED ...
```

Run the following: 
ros2 launch moveit2_tutorials mtc_demo.launch.py

and then run:
ros2 launch moveit2_tutorials pick_place_demo.launch.py

Execute any of the solutions:
[image]

### Expected behaviour
The robot should execute the pick and place :) 

### Actual behaviour
The task execution fails on " approach object" task component due to collision detection.

[image]

### Backtrace or Console output
```
[INFO] [launch]: All log files can be found below /home/trentpc/.ros/log/2024-03-09-19-25-56-756523-trentpc-199233
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [199236]
[INFO] [static_transform_publisher-2]: process started with pid [199238]
[INFO] [robot_state_publisher-3]: process started with pid [199240]
[INFO] [move_group-4]: process started with pid [199242]
[INFO] [ros2_control_node-5]: process started with pid [199246]
[INFO] [ros2 run controller_manager spawner panda_arm_controller-6]: process started with pid [199268]
[INFO] [ros2 run controller_manager spawner panda_hand_controller-7]: process started with pid [199273]
[... 232 lines cut ...]
[ros2_control_node-5] [INFO] [1710008767.270263887] [panda_arm_controller]: Goal reached, success!
[move_group-4] [INFO] [1710008767.283805301] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Controller 'panda_arm_controller' successfully finished
[move_group-4] [INFO] [1710008767.284193584] [move_group.moveit.add_time_optimal_parameterization]: Trajectory component '4/19' is invalid
[move_group-4] [INFO] [1710008767.284244942] [move_group.moveit.moveit_collision_detection_fcl]: Found a contact between 'object' (type 'Object') and 'panda_leftfinger' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[move_group-4] [INFO] [1710008767.284258583] [move_group.moveit.moveit_collision_detection_fcl]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[move_group-4] [INFO] [1710008767.284267993] [move_group.moveit.add_time_optimal_parameterization]: Upcoming trajectory component '4/19' is invalid
[move_group-4] [INFO] [1710008767.284283928] [move_group.moveit.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-4] [INFO] [1710008767.284300922] [move_group.moveit.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-4] [INFO] [1710008767.284427397] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: sending trajectory to panda_arm_controller
[ros2_control_node-5] [INFO] [1710008767.284600531] [panda_arm_controller]: Received new action goal
[ros2_control_node-5] [INFO] [1710008767.284630435] [panda_arm_controller]: Accepted new action goal
[move_group-4] [INFO] [1710008767.284758896] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: panda_arm_controller started execution
[move_group-4] [INFO] [1710008767.284769404] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Goal request accepted!
[move_group-4] [INFO] [1710008767.291016433] [move_group.moveit.add_time_optimal_parameterization]: Trajectory component '4/19' is invalid
[move_group-4] [INFO] [1710008767.291086856] [move_group.moveit.moveit_collision_detection_fcl]: Found a contact between 'object' (type 'Object') and 'panda_leftfinger' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[move_group-4] [INFO] [1710008767.291099803] [move_group.moveit.moveit_collision_detection_fcl]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[move_group-4] [INFO] [1710008767.291137571] [move_group.moveit.add_time_optimal_parameterization]: Trajectory component '4/19' is invalid after scene update
[move_group-4] [INFO] [1710008767.291145811] [move_group.moveit.add_time_optimal_parameterization]: Stopping execution because the path to execute became invalid(probably the environment changed)
[move_group-4] [INFO] [1710008767.291178856] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Cancelling execution for panda_arm_controller
[ros2_control_node-5] [INFO] [1710008767.291491290] [panda_arm_controller]: Got request to cancel goal
[ros2_control_node-5] [INFO] [1710008767.291531323] [panda_arm_controller]: Canceling active action goal because cancel callback received.
[move_group-4] [INFO] [1710008767.291789315] [move_group.moveit.trajectory_execution_manager]: Stopped trajectory execution.
[move_group-4] [INFO] [1710008767.335170954] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Controller 'panda_arm_controller' successfully finished
[move_group-4] [INFO] [1710008767.335222492] [move_group.moveit.trajectory_execution_manager]: Completed trajectory execution with status PREEMPTED ...
```

sea-bass, 2024-03-11:
I think I was able to reproduce this issue, have you recently checked out the MoveIt 2 tutorials?

As shown in https://github.com/ros-planning/moveit_task_constructor/issues/540, I suspect there might have been some recent breaking changes in the `moveit_task_constructor` repo.

On your system could you:

1. Go to the `src/moveit_task_constructor` folder in your repo
2. Do `git checkout ros2-old`
3. Rebuild your workspace
4. Try again

And let us know if that resolves the issue?

trentatsa, 2024-03-11:
Hi @sea-bass , I have tested it out. Still the same issue. With regards to the extra shape I took a screenshot from another demo (moveit_task_contructor pick_place demo) but the issue is the same. 

[image]

Based on https://github.com/ros-planning/moveit_task_constructor/issues/540 I would say that the issue is based on ExecuteTaskSolutionCapability plugin. More specifically it seems like after the gripper is opened,  ExecuteTaskSolutionCapability plugin is causing movegroup to not get the latest joint states of the gripper or hand and hence detects a false collision. Like I said if i make the initial joint position of the gripper open, the task execution is successful. You can replicate this by changing changing the contents of pand_hand.ros2_control.xacro to the following:

```
<?xml version="1.0"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro">

    <xacro:macro name="panda_hand_ros2_control" params="name prefix">
        <ros2_control name="${name}" type="system">
            <hardware>
                <plugin>mock_components/GenericSystem</plugin>
            </hardware>
            <joint name="${prefix}panda_finger_joint1">
                <command_interface name="position" />
                <state_interface name="position">
                  <param name="initial_value">0.035</param>
                </state_interface>
                <state_interface name="velocity">
                    <param name="initial_value">0.0</param>
                </state_interface>
            </joint>
            <joint name="${prefix}panda_finger_joint2">
                <param name="mimic">${prefix}panda_finger_joint1</param>
                <param name="multiplier">1</param>
                <command_interface name="position" />
                <state_interface name="position">
                  <param name="initial_value">0.035</param>
                </state_interface>
                <state_interface name="velocity">
                    <param name="initial_value">0.0</param>
                </state_interface>
            </joint>
        </ros2_control>
    </xacro:macro>

</robot>
```

This will start the gripper at "open state" and from that point on no matter what happen to the state of the gripper, movegroup sees it as this position/state.

I think @henningkayser should be able to tell us whats going on here.

rhaschke, 2024-03-11:
This might be related to https://github.com/ros-planning/moveit/pull/3538, which I strongly recommended for porting to ROS2.

rhaschke, 2024-03-11:
> I've just added the `allow_nonzero_velocity_at_trajectory_end: true`

Nonzero velocities at the end of the trajectory are useful for servoing, but not for MTC: you want the robot to stop at a stage's end state. Otherwise end and new start will have different velocity vectors resulting in high jerk!

sea-bass, 2024-03-11:
> Nonzero velocities at the end of the trajectory are useful for servoing, but not for MTC: you want the robot to stop at a stage's end state. Otherwise end and new start will have different velocity vectors resulting in high jerk!

Yep, I had to add this in unrelated to this PR to address a Servo issue.

Unsure why this pick and place demo is trying to set nonzero end velocities though, as I agree all these planned motions should stop at rest.

trentatsa, 2024-03-11:
@sea-bass @rhaschke so we dealing with two issues here:
1) non zero velocitys at end of trajectory.
2) MTC Task execution not getting getting updated planning scene using execution.

rhaschke, 2024-03-11:
> Unsure why this pick and place demo is trying to set nonzero end velocities though

Obviously, the time parameterization is broken.