moveit/moveit2_tutorials #884: ros2 launch moveit2_tutorials motion_planning_python_api_tutorial.launch.py

kavikode, 2024-03-17:
### Description
ros2 launch moveit2_tutorials motion_planning_python_api_tutorial.launch.py 
file 'motion_planning_python_api_tutorial.launch.py' was not found in the share directory of package 'moveit2_tutorials' which is at '/home/ubuntu/moveit2_tutorials_ws/install/moveit2_tutorials/share/moveit2_tutorials'

Overview of your issue here.

### Your environment
* ROS Distro: [Humble]
* OS Version:Ubuntu 22.04
* Source build humble branch

In the CMakeLists.txt file, the line code "add_subdirectory(doc/examples/motion_planning_api)"

is already included.

Would like to use Python. Any suggestions to resolve?

followthwhiterabbit, 2024-06-21:
I have the same problem, when i am trying to launch moveit2_tutorials, i am getting the following error 

`Package 'moveit2_tutorials' not found: "package 'moveit2_tutorials' not found, searching: ['/home/kaan/ws_moveit2/install/robotiq_hardware_tests', '/home/kaan/ws_moveit2/install/robotiq_driver', '/home/kaan/ws_moveit2/install/serial', '/home/kaan/ws_moveit2/install/moveit_visual_tools', '/home/kaan/ws_moveit2/install/rviz_visual_tools', '/home/kaan/ws_moveit2/install/rviz_marker_tools', '/home/kaan/ws_moveit2/install/kortex_bringup', '/home/kaan/ws_moveit2/install/kortex_description', '/home/kaan/ws_moveit2/install/robotiq_description', '/home/kaan/ws_moveit2/install/robotiq_controllers', '/home/kaan/ws_moveit2/install/pilz_industrial_motion_planner_testutils', '/home/kaan/ws_moveit2/install/moveit_runtime', '/home/kaan/ws_moveit2/install/moveit', '/home/kaan/ws_moveit2/install/moveit_planners', '/home/kaan/ws_moveit2/install/pilz_industrial_motion_planner', '/home/kaan/ws_moveit2/install/pick_ik', '/home/kaan/ws_moveit2/install/moveit_ros_control_interface', '/home/kaan/ws_moveit2/install/moveit_plugins', '/home/kaan/ws_moveit2/install/moveit_simple_controller_manager', '/home/kaan/ws_moveit2/install/moveit_setup_assistant', '/home/kaan/ws_moveit2/install/moveit_setup_srdf_plugins', '/home/kaan/ws_moveit2/install/moveit_setup_core_plugins', '/home/kaan/ws_moveit2/install/moveit_setup_controllers', '/home/kaan/ws_moveit2/install/moveit_setup_app_plugins', '/home/kaan/ws_moveit2/install/moveit_setup_framework', '/home/kaan/ws_moveit2/install/moveit_servo', '/home/kaan/ws_moveit2/install/moveit_ros', '/home/kaan/ws_moveit2/install/moveit_ros_visualization', '/home/kaan/ws_moveit2/install/moveit_hybrid_planning', '/home/kaan/ws_moveit2/install/moveit_ros_planning_interface', '/home/kaan/ws_moveit2/install/moveit_ros_benchmarks', '/home/kaan/ws_moveit2/install/moveit_ros_warehouse', '/home/kaan/ws_moveit2/install/moveit_ros_robot_interaction', '/home/kaan/ws_moveit2/install/moveit_ros_perception', '/home/kaan/ws_moveit2/install/moveit_resources_prbt_pg70_support', '/home/kaan/ws_moveit2/install/moveit_resources_prbt_moveit_config', '/home/kaan/ws_moveit2/install/moveit_ros_move_group', '/home/kaan/ws_moveit2/install/moveit_planners_ompl', '/home/kaan/ws_moveit2/install/moveit_ros_planning', '/home/kaan/ws_moveit2/install/moveit_ros_occupancy_map_monitor', '/home/kaan/ws_moveit2/install/moveit_resources_prbt_ikfast_manipulator_plugin', '/home/kaan/ws_moveit2/install/moveit_planners_chomp', '/home/kaan/ws_moveit2/install/moveit_kinematics', '/home/kaan/ws_moveit2/install/moveit_chomp_optimizer_adapter', '/home/kaan/ws_moveit2/install/chomp_motion_planner', '/home/kaan/ws_moveit2/install/moveit_core', '/home/kaan/ws_moveit2/install/generate_parameter_module_example', '/home/kaan/ws_moveit2/install/generate_parameter_library_example', '/home/kaan/ws_moveit2/install/cmake_generate_parameter_module_example', '/home/kaan/ws_moveit2/install/generate_parameter_library', '/home/kaan/ws_moveit2/install/parameter_traits', '/home/kaan/ws_moveit2/install/moveit_task_constructor_msgs', '/home/kaan/ws_moveit2/install/moveit_resources_prbt_support', '/home/kaan/ws_moveit2/install/moveit_resources', '/home/kaan/ws_moveit2/install/moveit_resources_pr2_description', '/home/kaan/ws_moveit2/install/moveit_resources_panda_moveit_config', '/home/kaan/ws_moveit2/install/moveit_resources_panda_description', '/home/kaan/ws_moveit2/install/moveit_resources_fanuc_moveit_config', '/home/kaan/ws_moveit2/install/moveit_resources_fanuc_description', '/home/kaan/ws_moveit2/install/moveit_msgs', '/home/kaan/ws_moveit2/install/moveit_configs_utils', '/home/kaan/ws_moveit2/install/moveit_common', '/home/kaan/ws_moveit2/install/kortex_driver', '/home/kaan/ws_moveit2/install/kortex_api', '/home/kaan/ws_moveit2/install/generate_parameter_library_py', '/opt/ros/humble']"
`

Benned-H, 2024-08-05:
I'm having the same issue, where `motion_planning_python_api_tutorial.launch.py` is not found when I run the first command in the [Motion Planning Python API](https://moveit.picknik.ai/main/doc/examples/motion_planning_python_api/motion_planning_python_api_tutorial.html) tutorial. Like @kavikode, I'm running ROS 2 Humble on Ubuntu 22.04, with `moveit2_tutorials` built from source on the `humble` branch.

It looks like the `doc/examples/motion_planning_python_api` folder doesn't exist on the `humble` branch. Because other tutorials *have* worked for the Humble install, this issue could be resolved by documenting which ROS 2 versions are/aren't supported by the Python API tutorial.

sea-bass, 2024-08-10:
As @Benned-H pointed out, the MoveIt Python bindings are not available on ROS 2 Humble. Only Iron and later.

So you have a few options (in order of difficulty):
1. Upgrade to Iron or later
2. Build the `main` branch of MoveIt and these tutorials, noting that you will need to manually modify some transitive Gazebo dependencies that are not available in Iron or earlier
3. Contribute by backporting the Python bindings to Humble! (this will be a lot of effort)

So, I would recommend either 1. or 2.

closed 2024-08-10 by sea-bass

kavikode, 2024-08-21:
Thank you for the message. 

For #2, how do we find the specific transitive Gazebo properties to manually modify for the MoveIt Python bindings to work in ROS2 Humble?

Specifically, which specific transitive Gazebo properties are necessary for Tiago, https://github.com/pal-robotics/tiago_simulation?

Thank you