moveit/moveit2_tutorials #891: Importing moveit.planning or etc. libraries

UgurYakut, 2024-04-05:
### Description
Hello i want to write a pick and place task, i looked the tutorials python pick and place. How i can find the moveit.planning library or etc. I installed binary Moveit2. I builded my robot and i can plan and execute via Rviz. How can i do with scripts? 

### Your environment
* ROS Distro: Humble
* OS Version:Ubuntu 22.04
* Binary build

### Expected behaviour
Detecting library.
### Backtrace or Console output
Traceback (most recent call last):
  File "/home/u/arm/dev_ws/src/paket/scripts/plan_execute.py", line 2, in <module>
    from moveit.planning import MoveItPy
ModuleNotFoundError: No module named 'moveit'

tanmayyb, 2024-04-11:
Same issue here. Any updates on this?
Best,

UgurYakut, 2024-04-17:
> Same issue here. Any updates on this? Best,

Still same issue,did you find any thing?

Jens-Yu, 2024-05-19:
Same issue，did you solve the problem?

tanmayyb, 2024-05-22:
@Jens-Yu , moveit_py package  is needed for scripting in Python. But:
1. moveit_py package was NOT available at Humble branch [[link](https://github.com/ros-planning/moveit2_tutorials/issues/715)][[link](https://github.com/ros-planning/moveit2/tree/humble)]
2. moveit2 [mainline](https://github.com/ros-planning/moveit2/tree/main) is not stable [[link](https://github.com/ros-planning/moveit2/tree/main)]. 

Therefore, I switched from Ubuntu20.04 w/ ROS Humble to Ubuntu22.04 w/ ROS Iron and I was able to use these tutorials. Hope it helps.

tanmayyb, 2024-05-22:
+@UgurYakut

mikeferguson, 2024-12-18:
That is correct, the python interfaces are newer than Humble and will not be backported.

closed 2024-12-18 by mikeferguson