moveit/moveit2_tutorials #899: High RAM usage during colcon build

matteogrechi, 2024-04-30:
### Description

I'm working in WSL with 3.74GiB of RAM and 1GiB of swap (Neither I cannot add more RAM to the machine nor increase the swap partition).
Whenever I try to compile the source the compiler gets killed, e.g.:
```
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [src/CMakeFiles/moveit_task_constructor_core.dir/build.make:174: src/CMakeFiles/moveit_task_constructor_core.dir/storage.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:256: src/CMakeFiles/moveit_task_constructor_core.dir/all] Error 2
```
I checked and both the RAM and the swap gets fully saturated.
Is there a way to tell the compiler to launch less threads (`--parallel-workers 1` it's not enough)?
Is it possible to provide a version working with pre-build binaries?

### Environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04 (inside WSL 2)
* Source, commit hash 20be39797b17e460fda48975a0a8ee25f8ad87ac

### Steps to reproduce
Run `colcon build --mixin release --parallel-workers 1`

Thank you very much!

matteogrechi, 2024-04-30:
Possible solution to question number 1:

Before running colcon build, set the MAKEFLAGS environment variable to `j1`:

```
$ export MAKEFLAGS=-j1
$ colcon build --mixin release --parallel-workers 1
```

Source:
- [Set number of threads used by make permanently](https://stackoverflow.com/questions/41098627/set-number-of-threads-used-by-make-permanently)

closed 2024-08-13 by sea-bass

Ken-Andre, 2024-09-20:
Hi, 
I'm new to ROS and also hit myself on this issues that is not yet modified in the docs.
[the colcon build ros2 humble](https://docs.ros.org/en/humble/Tutorials/Beginner-Client-Libraries/Colcon-Tutorial.html).
I've upgrade also my memory up to 16Gb and have the last intell processor with 12 thread, also a Dedicated Nvidia Graphics Card.