moveit/moveit2_tutorials #910: Parameter Type Exception in demo.launch.python

Santi-13, 2024-05-31:
### Description

I'm following the MoveIt2 tutorials and have encountered an issue when running the `demo.launch.python` file from the `moveit2_tutorials` package. Despite following all the steps in the quickstart guide, I receive a warning in RViz about "Requesting initial scene failed". The terminal log indicates a problem with a parameter type mismatch.

### Your environment
* ROS Distro: Iron
* OS Version: Ubuntu 22.04
* Source build

### Steps to reproduce
Just followed the instructions on the MoveIt Quickstart in RViz(https://moveit.picknik.ai/main/doc/tutorials/quickstart_in_rviz/quickstart_in_rviz_tutorial.html#moveit-quickstart-in-rviz) tutorial up to step 1. Where I noticed that the planning scene reported "Requesting initial scene failed"

### Backtrace or Console output
lt logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [43563]
[INFO] [static_transform_publisher-2]: process started with pid [43565]
[INFO] [robot_state_publisher-3]: process started with pid [43567]
[INFO] [move_group-4]: process started with pid [43569]
[INFO] [ros2_control_node-5]: process started with pid [43571]
[INFO] [spawner-6]: process started with pid [43573]
[INFO] [spawner-7]: process started with pid [43575]
[INFO] [spawner-8]: process started with pid [43577]
[static_transform_publisher-2] [INFO] [1717116430.267242645] [static_transform_publisher]: Spinning until stopped - publishing transform
[static_transform_publisher-2] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-2] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-2] from 'world' to 'base_link'
[ros2_control_node-5] [INFO] [1717116430.321668594] [controller_manager]: Subscribing to '/robot_description' topic for robot description.
[ros2_control_node-5] [INFO] [1717116430.321965845] [controller_manager]: update rate is 1000 Hz
[ros2_control_node-5] [INFO] [1717116430.322171145] [controller_manager]: RT kernel is recommended for better performance
[robot_state_publisher-3] [INFO] [1717116430.339275418] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1717116430.339402577] [robot_state_publisher]: got segment bracelet_link
[robot_state_publisher-3] [INFO] [1717116430.339415228] [robot_state_publisher]: got segment end_effector_link
[robot_state_publisher-3] [INFO] [1717116430.339421988] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-3] [INFO] [1717116430.339427778] [robot_state_publisher]: got segment half_arm_1_link
[robot_state_publisher-3] [INFO] [1717116430.339433451] [robot_state_publisher]: got segment half_arm_2_link
[robot_state_publisher-3] [INFO] [1717116430.339438964] [robot_state_publisher]: got segment robotiq_85_base_link
[robot_state_publisher-3] [INFO] [1717116430.339444808] [robot_state_publisher]: got segment robotiq_85_left_finger_link
[robot_state_publisher-3] [INFO] [1717116430.339450268] [robot_state_publisher]: got segment robotiq_85_left_finger_tip_link
[robot_state_publisher-3] [INFO] [1717116430.339455829] [robot_state_publisher]: got segment robotiq_85_left_inner_knuckle_link
[robot_state_publisher-3] [INFO] [1717116430.339461394] [robot_state_publisher]: got segment robotiq_85_left_knuckle_link
[robot_state_publisher-3] [INFO] [1717116430.339466937] [robot_state_publisher]: got segment robotiq_85_right_finger_link
[robot_state_publisher-3] [INFO] [1717116430.339472370] [robot_state_publisher]: got segment robotiq_85_right_finger_tip_link
[robot_state_publisher-3] [INFO] [1717116430.339477875] [robot_state_publisher]: got segment robotiq_85_right_inner_knuckle_link
[robot_state_publisher-3] [INFO] [1717116430.339483219] [robot_state_publisher]: got segment robotiq_85_right_knuckle_link
[robot_state_publisher-3] [INFO] [1717116430.339488760] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-3] [INFO] [1717116430.339494496] [robot_state_publisher]: got segment spherical_wrist_1_link
[robot_state_publisher-3] [INFO] [1717116430.339500131] [robot_state_publisher]: got segment spherical_wrist_2_link
[robot_state_publisher-3] [INFO] [1717116430.339518428] [robot_state_publisher]: got segment world
[ros2_control_node-5] [INFO] [1717116430.339811378] [controller_manager]: Received robot description from topic.
[ros2_control_node-5] [INFO] [1717116430.340566566] [resource_manager]: Loading hardware 'KortexMultiInterfaceHardware' 
[ros2_control_node-5] [INFO] [1717116430.342048482] [resource_manager]: Initialize hardware 'KortexMultiInterfaceHardware' 
[ros2_control_node-5] [WARN] [1717116430.342091176] [mock_generic_system]: Parameter 'fake_sensor_commands' has been deprecated from usage. Use'mock_sensor_commands' instead.
[ros2_control_node-5] [WARN] [1717116430.342141091] [mock_generic_system]: Parsing of optional initial interface values failed or uses a deprecated format. Add initial values for every state interface in the ros2_control.xacro. For example: 
[ros2_control_node-5] <state_interface name="velocity"> 
[ros2_control_node-5]   <param name="initial_value">0.0</param> 
[ros2_control_node-5] </state_interface>
[ros2_control_node-5] [INFO] [1717116430.342161246] [resource_manager]: Successful initialization of hardware 'KortexMultiInterfaceHardware'
[ros2_control_node-5] [INFO] [1717116430.342197198] [resource_manager]: Loading hardware 'RobotiqGripperHardwareInterface' 
[ros2_control_node-5] [INFO] [1717116430.342214023] [resource_manager]: Initialize hardware 'RobotiqGripperHardwareInterface' 
[ros2_control_node-5] [WARN] [1717116430.342231196] [mock_generic_system]: Parameter 'fake_sensor_commands' has been deprecated from usage. Use'mock_sensor_commands' instead.
[ros2_control_node-5] [INFO] [1717116430.342264075] [resource_manager]: Successful initialization of hardware 'RobotiqGripperHardwareInterface'
[ros2_control_node-5] [INFO] [1717116430.342356900] [resource_manager]: 'configure' hardware 'RobotiqGripperHardwareInterface' 
[ros2_control_node-5] [INFO] [1717116430.342372160] [resource_manager]: Successful 'configure' of hardware 'RobotiqGripperHardwareInterface'
[ros2_control_node-5] [INFO] [1717116430.342389600] [resource_manager]: 'activate' hardware 'RobotiqGripperHardwareInterface' 
[ros2_control_node-5] [INFO] [1717116430.342401851] [resource_manager]: Successful 'activate' of hardware 'RobotiqGripperHardwareInterface'
[ros2_control_node-5] [INFO] [1717116430.342414096] [resource_manager]: 'configure' hardware 'KortexMultiInterfaceHardware' 
[ros2_control_node-5] [INFO] [1717116430.342424631] [resource_manager]: Successful 'configure' of hardware 'KortexMultiInterfaceHardware'
[ros2_control_node-5] [INFO] [1717116430.342444807] [resource_manager]: 'activate' hardware 'KortexMultiInterfaceHardware' 
[ros2_control_node-5] [INFO] [1717116430.342455233] [resource_manager]: Successful 'activate' of hardware 'KortexMultiInterfaceHardware'
[ros2_control_node-5] [INFO] [1717116430.438406712] [controller_manager]: Received robot description from topic.
[... 71 lines cut ...]
[ERROR] [move_group-4]: process has died [pid 43569, exit code -6, cmd '/home/sanmaster/ws_moveit/install/moveit_ros_move_group/lib/moveit_ros_move_group/move_group --ros-args --params-file /tmp/launch_params_q793jq7l'].
[rviz2-1] [ERROR] [1717116434.373774845] [rviz2.moveit.ros.motion_planning_frame]: Action server: /recognize_objects not available
[rviz2-1] [INFO] [1717116434.396707512] [rviz2.moveit.ros.motion_planning_frame]: MoveGroup namespace changed: / -> . Reloading params.
[rviz2-1] [INFO] [1717116434.439370432] [rviz2.moveit.ros.rdf_loader]: Loaded robot model in 0.0342068 seconds
[rviz2-1] [INFO] [1717116434.439454770] [rviz2.moveit.core.robot_model]: Loading robot model 'gen3'...
[rviz2-1] [INFO] [1717116434.439484754] [rviz2.moveit.core.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[rviz2-1] [INFO] [1717116434.851120679] [rviz2.moveit.ros.rdf_loader]: Loaded robot model in 0.00390296 seconds
[rviz2-1] [INFO] [1717116434.851172141] [rviz2.moveit.core.robot_model]: Loading robot model 'gen3'...
[rviz2-1] [INFO] [1717116434.851185580] [rviz2.moveit.core.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[rviz2-1] [INFO] [1717116434.997956622] [rviz2.moveit.kinematics.kdl_kinematics_plugin]: Joint weights for group 'manipulator': 1 1 1 1 1 1 1
[... cut ...]
[rviz2-1] [INFO] [1717116440.385022304] [rviz2.moveit.ros.planning_scene_monitor]: Failed to call service get_planning_scene, have you launched move_group or called psm.providePlanningSceneService()?
[rviz2-1] [INFO] [1717116440.416608265] [rviz2.moveit.ros.planning_scene_monitor]: Failed to call service get_planning_scene, have you launched move_group or called psm.providePlanningSceneService()?
[rviz2-1] [INFO] [1717116440.416694902] [rviz2.moveit.ros.motion_planning_frame]: group gripper
[rviz2-1] [INFO] [1717116440.416704810] [rviz2.moveit.ros.motion_planning_frame]: Constructing new MoveGroup connection for group 'gripper' in namespace ''

Any help would be greatly appreciated!

Kaiser1401, 2024-06-06:
`[move_group-4] what(): expected [string_array] got [string]` seems to be because only ROS humble expects the parameter actually to be astring (not a string array)
--> Look at the [ompl_planning.yaml](https://github.com/Kinovarobotics/ros2_kortex/blob/main/kortex_moveit_config/kinova_gen3_7dof_robotiq_2f_85_moveit_config/config/ompl_planning.yaml#L7) and change the `request_adapters` to be a List accoring to the header meassage in the same file.

Had the same (and some) problems running the demo on Jazzy

sea-bass, 2024-08-02:
Should be fixed by https://github.com/moveit/moveit2_tutorials/pull/921 -- please try it out!

closed 2024-08-02 by sea-bass