moveit/moveit2_tutorials #912: Didn't receive robot state (joint angles) with recent timestamp within 1.000000 seconds

SyZbidi, 2024-06-06:
### Description

Followed the `motion_planning_python_api` tutorial example, the launch file is almost the same, just added `planning_pipelines` to moveit_config, and my `MoveitPy node` is only trying to plan and execute one example of named_target. I don't understand Python launch files much, and also new to ROS2 unfortunately. So I can't get to locate the reason behind having this error

` Didn't receive robot state (joint angles) with recent timestamp within 1.000000 seconds`

### Your environment
* ROS Distro: Humble
* OS Version: e.g. Ubuntu 22.04
* build: Source (main)

### Steps to reproduce

 I made my `moveit_config` package with `moveit_setup_assistant`. Then I only changed the `acceleration_limits` to True and set a value >0. After that I created a launch file similar to [motion_planning_python_api_tutorial.launch.py](https://github.com/moveit/moveit2_tutorials/blob/1834b79c14f2a068f7fbee6e3cbfb06e30a27f5c/doc/examples/motion_planning_python_api/launch/motion_planning_python_api_tutorial.launch.py)  with `moveit_utils` being my node

#### launch.py

```
"""
A launch file for running the motion planning python api tutorial
"""
import os
from ament_index_python.packages import get_package_share_directory
from launch import LaunchDescription
from launch_ros.actions import Node
from launch.actions import DeclareLaunchArgument, ExecuteProcess, IncludeLaunchDescription
from launch.substitutions import LaunchConfiguration
from moveit_configs_utils import MoveItConfigsBuilder
from launch.launch_description_sources import PythonLaunchDescriptionSource

def generate_launch_description():
    moveit_config = (
        MoveItConfigsBuilder(
            robot_name="thermoplast_abb_irb4600_camera_cell", package_name="abb_irb4600_webcam_moveit_config"
        )
        .robot_description(file_path="config/thermoplast_abb_irb4600_camera_cell.urdf.xacro")
        .trajectory_execution(file_path="config/moveit_controllers.yaml")
        .moveit_cpp(
            file_path=get_package_share_directory("abb_irb4600_webcam_moveit_config")
            + "/config/planning.yaml"
        )
        .to_moveit_configs()
    )

    example_file = DeclareLaunchArgument(
        "example_file",
        default_value="moveit_utils",
        description="Python API tutorial file name",
    )

    moveit_py_node = Node(
        name="moveit_utils",
        package="simple_motion_planner",
        executable=LaunchConfiguration("example_file"),
        output="both",
        parameters=[moveit_config.to_dict()],
    )

    rviz_config_file = os.path.join(
        get_package_share_directory("abb_irb4600_webcam_moveit_config"),
        "config",
        "moveit.rviz",
    )

    rviz_node = Node(
        package="rviz2",
        executable="rviz2",
        name="rviz2",
        output="log",
        arguments=["-d", rviz_config_file],
[... 325 lines cut ...]
[rviz2-4] [INFO] [1717702918.250131869] [rviz2.moveit.ros.robot_interaction]: No active joints or end effectors found for group 'manipulator'. Make sure that kinematics.yaml is loaded in this node's namespace.
[rviz2-4] [INFO] [1717702918.250898375] [rviz2.moveit.ros.robot_interaction]: No active joints or end effectors found for group 'manipulator'. Make sure that kinematics.yaml is loaded in this node's namespace.
[rviz2-4] [INFO] [1717702918.255510141] [rviz2.moveit.ros.motion_planning_frame]: group manipulator
[rviz2-4] [INFO] [1717702918.255528582] [rviz2.moveit.ros.motion_planning_frame]: Constructing new MoveGroup connection for group 'manipulator' in namespace ''
[rviz2-4] [INFO] [1717702918.271885302] [interactive_marker_display_94116680968320]: Sending request for interactive markers
[rviz2-4] [INFO] [1717702918.304191585] [interactive_marker_display_94116680968320]: Service response received for initialization
[moveit_utils-1] [INFO] [1717702918.877642308] [moveit_3672297761.moveit.ros.current_state_monitor]: Didn't receive robot state (joint angles) with recent timestamp within 1.000000 seconds. Requested time 1717702917.877563, but latest received state has time 0.000000.
[moveit_utils-1] Check clock synchronization if your are running ROS across multiple machines!
[moveit_utils-1] [WARN] [1717702918.877714628] [moveit_3672297761.moveit.ros.trajectory_execution_manager]: Failed to validate trajectory: couldn't receive full current joint state within 1s
[moveit_utils-1] [INFO] [1717702918.881850550] [moveit_3672297761.moveit.ros.moveit_cpp]: Deleting MoveItCpp
[moveit_utils-1] [INFO] [1717702918.899369883] [moveit_3672297761.moveit.ros.planning_scene_monitor]: Stopped publishing maintained planning scene.
[moveit_utils-1] [INFO] [1717702918.902589224] [moveit_3672297761.moveit.ros.planning_scene_monitor]: Stopping world geometry monitor
[moveit_utils-1] [INFO] [1717702918.904708542] [moveit_3672297761.moveit.ros.planning_scene_monitor]: Stopping planning scene monitor
[moveit_utils-1] Warning: class_loader.ClassLoader: SEVERE WARNING!!! Attempting to unload library while objects created by this loader exist in the heap! You should delete your objects before attempting to unload the library or destroying the ClassLoader. The library will NOT be unloaded.
[moveit_utils-1]          at line 127 in ./src/class_loader.cpp
[INFO] [moveit_utils-1]: process has finished cleanly [pid 234585]
[rviz2-4] [INFO] [1717702978.272028392] [rviz2.moveit.ros.move_group_interface]: Ready to take commands for planning group manipulator.
```

Am I missing parameters? I have no idea where to look to fix the `[rviz2-4] [ERROR] [1717702917.972281146] [rviz2.moveit.ros.trajectory_visualization]: No robot state or robot model loaded
` and `moveit_utils-1] [INFO] [1717702918.877642308] [moveit_3672297761.moveit.ros.current_state_monitor]: Didn't receive robot state (joint angles) with recent timestamp within 1.000000 seconds. Requested time 1717702917.877563, but latest received state has time 0.000000.`

By the way `use_sim_time` doesn't fix i, I already tried to add it to the node vut I'm l=not using simulation/gazebo

Thanks!

okvik, 2024-06-06:
Could be that your robot controller isn't up and publishing to `joint_states` topic by the time you are trying to plan. Seems plausible since you're launching everything from the same launch file and real robot drivers usually take a bit of time to establish connections to the hardware and get going.

You could easily test if this theory is correct by introducing a sleep of some duration before doing anything else in your program, just to give time for all the rest of the stuff in the launch file to come up. Alternatively you can subscribe to the `joint_states` topic yourself and wait until you get a first message (hint: this is more elegantly solved by liveliness QoS, which you may wanna look into at some date), at which point it may be safe to assume the controller / hardware is up and running and ready to receive commands. Pretty sure `ros2_control` provides a mechanism to establish the readyness of the controller / hardware, but I don't know whether MoveIt does anything with it -- I only worked with robots that don't use `ros2_control`.

SyZbidi, 2024-06-07:
> Sorry it's late and I overlooked that you are using FakeHardware so this is unlikely an issue, though you are free to try if sleep helps anyway. At least, confirm you are actually getting something published on the `joint_states` topic.

Your suggestion was actually helpful, thanks a lot! I let it sleep for 10 seconds and now I got to see the plan but no execution.

Also echoed the `/joint_states` and noticed that they're not being published so I added them to the launch file and finally got SUCCEEDED

```
[ros2_control_node-4] [INFO] [1717748497.252701086] [manipulator_controller]: Received new action goal
[ros2_control_node-4] [INFO] [1717748497.252793906] [manipulator_controller]: Accepted new action goal
[moveit_utils-1] [INFO] [1717748497.253029323] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: manipulator_controller started execution
[moveit_utils-1] [INFO] [1717748497.253073435] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Goal request accepted!
[ros2_control_node-4] [INFO] [1717748499.763253884] [manipulator_controller]: Goal reached, success!
[moveit_utils-1] [INFO] [1717748499.803428641] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Controller 'manipulator_controller' successfully finished
[moveit_utils-1] [INFO] [1717748500.052285755] [moveit_3805631290.moveit.ros.trajectory_execution_manager]: Completed trajectory execution with status SUCCEEDED ...
[moveit_utils-1] [INFO] [1717748500.060572846] [moveit_3805631290.moveit.ros.moveit_cpp]: Deleting MoveItCpp
[moveit_utils-1] [INFO] [1717748500.079781938] [moveit_3805631290.moveit.ros.planning_scene_monitor]: Stopped publishing maintained planning scene.
[moveit_utils-1] [INFO] [1717748500.080396127] [moveit_3805631290.moveit.ros.planning_scene_monitor]: Stopping world geometry monitor
[moveit_utils-1] [INFO] [1717748500.080647130] [moveit_3805631290.moveit.ros.planning_scene_monitor]: Stopping planning scene monitor
```

[image]

sjahr, 2024-06-07:
Sorry for the late reply, is your issue fixed now?

SyZbidi, 2024-06-07:
> Seems plausible since you're launching everything from the same launch file 

@okvik I actually wanted to split them I don't like the idea of launching the moveit setup and the `moveit_py` node from the same file even if it's for fakehardware, but I didn't know what should the node launch file have for parameters.

I'd appreciate if you have a suggestion but you already solved my main issue so thank you again for that :)

SyZbidi, 2024-06-07:
> Sorry for the late reply, is your issue fixed now?

Hi @sjahr, not really. there's something wrong in my launch file above. adding a sleep time and a `joint_state_publisher` node did help with executing the motion but the `/joint_states` topic is publishing all zeros.

So though the fix suggested by @okvik worked, it doesn't seem like the way to go

SyZbidi, 2024-06-07:
The solution was to add `joint_state_broadcaster` to `load_controllers` instead of `joint_state_publisher`. For some reason I removed that when I copied the tutorial launch files.

```
load_controllers = []
    for controller in [
        "manipulator_controller",
        "joint_state_broadcaster",
    ]:
        load_controllers += [
            ExecuteProcess(
                cmd=["ros2 run controller_manager spawner {}".format(controller)],
                shell=True,
                output="log",
            )
        ]
```

SyZbidi, 2024-06-07:
@sjahr this is off-topic but also related to my work, it would be helpful if you take a look at [this comment](https://github.com/moveit/moveit2/issues/2694#issuecomment-2149360967) when you have a chance. Thank you

okvik, 2024-06-07:
> @okvik I actually wanted to split them I don't like the idea of launching the moveit setup and the `moveit_py` node from the same file even if it's for fakehardware, but I didn't know what should the node launch file have for parameters.

The way I do it is indeed to split each major hardware component (like a robot) launch from the control program logic and any other supporting nodes such as rviz, potentially the HMI node, etc. Basically, split the system in a way that lets me rebuild / start / restart without having to reboot the entire system from scratch every time. For production I additionally wrap these launches in their own systemd units, which lets me deal with dependencies, starting the system at boot, have more meaningful logging experience, easily clean up broken process trees, etc.

devika-dudo, 2025-04-17:
Hi, I am getting the same issue , I have set use_sim_time to true for almost all nodes and my /joint_states -- anyone has any solutions?

JesusHdzzzz, 2025-05-09:
I also have the same issue, i technically have only like two days left to figure this out because my capstone class is about to end soon but I added use_sim_time: true to all my nodes and other packages. I've checked that ros2 param get /move_group use_sim_time equals true. I don't get zeroes when i run ros2 topic echo /joint_states --once, so idk what really is causing the issue in my case.