moveit/moveit2_tutorials #915: Update rolling CI job to use 24.04

sjahr, 2024-06-21:
Looks like the CI job is still using 22.04 

[image]

_Originally posted by @Abishalini in https://github.com/moveit/moveit2_tutorials/issues/913#issuecomment-2180882250_

stephanie-eng, 2024-07-25:
I wouldn't do this yet. We're using GitHub's runners, and using their `ubuntu-latest` (default) option gives 22.04. It's possible to switch to 24.04, but it's beta. I'd prefer to keep using `ubuntu-latest` and we'll move to 24.04 when it's out of beta and GitHub decides to make it the new default.

We run everything in Docker anyway, so the runner being 22.04 shouldn't hurt us.

See:
https://docs.github.com/en/actions/using-jobs/choosing-the-runner-for-a-job#standard-github-hosted-runners-for-public-repositories

Screenshot for folks viewing this issue after the GitHub link is updated:
[image]

130s, 2026-08-31:
In https://github.com/moveit/moveit2_tutorials/pull/1110/changes#diff-173f5db567f8459284c719180df8c27debad26456e63f98c5b87d94cf87e049bR19, 24.04 is skipped and upgraded to 26.04. Please report if there's any issue.

closed 2026-08-31 by 130s