moveit/moveit2_tutorials #918: Unauthorized from pulling Docker rolling-tutorial image from ghcr.io (GitHub container registry)

ukicomputers, 2024-07-03:
### Description

Hi! This is my first time to try MoveIt 2 for myself! As now, I use Docker for almost anything, because it keeps dependencies and software clean, working and OK like ROS 2. I started with tutorial on [How to Set Up MoveIt 2 Docker Containers in Ubuntu](https://moveit.picknik.ai/main/doc/how_to_guides/how_to_setup_docker_containers_in_ubuntu.html). I got the Docker Compose script, executed command `docker compose run ...`, and Docker started pulling from ghcr.io (GitHub container registry). After few seconds, I got error: `Head unauthorized` (please see console output). Did anyone encountered this problem, or it is just me? Thanks!

### Your environment
* ROS Distro: Rolling (but also, I tried with Jazzy, and it is not working)
* OS Version: Ubuntu 24.04 (codename **noble**)
* Using `moveit2_container` with image `rolling-tutorial` binary build from ghcr.io (Docker compose script)
* Rolling

### Steps to reproduce
```bash
wget https://raw.githubusercontent.com/ros-planning/moveit2_tutorials/main/.docker/docker-compose.yml
DOCKER_IMAGE=rolling-tutorial docker compose run --rm --name moveit2_container cpu
```

### Expected behaviour
While pulling image, I got `Error response from daemon: Head unauthorized` (see console output).

### Backtrace or Console output
When running `docker compose run` (see steps to reproduce):
```
[+] Running 1/1
 ✘ cpu Error Head "https://ghcr.io/v2/moveit/moveit2_tutorials/manifests/rolling-tutorial": unauthorized                                      1.8s 
Error response from daemon: Head "https://ghcr.io/v2/moveit/moveit2_tutorials/manifests/rolling-tutorial": unauthorized
```

ukicomputers, 2024-07-03:
I just saw, (https://github.com/moveit/moveit2/issues/1895) is probably related to this issue.

ukicomputers, 2024-07-03:
Also, when logged in with GitHub API token for ghcr.io in Docker, I get this (different) error now:
```
[+] Running 1/1
 ✘ cpu Error denied                                                                                                                           3.1s 
Error response from daemon: denied
```
ghcr.io credentials are not the problem

MartinsSmirnovs, 2024-07-12:
+1, I tried to go through tutorials using Docker (because manually built nodes crashed on startup despite the successful build) and got unauthorized error.

MartinsSmirnovs, 2024-07-13:
Well, it looks that moveit tutorial is unusable for now.

scrv, 2024-11-11:
+1, facing the same unauthorized request

sea-bass, 2024-11-17:
Fixed in https://github.com/moveit/moveit2_tutorials/pull/989

closed 2024-11-17 by sea-bass