moveit/moveit2_tutorials #925: Package 'moveit2_tutorials' not found

OmniLearner, 2024-07-17:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Binary: https://moveit.ros.org/install-moveit2/binary/

### Steps to reproduce
Followed instructions of this page:
https://moveit.picknik.ai/main/doc/tutorials/getting_started/getting_started.html

### Expected behaviour
It should run as the demo says

### Backtrace or Console output

$ ros2 launch moveit2_tutorials demo.launch.py

Package 'moveit2_tutorials' not found: "package 'moveit2_tutorials' not found, searching: ['/home/mraju42@tntech.edu/ws_moveit/install/moveit_resources_pr2_description', '/home/mraju42@tntech.edu/ws_moveit/install/moveit_resources_panda_description', '/home/mraju42@tntech.edu/ws_moveit/install/moveit_common', '/home/mraju42@tntech.edu/ros2_ws/install/my_robot_controller', '/opt/ros/humble']"

Can someone please help in this issue. Thanks in advance

muhidabid, 2024-07-27:
Hi, I am facing the same issue. I checked the folders and found `demo.launch.py` in `src -> moveit2_tutorials -> doc -> tutorials -> quickstart_in_rviz -> launch -> demo.launch.py`. What seems to be happening is that `ros2 launch` command only checks inside the `install` folder and not the `src` folder. 

I followed the [Getting Started Tutorial](https://moveit.picknik.ai/main/doc/tutorials/getting_started/getting_started.html#create-colcon-workspace) and it cloned the tutorials repo in the `src` folder.

nick-pape, 2024-11-19:
I'm facing the same issue on jazzy. I'm using the main branch of the tutorials repo. It looks like the tutorials actually build correctly (and can confirm they fail to build if I switch to humble branch), however, actually invoking the demo launch script fails.

```
> colcon build --mixin release
Starting >>> moveit_resources_panda_description
Starting >>> moveit_common
Starting >>> moveit_msgs
Starting >>> moveit_resources_pr2_description
Starting >>> srdfdom
Starting >>> geometric_shapes
Starting >>> moveit_resources_fanuc_description
Starting >>> moveit_resources_prbt_support
Starting >>> robotiq_description
Starting >>> kortex_api
Starting >>> rviz_visual_tools
Starting >>> serial
Starting >>> robotiq_controllers
Finished <<< moveit_common [0.67s]
Finished <<< moveit_resources_panda_description [0.70s]
Starting >>> rviz_marker_tools
Starting >>> moveit_resources_panda_moveit_config
Starting >>> dual_arm_panda_moveit_config
Finished <<< robotiq_description [0.71s]
Starting >>> kortex_description
Finished <<< moveit_resources_fanuc_description [0.76s]
Finished <<< moveit_resources_pr2_description [0.81s]
Finished <<< moveit_resources_prbt_support [0.79s]
Starting >>> moveit_resources_fanuc_moveit_config
Finished <<< kortex_api [0.80s]
Starting >>> kortex_driver
Finished <<< serial [0.89s]
Starting >>> robotiq_driver
Finished <<< robotiq_controllers [0.96s]
Finished <<< geometric_shapes [1.18s]
Finished <<< moveit_resources_panda_moveit_config [0.61s]
Finished <<< dual_arm_panda_moveit_config [0.60s]
Finished <<< kortex_description [0.58s]
Finished <<< rviz_marker_tools [0.68s]
Finished <<< moveit_resources_fanuc_moveit_config [0.57s]
Starting >>> moveit_resources
Finished <<< kortex_driver [0.62s]
Starting >>> kortex_bringup
Finished <<< srdfdom [1.52s]
Starting >>> moveit_configs_utils
Finished <<< rviz_visual_tools [1.49s]
Finished <<< robotiq_driver [0.68s]
Starting >>> robotiq_hardware_tests
Finished <<< moveit_resources [0.73s]
Finished <<< kortex_bringup [0.75s]
Finished <<< robotiq_hardware_tests [0.61s]
Finished <<< moveit_configs_utils [2.54s]
Finished <<< moveit_msgs [7.58s]
Starting >>> moveit_core
Starting >>> moveit_task_constructor_msgs
Finished <<< moveit_core [1.80s]
Starting >>> moveit_ros_occupancy_map_monitor
Starting >>> moveit_simple_controller_manager
Starting >>> moveit_resources_prbt_ikfast_manipulator_plugin
Starting >>> chomp_motion_planner
Starting >>> pilz_industrial_motion_planner_testutils
Starting >>> moveit_planners_stomp
Starting >>> pick_ik
Finished <<< moveit_task_constructor_msgs [2.34s]
Finished <<< moveit_resources_prbt_ikfast_manipulator_plugin [1.12s]
Finished <<< moveit_simple_controller_manager [1.18s]
Starting >>> moveit_plugins
Starting >>> moveit_ros_control_interface
Finished <<< moveit_ros_occupancy_map_monitor [1.25s]
Finished <<< chomp_motion_planner [1.24s]
Starting >>> moveit_ros_planning
[... 66 lines cut ...]
Finished <<< moveit_setup_assistant [3.16s]
Starting >>> kinova_gen3_7dof_robotiq_2f_85_moveit_config
Starting >>> moveit
Starting >>> kinova_gen3_6dof_robotiq_2f_85_moveit_config
Starting >>> kinova_gen3_lite_moveit_config
Finished <<< moveit_task_constructor_capabilities [2.83s]
Starting >>> moveit_task_constructor_demo
Finished <<< kinova_gen3_7dof_robotiq_2f_85_moveit_config [2.95s]
Finished <<< kinova_gen3_lite_moveit_config [3.02s]
Finished <<< moveit [3.14s]
Starting >>> moveit2_tutorials
Finished <<< kinova_gen3_6dof_robotiq_2f_85_moveit_config [3.16s]
Finished <<< moveit_task_constructor_visualization [3.63s]
Finished <<< moveit_task_constructor_demo [3.35s]
[Processing: moveit2_tutorials]
[Processing: moveit2_tutorials]
Finished <<< moveit2_tutorials [1min 16s]

Summary: 73 packages finished [1min 47s]
```

```
> ros2 launch moveit2_tutorials demo.launch.py
Package 'moveit2_tutorials' not found: "package 'moveit2_tutorials' not found, searching: ['/home/nickpape/ws_moveit2/install/pilz_industrial_motion_planner_testutils', '/home/nickpape/ws_moveit2/install/moveit_runtime', '/home/nickpape/ws_moveit2/install/moveit', '/home/nickpape/ws_moveit2/install/moveit_planners', '/home/nickpape/ws_moveit2/install/pilz_industrial_motion_planner', '/home/nickpape/ws_moveit2/install/moveit_ros_control_interface', '/home/nickpape/ws_moveit2/install/moveit_plugins', '/home/nickpape/ws_moveit2/install/moveit_simple_controller_manager', '/home/nickpape/ws_moveit2/install/moveit_setup_assistant', '/home/nickpape/ws_moveit2/install/moveit_setup_srdf_plugins', '/home/nickpape/ws_moveit2/install/moveit_setup_core_plugins', '/home/nickpape/ws_moveit2/install/moveit_setup_controllers', '/home/nickpape/ws_moveit2/install/moveit_setup_app_plugins', '/home/nickpape/ws_moveit2/install/moveit_setup_framework', '/home/nickpape/ws_moveit2/install/moveit_servo', '/home/nickpape/ws_moveit2/install/moveit_ros_trajectory_cache', '/home/nickpape/ws_moveit2/install/moveit_ros', '/home/nickpape/ws_moveit2/install/moveit_ros_visualization', '/home/nickpape/ws_moveit2/install/moveit_py', '/home/nickpape/ws_moveit2/install/moveit_hybrid_planning', '/home/nickpape/ws_moveit2/install/moveit_ros_planning_interface', '/home/nickpape/ws_moveit2/install/moveit_ros_benchmarks', '/home/nickpape/ws_moveit2/install/moveit_ros_warehouse', '/home/nickpape/ws_moveit2/install/moveit_ros_robot_interaction', '/home/nickpape/ws_moveit2/install/moveit_ros_perception', '/home/nickpape/ws_moveit2/install/moveit_resources_prbt_pg70_support', '/home/nickpape/ws_moveit2/install/moveit_resou [... cut ...] /moveit_resources_panda_moveit_config', '/home/nickpape/ws_moveit2/install/dual_arm_panda_moveit_config', '/home/nickpape/ws_moveit2/install/moveit_resources_panda_description', '/home/nickpape/ws_moveit2/install/moveit_resources_fanuc_moveit_config', '/home/nickpape/ws_moveit2/install/moveit_resources_fanuc_description', '/home/nickpape/ws_moveit2/install/moveit_msgs', '/home/nickpape/ws_moveit2/install/moveit_common', '/home/nickpape/ws_moveit2/install/geometric_shapes', '/opt/ros/jazzy']"
```

nick-pape, 2024-11-19:
I did get this working. Blew away everything in my workspace and started fresh. I followed these instructions:

https://moveit.ai/install-moveit2/source/

And then started from here: 

https://github.com/moveit/moveit2_tutorials/blob/main/doc/tutorials/getting_started/getting_started.rst#download-source-code-of-moveit-and-the-tutorials

sea-bass, 2024-11-23:
Are you running `source install/setup.bash` once the build is complete? This is needed to get the newly built packages to show up.

130s, 2026-11-16:
Closing as it's stale, also https://github.com/moveit/moveit2_tutorials/issues/925#issuecomment-2495524194 has got 3 :heart: emojis that I took as "that worked, thanks".