moveit/moveit2_tutorials #926: Colcon Build Error: Robotiq_driver package

shivaldube, 2024-07-18:
### Description

Not able to build the Colcon workspace, following the tutorial at
[Getting started tutorial](https://moveit.picknik.ai/main/doc/tutorials/getting_started/getting_started.html)

### Your environment
* ROS Distro: [Humble]
* OS Version: e.g. Ubuntu 22.04
* Source build (Tried both main and humble)

### Steps to reproduce
Follow the tutorial [Getting started tutorial](https://moveit.picknik.ai/main/doc/tutorials/getting_started/getting_started.html)

### Expected behaviour
Build colcon workspace without error.

### Backtrace or Console output
*
ws_moveit/src/ros2_robotiq_gripper/robotiq_driver/tests/test_robotiq_gripper_hardware_interface.cpp: In member function ‘virtual void robotiq_driver::test::TestRobotiqGripperHardwareInterface_load_urdf_Test::TestBody()’:
ws_moveit/src/ros2_robotiq_gripper/robotiq_driver/tests/test_robotiq_gripper_hardware_interface.cpp:79:114: error: no matching function for call to ‘hardware_interface::ResourceManager::ResourceManager(std::string&, rclcpp::node_interfaces::NodeClockInterface::SharedPtr, rclcpp::node_interfaces::NodeLoggingInterface::SharedPtr)’
   79 |   hardware_interface::ResourceManager rm(urdf, node.get_node_clock_interface(), node.get_node_logging_interface());
      |                                                                                                                  ^
In file included from /home/shivaldubey/ws_moveit/src/ros2_robotiq_gripper/robotiq_driver/tests/test_robotiq_gripper_hardware_interface.cpp:31:
/opt/ros/humble/include/hardware_interface/resource_manager.hpp:66:12: note: candidate: ‘hardware_interface::ResourceManager::ResourceManager(const string&, bool, bool)’
   66 |   explicit ResourceManager(
      |            ^~~~~~~~~~~~~~~
/opt/ros/humble/include/hardware_interface/resource_manager.hpp:67:36: note:   no known conversion for argument 2 from ‘rclcpp::node_interfaces::NodeClockInterface::SharedPtr’ {aka ‘std::shared_ptr<rclcpp::node_interfaces::NodeClockInterface>’} to ‘bool’
   67 |     const std::string & urdf, bool validate_interfaces = true, bool activate_all = false);
      |                               ~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~
/opt/ros/humble/include/hardware_interface/resource_manager.hpp:49:3: note: candidate: ‘hardware_interface::ResourceManager::ResourceManager()’
   49 |   ResourceManager();
      |   ^~~~~~~~~~~~~~~
/opt/ros/humble/include/hardware_interface/resource_manager.hpp:49:3: note:   candidate expects 0 arguments, 3 provided
gmake[2]: *** [tests/CMakeFiles/test_robotiq_gripper_hardware_interface.dir/build.make:76: tests/CMakeFiles/test_robotiq_gripper_hardware_interface.dir/test_robotiq_gripper_hardware_interface.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1037: tests/CMakeFiles/test_robotiq_gripper_hardware_interface.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< robotiq_driver [1.13s, exited with code 2]
Aborted  <<< moveit_msgs [1.67s]

Summary: 19 packages finished [1.91s]
  1 package failed: robotiq_driver
  1 package aborted: moveit_msgs
  1 package had stderr output: robotiq_driver
  48 packages not processed
*

Usimian, 2024-07-19:
This worked for me:
    git clone https://github.com/ros-planning/moveit2_tutorials --branch humble

MasterJinKim, 2024-07-21:
Same as you

`Starting >>> robotiq_driver
--- stderr: robotiq_driver                                                                             
/home/jin/ws_moveit/src/ros2_robotiq_gripper/robotiq_driver/tests/test_robotiq_gripper_hardware_interface.cpp: In member function ‘virtual void robotiq_driver::test::TestRobotiqGripperHardwareInterface_load_urdf_Test::TestBody()’:
/home/jin/ws_moveit/src/ros2_robotiq_gripper/robotiq_driver/tests/test_robotiq_gripper_hardware_interface.cpp:79:114: error: no matching function for call to ‘hardware_interface::ResourceManager::ResourceManager(std::string&, rclcpp::node_interfaces::NodeClockInterface::SharedPtr, rclcpp::node_interfaces::NodeLoggingInterface::SharedPtr)’
   79 |   hardware_interface::ResourceManager rm(urdf, node.get_node_clock_interface(), node.get_node_logging_interface());
      |                                                                                                                  ^
In file included from /home/jin/ws_moveit/src/ros2_robotiq_gripper/robotiq_driver/tests/test_robotiq_gripper_hardware_interface.cpp:31:
/opt/ros/iron/include/hardware_interface/hardware_interface/resource_manager.hpp:71:12: note: candidate: ‘hardware_interface::ResourceManager::ResourceManager(const string&, bool, bool, unsigned int, rclcpp::node_interfaces::NodeClockInterface::SharedPtr)’
   71 |   explicit ResourceManager(
      |            ^~~~~~~~~~~~~~~
/opt/ros/iron/include/hardware_interface/hardware_interface/resource_manager.hpp:72:36: note:   no known conversion for argument 2 from ‘rclcpp::node_interfaces::NodeClockInterface::SharedPtr’ {aka ‘std::shared_ptr<rclcpp::node_interfaces::NodeClockInterface>’} to ‘bool’
   72 |     const std::string & urdf, bool validate_interfaces = true, bool activate_all = false,
      |                               ~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~
/opt/ros/iron/include/hardware_interface/hardware_interface/resource_manager.hpp:52:3: note: candidate: ‘hardware_interface::ResourceManager::ResourceManager(unsigned int, rclcpp::node_interfaces::NodeClockInterface::SharedPtr)’
   52 |   ResourceManager(
      |   ^~~~~~~~~~~~~~~
/opt/ros/iron/include/hardware_interface/hardware_interface/resource_manager.hpp:52:3: note:   candidate expects 2 arguments, 3 provided
gmake[2]: *** [tests/CMakeFiles/test_robotiq_gripper_hardware_interface.dir/build.make:76：tests/CMakeFiles/test_robotiq_gripper_hardware_interface.dir/test_robotiq_gripper_hardware_interface.cpp.o] 错误 1
`

zzxmm99, 2024-07-23:
4 days ago,i colcon build --mixin release ,succeeded.Today,i tried it again in a new virtual machine,failed.Now,i meet the same quesition with you...

Mactarvish, 2024-07-24:
identical issue with humble under ubuntu22.04

zzxmm99, 2024-07-25:
> -

now，i think the code on official website(https://moveit.picknik.ai/main/doc/tutorials/getting_started/getting_started.html) can't use,because'If you are just getting started, we recommend you use the latest stable version of ROS (Iron) on Ubuntu 22.04 for the most seamless experience.'However,when i used Iron version and downloaded the code,'colcon build' still had problems.In a result,i think the official codes have problems.My solution is following this website(https://moveit.picknik.ai/humble/doc/tutorials/getting_started/getting_started.html),with Ubuntu 22.04 and ROS2 humble,the code'git clone --branch humble https://github.com/ros-planning/moveit2_tutorials' just clone the humble branch,after 'colcon build',we can follow the tutorials completely:),just the website no longer updated:(.

stephanie-eng, 2024-07-25:
If you're running main, what is your ros2_control version? You can check with:
```bash
apt list --installed | grep ros-rolling-ros2-control
```

This looks you may have the most recent [robotiq_driver](https://github.com/PickNikRobotics/ros2_robotiq_gripper) package (which you would get if you used `vcs import` as per the tutorial documentation recently, but not the compatible version of ros2_control.

MasterJinKim, 2024-07-27:
I found that the package robotiq_driver always fail to build. Here is my solution.
Remove the original ros_2robotiq_gripper from src and pull it again from the Humble branch.
`git clone  https://github.com/PickNikRobotics/ros2_robotiq_gripper.git  --branch humble`
It works for me.

closed 2024-08-10 by sea-bass