moveit/moveit2_tutorials #931: colcon build --mixin release CMake error gz_sim_vendor

muhidabid, 2024-07-27:
### Description

### Your environment
* ROS Distro: [humble|iron]
* OS Version: Ubuntu 22.04.4 Jammy Jellyfish (VirtualBox 19GB RAM, 50GB storage) 
* Build: Followed [Getting Started Tutorial](https://moveit.picknik.ai/main/doc/tutorials/getting_started/getting_started.html#create-colcon-workspace)

### Steps to reproduce
Installed ROS2 Iron following [this page](https://docs.ros.org/en/iron/Installation/Ubuntu-Install-Debians.html). Then setup moveit2 following [this page](https://moveit.picknik.ai/main/doc/tutorials/getting_started/getting_started.html#create-colcon-workspace)

### Expected behaviour
The project should build. I did successfully build it once following the [linux source build](https://moveit.ros.org/install-moveit2/source/) however I am failing to do so again.

### Backtrace or Console output

[Gist link](https://gist.github.com/muhidabid/fc0d3eb6c1aada47d0f515fe89427f73)
<script src="https://gist.github.com/muhidabid/fc0d3eb6c1aada47d0f515fe89427f73.js"></script>

christiandumasrobo, 2024-07-28:
Exactly the same issue today, will be following this thread.

yusufkizgin, 2024-07-30:
Did you solve this issue? I have same issue.

muhidabid, 2024-07-30:
I've been trying but no luck so far @yusufkizgin

T3rO, 2024-07-31:
For ROS Iron I deleted gz_ros2_control from ws_moveit/src then download again, but the iron branch: git clone https://github.com/ros-controls/gz_ros2_control -b iron

muhidabid, 2024-07-31:
@T3rO I just tried that but it gives me a new error upon colcon build: [Error output gist](https://gist.github.com/muhidabid/b66674ea0f81c0ac04623db913726758)

li-huanhuan, 2024-08-01:
me too

ubuntu22.04.4 
humble

[image]

sea-bass, 2024-08-01:
Are you all checking out the `humble` branch of this repo?

There have been some recent changes on the `main` branch here regarding the `ros2_kortex` repo (which pulls in the offending dependency), but there are meant to track ROS 2 Rolling on Ubuntu 24.04 going forward.

T3rO, 2024-08-01:
> @T3rO I just tried that but it gives me a new error upon colcon build: [Error output gist](https://gist.github.com/muhidabid/b66674ea0f81c0ac04623db913726758)

Yes I had this error too. I used this fix: https://github.com/PickNikRobotics/ros2_robotiq_gripper/issues/60

T3rO, 2024-08-01:
@sea-bass I think an Iron branch in this repo would be nice, and then also mention that a specific branch has to be checked out in the installation guide

sea-bass, 2024-08-01:
> @sea-bass I think an Iron branch in this repo would be nice, and then also mention that a specific branch has to be checked out in the installation guide

I see. That one is also due to there being newer changes on Rolling, and you could grab the humble branch of that repo.

All in all, it seems that all these new updates need to be handled correctly on non-rolling versions of the tutorials repo.

MartinsSmirnovs, 2024-08-01:
I have a feeling that I will never be able to go through moveit tutorials. First #918 which does not allow to get the environment itself and now this.

muhidabid, 2024-08-01:
Wanted to try @T3rO method but I am getting a new error upon redoing the setup and building :')

[image]

sea-bass, 2024-08-01:
OK yeah so the issue is that the `gz_sim_vendor` package is only available on Jazzy and Rolling, per https://index.ros.org/search/?term=gz_sim_vendor

Meaning the `main` branch of this repo is not currently compatible with humble/iron.

T3rO, 2024-08-02:
Is there a branch in this repo that is compatible with iron?

muhidabid, 2024-08-02:
@sea-bass I tried rolling but that gives me a new error 

[image]

sea-bass, 2024-08-02:
> Is there a branch in this repo that is compatible with iron?

Nope, I think there are only branches for LTS releases, meaning a `humble` branch and then a `main` (which right now is supposed to work for Jazzy/Rolling).

With luck, the `humble` branch should also work with Iron as the changes were fairly minimal and they use the same supported Ubuntu version (22.04).

sea-bass, 2024-08-02:
> @sea-bass I tried rolling but that gives me a new error

As documented in the [getting started tutorial](rosdep install --from-paths src -y --ignore-src), try this from your workspace root to install any missing packages like the one you're pointing out?

```
rosdep install --from-paths src -y --ignore-src
```

muhidabid, 2024-08-02:
@sea-bass I followed the getting started tutorial and installed from my workspace root using `rosdep install --from-paths src -y --ignore-src` but I am still getting the same error

sea-bass, 2024-08-02:
> @sea-bass I followed the getting started tutorial and installed from my workspace root using `rosdep install --from-paths src -y --ignore-src` but I am still getting the same error

You should check if the command ran successfully, i.e., it found all the packages or it failed to resolve some. Also it helps to do a `sudo apt update` first to make sure your repos are up to date.

Anyways, I am unable to debug every small issue for everyone, so just sharing some pointers for people to consider and/or contribute fixes as you find them.

All the best!

Thgm01, 2024-08-03:
I have the same error, but first I deleted the rosdep list using `sudo rm /etc/ros/rosdep/sources.list.d/20-default.list` and deleted the ws with `rm -rf ws_moveit/` then, I Followed [Getting Started Tutorial](https://moveit.picknik.ai/main/doc/tutorials/getting_started/getting_started.html#create-colcon-workspace) but instead use `git clone https://github.com/moveit/moveit2_tutorials` i used `git clone https://github.com/moveit/moveit2_tutorials -b humble` and it worked for me.

yusufkizgin, 2024-08-03:
actually i deleted even my virtual machine but it doesn't work :/

kimhw1009, 2024-08-04:
> I have the same error, but first I deleted the rosdep list using `sudo rm /etc/ros/rosdep/sources.list.d/20-default.list` and deleted the ws with `rm -rf ws_moveit/` then, I Followed the [Getting Started Tutorial](https://moveit.picknik.ai/main/doc/tutorials/getting_started/getting_started.html#create-colcon-workspace) but instead use `git clone https://github.com/moveit/moveit2_tutorials` i used `git clone https://github.com/moveit/moveit2_tutorials -b humble` and it worked for me.
> 
> edit: I'm using ROS Humble and Ubuntu 22.04

I tried it this way.
It looked like it would be resolved, but there were still some failures.
[image]

kimhw1009, 2024-08-04:
> I have the same error, but first I deleted the rosdep list using `sudo rm /etc/ros/rosdep/sources.list.d/20-default.list` and deleted the ws with `rm -rf ws_moveit/` then, I Followed the [Getting [...]

I tried this way and it was successful.
Change to ROS Humble.

yusufkizgin, 2024-08-04:
> I have the same error, but first I deleted the rosdep list using `sudo rm /etc/ros/rosdep/sources.list.d/20-default.list` and deleted the ws with `rm -rf ws_moveit/` then, I Followed the [Getting [...]

I tried this and it worked. obrigado :)

muhidabid, 2024-08-04:
> I have the same error, but first I deleted the rosdep list using `sudo rm /etc/ros/rosdep/sources.list.d/20-default.list` and deleted the ws with `rm -rf ws_moveit/` then, I Followed the [Getting [...]

It took a few tries but this worked for me as well. Thank you! @Thgm01

sea-bass, 2024-08-10:
Thanks to everyone for sharing your solutions!

As a recap: Since the `main` branch of this repo tracks ROS 2 rolling, it is not guaranteed to work on Humble. So, as many have pointed out, you can use the `humble` branch of the tutorials on Humble.

closed 2024-08-10 by sea-bass

Ivan-Rod-Diz, 2024-08-29:
> I have the same error, but first I deleted the rosdep list using `sudo rm /etc/ros/rosdep/sources.list.d/20-default.list` and deleted the ws with `rm -rf ws_moveit/` then, I Followed the [Getting [...]

This also worked fo me :)

navdotnetreqs, 2024-09-30:
Out of luck with iron, not able to get it to work. Have to say the entire several months I've been learning ros2 have been littered with issues of broken dependencies, compile errors, breaking changes, non-compatible options, outdated tutorials that don't work, nodes that don't start, debugging, opening issues, stabbing config files and hairs that fall out..