moveit/moveit2_tutorials #936: MoveIt2 Docker Tutorial Failing -- Can't Download Image

MichaelrMentele, 2024-08-01:
### Description

Second attempt to do MoveIt2 tutorial from #935. Following along with https://moveit.picknik.ai/main/doc/how_to_guides/how_to_setup_docker_containers_in_ubuntu.html

Getting stuck unable to pull down the moveit2_tutorials image.

### Your environment
* ROS Distro: Not relevant just pulling docker image for move it
* OS Version: Ubuntu 24.04
* Source or Binary build? n/a -- using docker
* If binary, which release version?
* If source, which git commit or tag?

### Steps to reproduce
Trying to run:
```
DOCKER_IMAGE=rolling-tutorial docker compose run --rm --name moveit2_container gpu
```

### Expected behaviour
The image should pull down! I can pull other public images down. Is this thing private by accident..?

### Backtrace or Console output

Running the docker compose command gives:
```
DOCKER_IMAGE=rolling-tutorial docker compose run --rm --name moveit2_container gpu

[+] Running 1/1
 ✘ gpu Error denied                                                                                                       0.7s
Error response from daemon: denied
```

If I try to directly pull the image:
```
sudo docker pull ghcr.io/moveit/moveit2_tutorials:rolling-tutorial
Error response from daemon: Head "https://ghcr.io/v2/moveit/moveit2_tutorials/manifests/rolling-tutorial": unauthorized
```

zzv2, 2024-08-26:
This happened to me as well, the fix was to change the `docker-compose.yaml` file to point to the new location of the image.

These lines:
```
image: ghcr.io/moveit/moveit2_tutorials:$DOCKER_IMAGE
```
needed to change to this:
```
image: ghcr.io/ros-planning/moveit2_tutorials:$DOCKER_IMAGE
```

I guess the image was moved from `moveit` org to `ros-planning`?

brammmieee, 2024-09-20:
I created a pull request to fix this bug [here](https://github.com/moveit/moveit2_tutorials/pull/972)

sea-bass, 2024-11-17:
Fixed in https://github.com/moveit/moveit2_tutorials/pull/989

closed 2024-11-17 by sea-bass