moveit/moveit2_tutorials #942: "MoveIt Quickstart in RViz" seems to be out of date

muhidabid, 2024-08-08:
### Description

I successfully ran and went through the [MoveIt Quickstart in RViz](https://moveit.picknik.ai/main/doc/tutorials/quickstart_in_rviz/quickstart_in_rviz_tutorial.html) however I noticed a few details in the tutorial to be different than the demo I ran:

- World is not empty upon launching the demo for the first time
- There seems to be no `kinova` robot but a `panda` robot
- `Fixed Frame` field does not have a `/base_link` option but rather is already set to `panda_0`
- `Planned Path` menu drop-down in Displays seems to have been renamed to `Trajectory`
- under MotionPlanning in Planning Request there is no Planning Group called `manipulator` but rather `panda_arm`

There may be few more discrepancies which I may have missed. 

**🚀 I am new to opensource and if this page needs to be updated I would like to volunteer to help do so. Please guide me accordingly.**

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04

### Steps to reproduce
Step 1: [Getting Started](https://moveit.picknik.ai/main/doc/tutorials/getting_started/getting_started.html)
Step 2: [MoveIt Quickstart in RViz](https://moveit.picknik.ai/main/doc/tutorials/quickstart_in_rviz/quickstart_in_rviz_tutorial.html) 

### Expected behaviour
Documentation must be up to date with the current demo.

sea-bass, 2024-08-08:
If you see a Franka robot, then you must be using the Humble version of the tutorials. Main used a Kinova.

So you can either use the main branch of the tutorials (noting this only works on Jazzy/Rolling now), or follow along with the Humble tutorials.

https://moveit.picknik.ai/humble/doc/tutorials/quickstart_in_rviz/quickstart_in_rviz_tutorial.html

muhidabid, 2024-08-08:
Oh makes sense. Thank you

closed 2024-08-08 by muhidabid