moveit/moveit2_tutorials #945: Isaac Panda Hand Controller Issue

BingPeng0, 2024-08-14:
### Description

I am trying to control Franka Panda in Omniverse IssacSim. Plan & Execute for arm works perfectly fine except for hand. Occasionally there will be error when executing the command for the hand to open or close state where the hand will not do so immediately, delay time is about 5-10s. If I plan&execute the arm to current state and plan hand again it will work, but when i move the arm to another state and execute the hand to another state again there will be delay again.

Error log:
[move_group-5] [INFO] [1723619674.828857349] [move_group.moveit.moveit.ros.move_group.move_action]: Solution was found and executed.
[rviz2-1] [INFO] [1723619674.829303086] [rviz2.moveit.ros.move_group_interface]: Plan and Execute request complete!
[rviz2-1] [INFO] [1723619676.850433565] [rviz2.moveit.ros.robot_interaction]: No active joints or end effectors found for group 'hand'. Make sure that kinematics.yaml is loaded in this node's namespace.
[rviz2-1] [INFO] [1723619676.850790057] [rviz2.moveit.ros.motion_planning_frame]: group hand
[rviz2-1] [INFO] [1723619676.850806197] [rviz2.moveit.ros.motion_planning_frame]: Constructing new MoveGroup connection for group 'hand' in namespace ''
[rviz2-1] [INFO] [1723619676.854240231] [rviz2.moveit.ros.move_group_interface]: Ready to take commands for planning group hand.
[move_group-5] [INFO] [1723619681.462298786] [move_group.moveit.moveit.ros.move_group.move_action]: MoveGroupMoveAction: Received request
[move_group-5] [INFO] [1723619681.462401024] [move_group.moveit.moveit.ros.move_group.move_action]: executing..
[rviz2-1] [INFO] [1723619681.462424190] [rviz2.moveit.ros.move_group_interface]: Plan and Execute request accepted
[move_group-5] [INFO] [1723619681.468468300] [move_group.moveit.moveit.ros.move_group.move_action]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[move_group-5] [INFO] [1723619681.468522174] [move_group.moveit.moveit.ros.add_time_optimal_parameterization]: Planning attempt 1 of at most 1
[move_group-5] [INFO] [1723619681.468544770] [move_group.moveit.moveit.ros.move_group.capability]: Using planning pipeline 'ompl'
[move_group-5] [INFO] [1723619681.468565404] [move_group]: Calling PlanningRequestAdapter 'ResolveConstraintFrames'
[move_group-5] [INFO] [1723619681.468582277] [move_group]: Calling PlanningRequestAdapter 'ValidateWorkspaceBounds'
[move_group-5] [INFO] [1723619681.468589087] [move_group]: Calling PlanningRequestAdapter 'CheckStartStateBounds'
[move_group-5] [INFO] [1723619681.468601364] [move_group]: Calling PlanningRequestAdapter 'CheckStartStateCollision'
[move_group-5] [INFO] [1723619681.468726957] [move_group.moveit.moveit.planners.ompl.model_based_planning_context]: Planner configuration 'hand' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[move_group-5] [INFO] [1723619681.468770972] [move_group]: Calling Planner 'OMPL'
[move_group-5] [INFO] [1723619681.488519276] [move_group]: Calling PlanningResponseAdapter 'AddTimeOptimalParameterization'
[move_group-5] [INFO] [1723619681.492443035] [move_group]: Calling PlanningResponseAdapter 'ValidateSolution'
[move_group-5] [INFO] [1723619681.492654699] [move_group]: Calling PlanningResponseAdapter 'DisplayMotionPath'
[move_group-5] [WARN] [1723619681.492710793] [move_group.moveit.moveit.ros.planning_pipeline]: The planner plugin did not fill out the 'planner_id' field of the MotionPlanResponse. Setting it to the planner ID name of the MotionPlanRequest assuming that the planner plugin does warn you if it does not use the requested planner.
[move_group-5] [INFO] [1723619681.492752719] [move_group.moveit.moveit.plugins.simple_controller_manager]: Returned 2 controllers in list
[move_group-5] [INFO] [1723619681.492761883] [move_group.moveit.moveit.plugins.simple_controller_manager]: Returned 2 controllers in list
[move_group-5] [INFO] [1723619681.492788276] [move_group.moveit.moveit.ros.trajectory_execution_manager]: Validating trajectory with allowed_start_tolerance 0.01
[move_group-5] [INFO] [1723619681.498092843] [move_group.moveit.moveit.ros.trajectory_execution_manager]: Starting trajectory execution ...
[move_group-5] [INFO] [1723619681.498123937] [move_group.moveit.moveit.plugins.simple_controller_manager]: Returned 2 controllers in list
[move_group-5] [INFO] [1723619681.498136249] [move_group.moveit.moveit.plugins.simple_controller_manager]: Returned 2 controllers in list
[ros2_control_node-6] [INFO] [1723619681.498293083] [panda_hand_controller]: Received & accepted new action goal
[move_group-5] [WARN] [1723619682.548594447] [move_group.moveit.moveit.ros.trajectory_execution_manager]: Controller handle panda_hand_controller reports status ABORTED
[move_group-5] [INFO] [1723619682.548613883] [move_group.moveit.moveit.ros.trajectory_execution_manager]: Completed trajectory execution with status ABORTED ...
[move_group-5] [INFO] [1723619682.558217201] [move_group.moveit.moveit.ros.move_group.move_action]: CONTROL_FAILED
[rviz2-1] [INFO] [1723619682.558509056] [rviz2.moveit.ros.move_group_interface]: Plan and Execute request aborted
[rviz2-1] [ERROR] [1723619682.559070520] [rviz2.moveit.ros.move_group_interface]: MoveGroupInterface::move() failed or timeout reached

Successful log:
[move_group-5] [INFO] [1723619952.053674230] [move_group.moveit.moveit.ros.move_group.move_action]: MoveGroupMoveAction: Received request
[rviz2-1] [INFO] [1723619952.053785150] [rviz2.moveit.ros.move_group_interface]: Plan and Execute request accepted
[move_group-5] [INFO] [1723619952.053813042] [move_group.moveit.moveit.ros.move_group.move_action]: executing..
[move_group-5] [INFO] [1723619952.058187350] [move_group.moveit.moveit.ros.move_group.move_action]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[move_group-5] [INFO] [1723619952.058239791] [move_group.moveit.moveit.ros.add_time_optimal_parameterization]: Planning attempt 1 of at most 1
[move_group-5] [INFO] [1723619952.058272732] [move_group.moveit.moveit.ros.move_group.capability]: Using planning pipeline 'ompl'
[move_group-5] [INFO] [1723619952.058299047] [move_group]: Calling PlanningRequestAdapter 'ResolveConstraintFrames'
[move_group-5] [INFO] [1723619952.058316891] [move_group]: Calling PlanningRequestAdapter 'ValidateWorkspaceBounds'
[move_group-5] [INFO] [1723619952.058324711] [move_group]: Calling PlanningRequestAdapter 'CheckStartStateBounds'
[move_group-5] [INFO] [1723619952.058335697] [move_group]: Calling PlanningRequestAdapter 'CheckStartStateCollision'
[move_group-5] [INFO] [1723619952.058475900] [move_group.moveit.moveit.planners.ompl.model_based_planning_context]: Planner configuration 'hand' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[move_group-5] [INFO] [1723619952.058511554] [move_group]: Calling Planner 'OMPL'
[move_group-5] [INFO] [1723619952.074668788] [move_group]: Calling PlanningResponseAdapter 'AddTimeOptimalParameterization'
[move_group-5] [INFO] [1723619952.076859958] [move_group]: Calling PlanningResponseAdapter 'ValidateSolution'
[move_group-5] [INFO] [1723619952.077066427] [move_group]: Calling PlanningResponseAdapter 'DisplayMotionPath'
[move_group-5] [WARN] [1723619952.077181199] [move_group.moveit.moveit.ros.planning_pipeline]: The planner plugin did not fill out the 'planner_id' field of the MotionPlanResponse. Setting it to the planner ID name of the MotionPlanRequest assuming that the planner plugin does warn you if it does not use the requested planner.
[move_group-5] [INFO] [1723619952.077241832] [move_group.moveit.moveit.plugins.simple_controller_manager]: Returned 2 controllers in list
[move_group-5] [INFO] [1723619952.077253661] [move_group.moveit.moveit.plugins.simple_controller_manager]: Returned 2 controllers in list
[move_group-5] [INFO] [1723619952.077289502] [move_group.moveit.moveit.ros.trajectory_execution_manager]: Validating trajectory with allowed_start_tolerance 0.01
[move_group-5] [INFO] [1723619952.078294374] [move_group.moveit.moveit.ros.trajectory_execution_manager]: Starting trajectory execution ...
[move_group-5] [INFO] [1723619952.078356409] [move_group.moveit.moveit.plugins.simple_controller_manager]: Returned 2 controllers in list
[... cut ...]
[main-humble-tutorial-source](https://hub.docker.com/layers/moveit/moveit2/main-humble-tutorial-source/images/sha256-62ce5f8ab1de0b2091fa335e099c5de4e7de38bd49927e3d9fd27b06103d7e97?context=explore)

### Steps to reproduce
* Docker : tag - main-humble-tutorial-source 
[main-humble-tutorial-source](https://hub.docker.com/layers/moveit/moveit2/main-humble-tutorial-source/images/sha256-62ce5f8ab1de0b2091fa335e099c5de4e7de38bd49927e3d9fd27b06103d7e97?context=explore)

### Expected behaviour
The hand will execute immediately and no error.

BingPeng0, 2024-08-14:
Have tried to increase the allowed_execution_duration_scaling from 1.2 to 1.5, 2.0, 10.0. But still does not work.

assaulz, 2024-11-02:
I have the same issue.

joanne40226, 2025-07-08:
hi, i also encountered the same issue. Have you solved it? Thank you.