moveit/moveit2_tutorials #947: Failed to build moveit2_tutorials

moisesrc13, 2024-08-25:
### Description

Failed to build

### Your environment
* ROS Distro: [Foxy|Galactic|Humble|Rolling]
* OS Version: e.g. Ubuntu 22.04
* Source or Binary build?
* If binary, which release version?
* If source, which git commit or tag?

### Steps to reproduce
following instructions here https://moveit.picknik.ai/main/doc/tutorials/getting_started/getting_started.html
then edit the file ` moveit2_tutorials/moveit2_tutorials.repos` to point to `iron` version instead of master

### Expected behaviour
Expected all packages to build built

### Backtrace or Console output

```
--- stderr: robotiq_driver
/home/moisesrc/ws_moveit/src/ros2_robotiq_gripper/robotiq_driver/tests/test_robotiq_gripper_hardware_interface.cpp: In member function ‘virtual void robotiq_driver::test::TestRobotiqGripperHardwareInterface_load_urdf_Test::TestBody()’:
/home/moisesrc/ws_moveit/src/ros2_robotiq_gripper/robotiq_driver/tests/test_robotiq_gripper_hardware_interface.cpp:79:114: error: no matching function for call to ‘hardware_interface::ResourceManager::ResourceManager(std::string&, rclcpp::node_interfaces::NodeClockInterface::SharedPtr, rclcpp::node_interfaces::NodeLoggingInterface::SharedPtr)’
   79 | erface::ResourceManager rm(urdf, node.get_node_clock_interface(), node.get_node_logging_interface());
      |                                                                                                    ^

In file included from /home/moisesrc/ws_moveit/src/ros2_robotiq_gripper/robotiq_driver/tests/test_robotiq_gripper_hardware_interface.cpp:31:
/opt/ros/iron/include/hardware_interface/hardware_interface/resource_manager.hpp:71:12: note: candidate: ‘hardware_interface::ResourceManager::ResourceManager(const string&, bool, bool, unsigned int, rclcpp::node_interfaces::NodeClockInterface::SharedPtr)’
   71 |   explicit ResourceManager(
      |            ^~~~~~~~~~~~~~~
/opt/ros/iron/include/hardware_interface/hardware_interface/resource_manager.hpp:72:36: note:   no known conversion for argument 2 from ‘rclcpp::node_interfaces::NodeClockInterface::SharedPtr’ {aka ‘std::shared_ptr<rclcpp::node_interfaces::NodeClockInterface>’} to ‘bool’
   72 |     const std::string & urdf, bool validate_interfaces = true, bool activate_all = false,
      |                               ~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~
/opt/ros/iron/include/hardware_interface/hardware_interface/resource_manager.hpp:52:3: note: candidate: ‘hardware_interface::ResourceManager::ResourceManager(unsigned int, rclcpp::node_interfaces::NodeClockInterface::SharedPtr)’
   52 |   ResourceManager(
      |   ^~~~~~~~~~~~~~~
/opt/ros/iron/include/hardware_interface/hardware_interface/resource_manager.hpp:52:3: note:   candidate expects 2 arguments, 3 provided
gmake[2]: *** [tests/CMakeFiles/test_robotiq_gripper_hardware_interface.dir/build.make:76: tests/CMakeFiles/test_robotiq_gripper_hardware_interface.dir/test_robotiq_gripper_hardware_interface.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1037: tests/CMakeFiles/test_robotiq_gripper_hardware_interface.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< robotiq_driver [53.6s, exited with code 2]
Aborted  <<< gz_ros2_control [1min 31s]
Aborted  <<< rviz_visual_tools [2min 50s]
Aborted  <<< moveit_msgs [3min 8s]

Summary: 16 packages finished [3min 8s]
  1 package failed: robotiq_driver
  3 packages aborted: gz_ros2_control moveit_msgs rviz_visual_tools
  1 package had stderr output: robotiq_driver
  53 packages not processed
  ```

sea-bass, 2024-08-25:
This error has to do with building a package that requires the latest (Jazzy/Rolling) version of `ros2_control`.

Which ROS distro are you trying to do this in? The `main` branch of this repo currently only works on Jazzy/Rolling, and if you are on Humble/Iron you want to use the `humble` branch of this repo instead.

moisesrc13, 2024-08-26:
> This error has to do with building a package that requires the latest (Jazzy/Rolling) version of `ros2_control`.
>
> Which ROS distro are you trying to do this in? The `main` branch of this repo [...]

Thanks a lot!, yes it worked after I moved to the correct branch.

closed 2024-08-26 by moisesrc13